MCUdude/MegaCore #2: collaborate

frankensteinXOXO, 2016-04-08:
I am working on few atmega128 boards, do you want to collaborate? also, your circuit, the ISCP part is wrong. https://www.youtube.com/watch?v=lfK-5zmFMo4

MCUdude, 2016-04-08:
Hi! How can i collaborate? :smile: I've been looking through the bare bone circuit and the pinout diagram. How's the ISP part wrong?

frankensteinXOXO, 2016-04-08:
I have tested your core with arduino, not extensively, but it works great. I am sure there are some optimizations. or is there any function from arduino that have not included that I can help you working on? Have you tested your circuit before posting it? Here is the correct isp header. https://github.com/frankensteinXOXO/ArduinoProject/blob/master/Juno_Pro/atmega128-minimal-lg.png

MCUdude, 2016-04-08:
Hmm. How's that possible? The data sheet for the ATmega128 specifies that the SPI / ISP pins are MOSI, MISO and SCK. These are located at PB1, PB2 and PB3

MCUdude, 2016-04-08:
The circuit error is now fixed :smile: I don't know if It's possible to do some more optimizations on the core. It's a recent fork of the [Arduino Github page](https://github.com/arduino/Arduino), so their the one handling all the changes. My job is to keep this core synced with their while still making it work for the ATmega64/128.

frankensteinXOXO, 2016-04-08:
Okay sounds great. I saw your circuit. Maybe put the LED on SCK/PB1 is better? so you can see it flashing when uploading with ISCP. Also for the pinout, I think it's more logical to group them using ports, ie, PA0=D0; PA1=D2.. and etc, but it's fine as-is. I don't know if arduino core is optimized for an AVR with 128kb of flash memory, not to say many hidden functions within the MCU. Maybe we can work together to bring out a better product while being 100% compatible with arduino? I was thinking about adding more functions. For starter the RTC function within the atmega128 is very interesting. Least we can do are adding some standard code for buttons and LED display? Meanwhile, I will test your core, and will let you know if there is any bug or problem. I am also working on an atmega16 board. I will try your other core out. I am sure it's also great. Maybe make a core for attiny chips too? Few out there are not very functional, with large memory usage and lacking of arduino functions. keep up the good work.

MCUdude, 2016-04-08:
The reason why I made the pinout like I did was to match the "MightyCore" way to define Arduino pins (the same counter clockwise naming is also done for the Arduino UNO).
I've been working with the DIP-40 series (ATmega16/32/164/324/644/1284) for a long time, but I recently discovered the ATmega128's really low price at Ebay and AliExpress. I don't think the core can be optimized for a specific flash size, but I'm a fan of the idea to add "Arduino like" functions (or maybe a core library, just to make it clean) to the core by adding "hidden" functions! I don't think I'll create an ATtiny core. (Dr Azzy's core)[https://github.com/SpenceKonde/ATTinyCore] is definitely the best core for the ATtinys out there. I know he's done some fantastic work optimizing the core functions :smile: If you want to have a look at some reeealy optimized code, have a look at the (AVR examples)[https://github.com/MCUdude/MegaCore/tree/master/avr/libraries/AVR_examples].

frankensteinXOXO, 2016-04-09:
For a small 28 pins uno, it's logical to use pin number as pinout numbering. but for a 64 pins 128, it's sort of confusing to someone who used to AVR ports. For someone came from arduino, it's totally fine. I am not complaining. it's great as is.  I already made 2 boards using your pinout numbers. Just keep in mind that using port as numbering is best of both world. ATmega128 is really a great chip, cheap (you can get better price if you buy from some wholesaler), lot of pins, and 128kb of memory is huge compared to 32kb for most arduino. I am going to make few more boards using it as educational purpose for both electronic and programming. "arduino like" functions that you mentioned is exactly what I have been talking about. Arduino IDE has been updating very frequently since, but their function is still pretty much the same. no new function added. very disappointing. getting more out of the mcu is one thing. enhancing and adding more feature is another thing that I want to work on. For example: digitalWrite(13, 1); turn on pin 13; how about having a function to turn on all 8 pins in a port at the same time? : digitalWritePort (portd, 0B11111111); instead of using a bool use a byte. that is a silly example, but I hope you get my point. adding some essential library as default for the core is another thing. since there are 128kb in the 128, memory is not a problem.   There are a lot of great libraries out there already. we can pick some and make a solid collection of essential and useful library, from blinking a led to pushing a button. making it much easier for beginners to use, yet powerful for advance user.

MCUdude, 2016-06-17:
I'm not going to add additional Arduino functions to this core. All core changes should come from the Arduino team to make sure all Arduino cores includes the same functionality. Additional functions, such as digitalWritePort can be implemented using an external header file

closed 2016-06-17 by MCUdude