MCUdude/MegaCore #3: out of range  address 0x200000

roberto74, 2016-04-13:
hi everyone i try to upload the bootloader on atmega64, i make the minimal board with 16Mhz cristal, an external 5v power supply, modify the fuses on board.txt to up to 4k space for bootloader and use an arduino uno as isp, but when try to upload i have this issue..

avrdude: ERROR: address 0x20000 out of range at line 40 of C:\Users.......10/bootloaders/optiboot/atmega64/optiboot_atmega64_16MHz.hex

someone can help to me please? thanks

MCUdude, 2016-04-13:
I haven't tested this core at the ATmega64 yet, I'm waiting for it to arrive in the mail. But why do you need to modify the fuse setting to make the bootloader space larger? The optiboot bootloader fits well within 512 words

roberto74, 2016-04-13:
Thank you so much for fast reply,"I am reading out of range" so in many desperate attemps ('m not a programmer) I try to make space larger...
Yes this error is the same with the standard fuse settings or other clock frequency options
this is the actual settings 
#make atmega64 AVR_FREQ=16000000L BAUD_RATE=57600 UART=0 LED=B5 LED_START_FLASHES=2 -BIGBOOT
64.menu.clock.16MHz_external=16MHz external (BOD 2.7v)
64.menu.clock.16MHz_external.upload.speed=57600
64.menu.clock.16MHz_external.bootloader.low_fuses=0xbf
64.menu.clock.16MHz_external.bootloader.high_fuses=0xc8
64.menu.clock.16MHz_external.bootloader.extended_fuses=0xff
64.menu.clock.16MHz_external.bootloader.file=optiboot/atmega64/optiboot_atmega64_16MHz.hex
64.menu.clock.16MHz_external.build.f_cpu=16000000L

MCUdude, 2016-05-05:
I'm still waiting for the ATmega64 to arrive in the mail. I'll look into this as soon as I get it 😉

MCUdude, 2016-05-18:
I think it should be OK now. The address in the makefile was out of range

roberto74, 2016-05-18:
Thanks i try it in the next day and give to you a feedback

Roberto S.

-------- Messaggio originale --------

CRI73, 2016-05-19:
hi guys 
I try new version but still not work :-(

This is avrdude msg:
******************************************************************
avrdude: verifying ...
avrdude: verification error, first mismatch at byte 0xfd00
         0xff != 0x84
avrdude: verification error; content mismatch
******************************************************************
tnks

MCUdude, 2016-05-19:
hmm. That seems like a hardware related issue. What bootloader version did you burn? 16MHz, 8MHz?

please turn on verbose upload in the IDE settings and pust the full output :)

chamanirad, 2016-05-19:
Hi ,
First of all a million thanks for providing the boot loader. Burnt it using Arduino interface and USBASP programmer worked like a charm.
One question: How can I change the USART from 0 to 1. The usb to serial module on my Atmega128 board is connected to USART 1 so I have to wire every thing again to get to USART0. Also do I need Linux to compile the boot-loader again?
Many thanks

MCUdude, 2016-05-19:
Hi! If you want to use UART 1, you'll have to recompile the bootloader. I'm using a mac, so I didn't have to install extra software in order to compile it (It's UNIX based). If you want I can compile the bootloader for you. What clock speed are you running?

CRI73, 2016-05-20:
Hi
This is a verbose report:

        AVR Part                      : ATmega64
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PA0
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom         4    20    64    0 no       2048    8      0  9000  9000 0xff 0xff
           flash         33     6   128    0 yes     65536  256    256  4500  4500 0xff 0xff
           lfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           lock           0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           calibration    0     0     0    0 no          4    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : usbasp
         Description     : USBasp, http://www.fischl.de/usbasp/

avrdude: auto set sck period (because given equals null)
avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude: Device signature = 0x1e9602
avrdude: NOTE: "flash" memory has been specified, an erase cycle will be performed
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: auto set sck period (because given equals null)
avrdude: reading input file "/Users/omegapro/Google Drive/Arduino/hardware/MegaCore/avr/bootloaders/optiboot/atmega64/optiboot_atmega64_16MHz.hex"
avrdude: writing flash (65536 bytes):

Writing | ################################################## | 100% 0.00s

avrdude: 65536 bytes of flash written
avrdude: verifying flash memory against /Users/omegapro/Google Drive/Arduino/hardware/MegaCore/avr/bootloaders/optiboot/atmega64/optiboot_atmega64_16MHz.hex:
avrdude: load data flash data from input file /Users/omegapro/Google Drive/Arduino/hardware/MegaCore/avr/bootloaders/optiboot/atmega64/optiboot_atmega64_16MHz.hex:
avrdude: input file /Users/omegapro/Google Drive/Arduino/hardware/MegaCore/avr/bootloaders/optiboot/atmega64/optiboot_atmega64_16MHz.hex contains 65536 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 0.00s

avrdude: verifying ...
avrdude: verification error, first mismatch at byte 0xfd00
         0xff != 0x84
avrdude: verification error; content mismatch

tnks

MCUdude, 2016-05-20:
Please try burning the 1MHz bootloader. Does this give the same error too?

CRI73, 2016-05-20:
This is it

avrdude: Version 6.0.1, compiled on Apr 14 2015 at 16:30:25
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2009 Joerg Wunsch

         System wide configuration file is "/Users/omegapro/Google Drive/Arduino/hardware/MegaCore/avr/avrdude.conf"
         User configuration file is "/Users/omegapro/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : usb
         Using Programmer              : usbasp
         AVR Part                      : ATmega64
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PA0
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom         4    20    64    0 no       2048    8      0  9000  9000 0xff 0xff
           flash         33     6   128    0 yes     65536  256    256  4500  4500 0xff 0xff
           lfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           lock           0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           calibration    0     0     0    0 no          4    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : usbasp
         Description     : USBasp, http://www.fischl.de/usbasp/

avrdude: auto set sck period (because given equals null)
avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.02s

avrdude: Device signature = 0x1e9602
avrdude: NOTE: "flash" memory has been specified, an erase cycle will be performed
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: auto set sck period (because given equals null)
avrdude: reading input file "/Users/omegapro/Google Drive/Arduino/hardware/MegaCore/avr/bootloaders/optiboot/atmega64/optiboot_atmega64_1MHz.hex"
avrdude: writing flash (65536 bytes):

Writing | ################################################## | 100% 0.00s

avrdude: 65536 bytes of flash written
avrdude: verifying flash memory against /Users/omegapro/Google Drive/Arduino/hardware/MegaCore/avr/bootloaders/optiboot/atmega64/optiboot_atmega64_1MHz.hex:
avrdude: load data flash data from input file /Users/omegapro/Google Drive/Arduino/hardware/MegaCore/avr/bootloaders/optiboot/atmega64/optiboot_atmega64_1MHz.hex:
avrdude: input file /Users/omegapro/Google Drive/Arduino/hardware/MegaCore/avr/bootloaders/optiboot/atmega64/optiboot_atmega64_1MHz.hex contains 65536 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 0.00s

avrdude: verifying ...
avrdude: verification error, first mismatch at byte 0xfd00
         0xff != 0x84
avrdude: verification error; content mismatch

CRI73, 2016-05-20:
... it seems the same one ...

MCUdude, 2016-05-20:
Hmm. This is weird. It works perfectly for me with a USBtinyISP programmer. You should probably try using another programmer (e.g Arduino as ISP)

CRI73, 2016-05-20:
Ok I'll try with Arduino as ISP... Than I tell you

CRI73, 2016-05-20:
... unfortunately nothing change with Arduino uno as ISP, but I've buy USBtinyISP on eBay... fingers crossed... :-D

MCUdude, 2016-05-20:
What hardware are you trying to program? I'm just flashing a small breakoutboard I've made.
[image]

I'm also using[ THIS USBtinyISP](http://www.ebay.com/itm/161947116048) board from Ebay. It's small, cheap and works like a charm :)

CRI73, 2016-05-20:
[image]

CRI73, 2016-05-20:
It's a custom made board... Some bypass capacitor & Quartz, nothing more, it's was a test... :-)

chamanirad, 2016-05-20:
That would be great. I am running at 16Mhz 
so I have
atmega128a
16mhz
usart1
Thanks again.

MCUdude, 2016-05-20:
This one is compiled with UART1. Just replace the 16MHz bootloader with this one!

[optiboot_atmega128_16MHz.hex.zip](https://github.com/MCUdude/MegaCore/files/275486/optiboot_atmega128_16MHz.hex.zip)

chamanirad, 2016-05-21:
Many thanks

closed 2016-06-03 by MCUdude

MCUdude, 2016-06-08:
@CRI73 would you pleas post your schematic (preferably in PDF)? there _might_ be a hardware error.