MCUdude/MegaCore #7: verification error; content mismatch

frankensteinXOXO, 2016-06-18:
Gives this error when uploading codes using arduino IDE with USBASP, and when PORTE is used in the code, the code will not upload at all. but It uploads fine using the hex file generated in arduino IDE directly via avrdude. 

`avrdude: verifying ...
avrdude: verification error, first mismatch at byte 0x0100
         0xff != 0x20
avrdude: verification error; content mismatch`

[bad.txt](https://github.com/MCUdude/MegaCore/files/321541/bad.txt)
[good.txt](https://github.com/MCUdude/MegaCore/files/321540/good.txt)

MCUdude, 2016-06-18:
are you using the latest version of MegaCore? Both codes uploads fine on my ATmega128 running at 16MHz.

Here is the verbose output:
``` bash
Build options changed, rebuilding all

/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-g++ -c -g -Os -Wall -Wextra -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega128 -DF_CPU=16000000L -DARDUINO=10605 -DARDUINO_AVR_ATmega128 -DARDUINO_ARCH_AVR -I/Users/Hans/Documents/Arduino/hardware/MegaCore 2561/avr/cores/MegaCore -I/Users/Hans/Documents/Arduino/hardware/MegaCore 2561/avr/variants/standard /var/folders/_p/s3kykxss20bbhfmqcg0ctrp80000gn/T/build7594825566520261610.tmp/sketch_jun18b.cpp -o /var/folders/_p/s3kykxss20bbhfmqcg0ctrp80000gn/T/build7594825566520261610.tmp/sketch_jun18b.cpp.o 
/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-gcc -c -g -x assembler-with-cpp -mmcu=atmega128 -DF_CPU=16000000L -DARDUINO=10605 -DARDUINO_AVR_ATmega128 -DARDUINO_ARCH_AVR -I/Users/Hans/Documents/Arduino/hardware/MegaCore 2561/avr/cores/MegaCore -I/Users/Hans/Documents/Arduino/hardware/MegaCore 2561/avr/variants/standard /Users/Hans/Documents/Arduino/hardware/MegaCore 2561/avr/cores/MegaCore/wiring_pulse.S -o /var/folders/_p/s3kykxss20bbhfmqcg0ctrp80000gn/T/build7594825566520261610.tmp/wiring_pulse.S.o 
/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-gcc -c -g -Os -Wall -Wextra -std=gnu11 -ffunction-sections -fdata-sections -MMD -mmcu=atmega128 -DF_CPU=16000000L -DARDUINO=10605 -DARDUINO_AVR_ATmega128 -DARDUINO_ARCH_AVR -I/Users/Hans/Documents/Arduino/hardware/MegaCore 2561/avr/cores/MegaCore -I/Users/Hans/Documents/Arduino/hardware/MegaCore 2561/avr/variants/standard /Users/Hans/Documents/Arduino/hardware/MegaCore 2561/avr/cores/MegaCore/hooks.c -o /var/folders/_p/s3kykxss20bbhfmqcg0ctrp80000gn/T/build7594825566520261610.tmp/hooks.c.o 
/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-gcc -c -g -Os -Wall -Wextra -std=gnu11 -ffunction-sections -fdata-sections -MMD -mmcu=atmega128 -DF_CPU=16000000L -DARDUINO=10605 -DARDUINO_AVR_ATmega128 -DARDUINO_ARCH_AVR -I/Users/Hans/Documents/Arduino/hardware/MegaCore 2561/avr/cores/MegaCore -I/Users/Hans/Documents/Arduino/hardware/MegaCore 2561/avr/variants/standard /Users/Hans/Documents/Arduino/hardware/MegaCore 2561/avr/cores/MegaCore/WInterrupts.c -o /var/folders/_p/s3kykxss20bbhfmqcg0ctrp80000gn/T/build7594825566520261610.tmp/WInterrupts.c.o 
/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-gcc -c -g -Os -Wall -Wextra -std=gnu11 -ffunction-sections -fdata-sections -MMD -mmcu=atmega128 -DF_CPU=16000000L -DARDUINO=10605 -DARDUINO_AVR_ATmega128 -DARDUINO_ARCH_AVR -I/Users/Hans/Documents/Arduino/hardware/MegaCore 2561/avr/cores/MegaCore -I/Users/Hans/Documents/Arduino/hardware/MegaCore 2561/avr/variants/standard /Users/Hans/Documents/Arduino/hardware/MegaCore 2561/avr/cores/MegaCore/wiring.c -o /var/folders/_p/s3kykxss20bbhfmqcg0ctrp80000gn/T/build7594825566520261610.tmp/wiring.c.o 
/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-gcc -c -g -Os -Wall -Wextra -std=gnu11 -ffunction-sections -fdata-sections -MMD -mmcu=atmega128 -DF_CPU=16000000L -DARDUINO=10605 -DARDUINO_AVR_ATmega128 -DARDUINO_ARCH_AVR -I/Users/Hans/Documents/Arduino/hardware/MegaCore 2561/avr/cores/MegaCore -I/Users/Hans/Documents/Arduino/hardware/MegaCore 2561/avr/variants/standard /Users/Hans/Documents/Arduino/hardware/MegaCore 2561/avr/cores/MegaCore/wiring_analog.c -o /var/folders/_p/s3kykxss20bbhfmqcg0ctrp80000gn/T/build7594825566520261610.tmp/wiring_analog.c.o 
/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-gcc -c -g -Os -Wall -Wextra -std=gnu11 -ffunction-sections -fdata-sections -MMD -mmcu=atmega128 -DF_CPU=16000000L -DARDUINO=10605 -DARDUINO_AVR_ATmega128 -DARDUINO_ARCH_AVR -I/Users/Hans/Documents/Arduino/hardware/MegaCore 2561/avr/cores/MegaCore -I/Users/Hans/Documents/Arduino/hardware/MegaCore 2561/avr/variants/standard /Users/Hans/Documents/Arduino/hardware/MegaCore 2561/avr/cores/MegaCore/wiring_digital.c -o /var/folders/_p/s3kykxss20bbhfmqcg0ctrp80000gn/T/build7594825566520261610.tmp/wiring_digital.c.o 
/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-gcc -c -g -Os -Wall -Wextra -std=gnu11 -ffunction-sections -fdata-sections -MMD -mmcu=atmega128 -DF_CPU=16000000L -DARDUINO=10605 -DARDUINO_AVR_ATmega128 -DARDUINO_ARCH_AVR -I/Users/Hans/Documents/Arduino/hardware/MegaCore 2561/avr/cores/MegaCore -I/Users/Hans/Documents/Arduino/hardware/MegaCore 2561/avr/variants/standard /Users/Hans/Documents/Arduino/hardware/MegaCore 2561/avr/cores/MegaCore/wiring_pulse.c -o /var/folders/_p/s3kykxss20bbhfmqcg0ctrp80000gn/T/build7594825566520261610.tmp/wiring_pulse.c.o 
/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-gcc -c -g -Os -Wall -Wextra -std=gnu11 -ffunction-sections -fdata-sections -MMD -mmcu=atmega128 -DF_CPU=16000000L -DARDUINO=10605 -DARDUINO_AVR_ATmega128 -DARDUINO_ARCH_AVR -I/Users/Hans/Documents/Arduino/hardware/MegaCore 2561/avr/cores/MegaCore -I/Users/Hans/Documents/Arduino/hardware/MegaCore 2561/avr/variants/standard /Users/Hans/Documents/Arduino/hardware/MegaCore 2561/avr/cores/MegaCore/wiring_shift.c -o /var/folders/_p/s3kykxss20bbhfmqcg0ctrp80000gn/T/build7594825566520261610.tmp/wiring_shift.c.o 
[... cut ...]
avrdude: verifying flash memory against /var/folders/_p/s3kykxss20bbhfmqcg0ctrp80000gn/T/build7594825566520261610.tmp/sketch_jun18b.cpp.hex:
avrdude: load data flash data from input file /var/folders/_p/s3kykxss20bbhfmqcg0ctrp80000gn/T/build7594825566520261610.tmp/sketch_jun18b.cpp.hex:
avrdude: input file /var/folders/_p/s3kykxss20bbhfmqcg0ctrp80000gn/T/build7594825566520261610.tmp/sketch_jun18b.cpp.hex contains 228 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 0.05s

avrdude: verifying ...
avrdude: 228 bytes of flash verified

avrdude done.  Thank you.

```

MCUdude, 2016-06-18:
I found the Issue! The good news is that there's nothing wrong with my core. The bad new is that there's nothing I can do about it.. 

You see, there is some kind of bug in AVRdude 6.0.1 (which Arduino IDE ships with) that gives this error. Running this code:
``` Arduino

void setup() {
  DDRA = 0xff;
  DDRB = 0xff;
  DDRC = 0xff;
  DDRD = 0xff;
  DDRE = 0xff;
  DDRF = 0xff;
}

void loop() {
  PORTA ^= 0xff;
  PORTB ^= 0xff;
  PORTC ^= 0xff;
  PORTD ^= 0xff;
  PORTE ^= 0xff;
  PORTF ^= 0xff;

  delay(1000);
}
```

and uploading with AVRdude 6.0.1 using this command (generated by Arduino IDE):
``` bash
/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avrdude -C/Users/Hans/Documents/Arduino/hardware/MegaCore/avr/avrdude.conf -v -patmega128 -cusbasp -Pusb -Uflash:w:/var/folders/_p/s3kykxss20bbhfmqcg0ctrp80000gn/T/build6448819071412050804.tmp/sketch_jun18a.cpp.hex:i 
```

gives this error:
``` bash
Reading | ################################################## | 100% 0.28s

avrdude: verifying ...
avrdude: verification error, first mismatch at byte 0x0100
         0xff != 0x0f
avrdude: verification error; content mismatch

avrdude done.  Thank you.
```

If I open Terminal (I'm using a mac by the way) and use the latest version of AVRdud, but still sticks with the same hex file and parameters:
``` bash
avrdude -C/Users/Hans/Documents/Arduino/hardware/MegaCore/avr/avrdude.conf -v -patmega128 -cusbasp -Pusb -Uflash:w:/var/folders/_p/s3kykxss20bbhfmqcg0ctrp80000gn/T/build6448819071412050804.tmp/sketch_jun18a.cpp.hex:i 
```

This is the output:
``` bash
Reading | ################################################## | 100% 0.21s

avrdude: verifying ...
avrdude: 720 bytes of flash verified

avrdude: safemode: hfuse reads as CE
avrdude: safemode: efuse reads as FF
avrdude: safemode: Fuses OK (E:FF, H:CE, L:BF)

avrdude done.  Thank you.
```

If you won't use the bootloader I guess you'll have to use a newer version of AVRdude

frankensteinXOXO, 2016-06-18:
okay, I see. thank you. Seems to me that this bug only happens when I use 0xff, the default value for all the memory.

closed 2016-07-05 by MCUdude