MCUdude/MegaCore #8: arduino 1.6.9 cpu:mega128 collect2.exe: error: ld returned 5 exit status

yonsight, 2016-06-19:
when select  atmega128 in arduino 1.6.9, return **collect2.exe: error: ld returned 5 exit status**
but atmega64 has no error.

MCUdude, 2016-06-19:
Can you post the full error message please? I'm using. Mac,  so I night have a hard time reproducing this error

yonsight, 2016-06-20:
Arduino:1.6.9 (Windows XP), 开发板："ATmega128, 16MHz external (BOD 2.7v)"

D:\Program Files\arduino\arduino-builder -dump-prefs -logger=machine 
-hardware "D:\Program Files\arduino\hardware" -hardware "C:\Documents 
and Settings\yong\Local Settings\Application Data\Arduino15\packages" 
-hardware "C:\Documents and Settings\yong\My Documents\Arduino\hardware" 
-tools "D:\Program Files\arduino\tools-builder" -tools "D:\Program 
Files\arduino\hardware\tools\avr" -tools "C:\Documents and 
Settings\yong\Local Settings\Application Data\Arduino15\packages" 
-built-in-libraries "D:\Program Files\arduino\libraries" -libraries 
"C:\Documents and Settings\yong\My Documents\Arduino\libraries" 
-fqbn=MegaCore:avr:128:clock=16MHz_external -ide-version=10609 
-build-path 
"C:\DOCUME~1\yong\LOCALS~1\Temp\build1717c675d58f482eb43b879989bee4a1.tmp" 
-warnings=none -prefs=build.warn_data_percentage=75 -verbose 
"C:\Documents and Settings\yong\My 
Documents\Arduino\sketch_jun16a\sketch_jun16a.ino"
D:\Program Files\arduino\arduino-builder -compile -logger=machine 
-hardware "D:\Program Files\arduino\hardware" -hardware "C:\Documents 
and Settings\yong\Local Settings\Application Data\Arduino15\packages" 
-hardware "C:\Documents and Settings\yong\My Documents\Arduino\hardware" 
-tools "D:\Program Files\arduino\tools-builder" -tools "D:\Program 
Files\arduino\hardware\tools\avr" -tools "C:\Documents and 
Settings\yong\Local Settings\Application Data\Arduino15\packages" 
-built-in-libraries "D:\Program Files\arduino\libraries" -libraries 
"C:\Documents and Settings\yong\My Documents\Arduino\libraries" 
-fqbn=MegaCore:avr:128:clock=16MHz_external -ide-version=10609 
-build-path 
"C:\DOCUME~1\yong\LOCALS~1\Temp\build1717c675d58f482eb43b879989bee4a1.tmp" 
-warnings=none -prefs=build.warn_data_percentage=75 -verbose 
"C:\Documents and Settings\yong\My 
Documents\Arduino\sketch_jun16a\sketch_jun16a.ino"
"D:\Program Files\arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w 
-std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections 
-fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=atmega128 
-DF_CPU=16000000L -DARDUINO=10609 -DARDUINO_AVR_ATmega128 
-DARDUINO_ARCH_AVR   "-IC:\Documents and Settings\yong\Local 
Settings\Application 
Data\Arduino15\packages\MegaCore\hardware\avr\1.0.1\cores\MegaCore" 
"-IC:\Documents and Settings\yong\Local Settings\Application 
Data\Arduino15\packages\MegaCore\hardware\avr\1.0.1\variants\standard" 
"C:\DOCUME~1\yong\LOCALS~1\Temp\build1717c675d58f482eb43b879989bee4a1.tmp\sketch\sketch_jun16a.ino.cpp" 
-o "nul"
"D:\Program Files\arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w 
-std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections 
-fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=atmega128 
-DF_CPU=16000000L -DARDUINO=10609 -DARDUINO_AVR_ATmega128 
-DARDUINO_ARCH_AVR   "-IC:\Documents and Settings\yong\Local 
Settings\Application 
Data\Arduino15\packages\MegaCore\hardware\avr\1.0.1\cores\MegaCore" 
"-IC:\Documents and Settings\yong\Local Settings\Application 
Data\Arduino15\packages\MegaCore\hardware\avr\1.0.1\variants\standard" 
"C:\DOCUME~1\yong\LOCALS~1\Temp\build1717c675d58f482eb43b879989bee4a1.tmp\sketch\sketch_jun16a.ino.cpp" 
-o "nul"
"D:\Program Files\arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w 
-std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections 
-fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=atmega128 
-DF_CPU=16000000L -DARDUINO=10609 -DARDUINO_AVR_ATmega128 
-DARDUINO_ARCH_AVR   "-IC:\Documents and Settings\yong\Local 
Settings\Application 
Data\Arduino15\packages\MegaCore\hardware\avr\1.0.1\cores\MegaCore" 
"-IC:\Documents and Settings\yong\Local Settings\Application 
Data\Arduino15\packages\MegaCore\hardware\avr\1.0.1\variants\standard" 
"C:\DOCUME~1\yong\LOCALS~1\Temp\build1717c675d58f482eb43b879989bee4a1.tmp\sketch\sketch_jun16a.ino.cpp" 
-o 
"C:\DOCUME~1\yong\LOCALS~1\Temp\build1717c675d58f482eb43b879989bee4a1.tmp\preproc\ctags_target_for_gcc_minus_e.cpp"
"D:\Program Files\arduino\tools-builder\ctags\5.8-arduino10/ctags" -u 
--language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns 
--line-directives 
"C:\DOCUME~1\yong\LOCALS~1\Temp\build1717c675d58f482eb43b879989bee4a1.tmp\preproc\ctags_target_for_gcc_minus_e.cpp"
[... 371 lines cut ...]
"C:\DOCUME~1\yong\LOCALS~1\Temp\build1717c675d58f482eb43b879989bee4a1.tmp\core\core.a" 
"C:\DOCUME~1\yong\LOCALS~1\Temp\build1717c675d58f482eb43b879989bee4a1.tmp\core\WMath.cpp.o"
"D:\Program Files\arduino\hardware\tools\avr/bin/avr-ar" rcs 
"C:\DOCUME~1\yong\LOCALS~1\Temp\build1717c675d58f482eb43b879989bee4a1.tmp\core\core.a" 
"C:\DOCUME~1\yong\LOCALS~1\Temp\build1717c675d58f482eb43b879989bee4a1.tmp\core\WString.cpp.o"
"D:\Program Files\arduino\hardware\tools\avr/bin/avr-ar" rcs 
"C:\DOCUME~1\yong\LOCALS~1\Temp\build1717c675d58f482eb43b879989bee4a1.tmp\core\core.a" 
"C:\DOCUME~1\yong\LOCALS~1\Temp\build1717c675d58f482eb43b879989bee4a1.tmp\core\abi.cpp.o"
"D:\Program Files\arduino\hardware\tools\avr/bin/avr-ar" rcs 
"C:\DOCUME~1\yong\LOCALS~1\Temp\build1717c675d58f482eb43b879989bee4a1.tmp\core\core.a" 
"C:\DOCUME~1\yong\LOCALS~1\Temp\build1717c675d58f482eb43b879989bee4a1.tmp\core\main.cpp.o"
"D:\Program Files\arduino\hardware\tools\avr/bin/avr-ar" rcs 
"C:\DOCUME~1\yong\LOCALS~1\Temp\build1717c675d58f482eb43b879989bee4a1.tmp\core\core.a" 
"C:\DOCUME~1\yong\LOCALS~1\Temp\build1717c675d58f482eb43b879989bee4a1.tmp\core\new.cpp.o"
"D:\Program Files\arduino\hardware\tools\avr/bin/avr-gcc" -w -Os 
-Wl,--gc-sections -mmcu=atmega128  -o 
"C:\DOCUME~1\yong\LOCALS~1\Temp\build1717c675d58f482eb43b879989bee4a1.tmp/sketch_jun16a.ino.elf" 
"C:\DOCUME~1\yong\LOCALS~1\Temp\build1717c675d58f482eb43b879989bee4a1.tmp\sketch\sketch_jun16a.ino.cpp.o" 
"C:\DOCUME~1\yong\LOCALS~1\Temp\build1717c675d58f482eb43b879989bee4a1.tmp/core\core.a" 
"-LC:\DOCUME~1\yong\LOCALS~1\Temp\build1717c675d58f482eb43b879989bee4a1.tmp" 
-lm
collect2.exe: error: ld returned 5 exit status

exit status 1
为开发板 ATmega128 编译时出错。

MCUdude, 2016-06-20:
It seems like an IDE related issue when using Windows XP:
[http://forum.arduino.cc/index.php?topic=310950.0](http://forum.arduino.cc/index.php?topic=310950.0)

closed 2016-06-25 by MCUdude