MCUdude/MegaCore #12: Compile problem

wothke, 2016-06-27:
I just installed Arduino 1.6.4 IDE (on WinXP) and MegaCore. I selected Board: "ATmega128", Clock: "16MHz external", Port: "COM1" and am using Programmer: "USBasp" (I already successfully installed the bootloader.).

When trying to compile the "BareMinimum" example the respective compile fails:

'ld.exe has encountered a problem and needs to close. ...'

last console message is:

'N:\Program Files\ArduinoWindows\arduino-1.6.4\hardware\tools\avr/bin/avr-ar rcs n:\Temp\build8835498005943340668.tmp/core.a n:\Temp\build8835498005943340668.tmp\WString.cpp.o 
N:\Program Files\ArduinoWindows\arduino-1.6.4\hardware\tools\avr/bin/avr-gcc -w -Os -Wl,--gc-sections -mmcu=atmega128 -o n:\Temp\build8835498005943340668.tmp/BareMinimum.cpp.elf n:\Temp\build8835498005943340668.tmp\BareMinimum.cpp.o n:\Temp\build8835498005943340668.tmp/core.a -Ln:\Temp\build8835498005943340668.tmp -lm 
collect2.exe: error: ld returned 5 exit status
Error compiling.'

However other examples like "AnalogReadSerial" do successfully compile. I don't know if the compile problem only affects the empty "BareMinimum" example or if the underlying root cause might also surface in other cases.

MCUdude, 2016-06-27:
Have a look at issue #8. It's an issue that only appears on Windows XP, and has nothing to do with my core. I suggest you replace the 15 years old OS 😉

wothke, 2016-06-27:
good to know.. in case it ever becomes a problem I'll just need to restart from one of the other boot partitions :-)

closed 2016-07-14 by MCUdude