MCUdude/MegaCore #13: digitalPinToInterrupt() issue

wothke, 2016-07-02:
see pins_arduino.h:
´#define digitalPinToInterrupt(p)   (((p) >= 4 && (p) <= 7) ? (p) : ((p) >= 18 && (p) <= 21) ? (p) - 18 : NOT_AN_INTERRUPT)´

shouldn't it rather be:

´#define digitalPinToInterrupt(p)   (((p) >= 3 && (p) <= 6) ? (p+1) : ((p) >= 18 && (p) <= 21) ? (p) - 18 : NOT_AN_INTERRUPT)´

wothke, 2016-07-02:
Also, when looking at the MegaCore specific SPI.cpp (see lines 84ff) implementation I wonder if the SPI_INT7_MASK might not always be setup properly (I did not investigate what the used defines exacly mean, so I don't know with of the 4 cases will actually be used for the ATmega128). 
I am trying to use INT7 to control a radio in my code and for some reason the radio seems to transmit but is does not seem to receive data - which may or may not be related to the above.

MCUdude, 2016-07-02:
> shouldn't it rather be:

#define digitalPinToInterrupt(p) (((p) >= 3 && (p) <= 6) ? (p+1) : ((p) >= 18 && (p) <= 21) ? (p) - 18 : NOT_AN_INTERRUPT)

Yes, that's a clear bug right there 😉 Will fix it soon.

> Also, when looking at the MegaCore specific SPI.cpp (see lines 84ff) implementation I wonder if the SPI_INT7_MASK might not always be setup properly (I did not investigate what the used defines exacly mean, so I don't know with of the 4 cases will actually be used for the ATmega128). 
I am trying to use INT7 to control a radio in my code and for some reason the radio seems to transmit but is does not seem to receive data - which may or may not be related to the above.

Have a look at [line 99](https://github.com/MCUdude/MegaCore/blob/cd0177a53c6022627889886e9ba1ccd2b66fa260/avr/libraries/SPI/SPI.cpp#L99). The ATmega64/128 got the EICRA, EICRB and EIMSK registers, so all these SPI interrupt masks should be defined in the following lines. Please keep me updated on this topic. The port of this library was an easy process, but I can't guarantee it's bug free

wothke, 2016-07-03:
I applied the above fix to my local configuration but unfortunately it does not seem to make and difference to my problem.

wothke, 2016-07-04:
it is the "standard pinout" graph that needs fixing.. the original digitalPinToInterrupt() impl is actually correct.

yonsight, 2016-07-07:
I changed to

#define digitalPinToInterrupt(p) (((p) >= 6 && (p) <= 7) ? (p) : ((p) >= 4 && (p) <= 5) ? (p - 4) : ((p) >= 18 && (p) <= 21) ? (p) - 16 : NOT_AN_INTERRUPT)

use this macro in externl interrupt.

arduino Pin    Interrupt    WInterrupt.c
4                INT4            case 0        (p-4)
5                INT5            case 1        (p-4)
6                INT6            case 6        (p)
7                INT7            case 7        (p)

18               INT0            case 2        (p-16)
19               INT1            case 3        (p-16)
20               INT2            case 4        (p-16)
21               INT3            case 5        (p-16)

在 2016-7-3 2:20, Hans 写道:

shouldn't it rather be:

#define digitalPinToInterrupt(p) (((p) >= 3 && (p) <= 6) ? (p+1) : ((p) >= 18 && (p) <= 21) ? (p) - 18 : NOT_AN_INTERRUPT)

Yes, that's a clear bug right there 😉 Will fix it soon.

Also, when looking at the MegaCore specific SPI.cpp (see lines 84ff) implementation I wonder if the SPI_INT7_MASK might not always be setup properly (I did not investigate what the used defines exacly mean, so I don't know with of the 4 cases will actually be used for the ATmega128).
I am trying to use INT7 to control a radio in my code and for some reason the radio seems to transmit but is does not seem to receive data - which may or may not be related to the above.

Have a look at line 99<https://github.com/MCUdude/MegaCore/blob/cd0177a53c6022627889886e9ba1ccd2b66fa260/avr/libraries/SPI/SPI.cpp#L99>. The ATmega64/128 got the EICRA, EICRB and EIMSK registers, so all these SPI interrupt masks should be defined in the following lines. Please keep me updated on this topic. The port of this library was an easy process, but I can't guarantee it's bug free

MCUdude, 2016-07-07:
Maybe it's best to just change the content of WInterrupts.h to:

``` c++
#elif defined(EICRA) && defined(EICRB) && defined(EIMSK)
    case 0:
      EICRA = (EICRA & ~((1 << ISC00) | (1 << ISC01))) | (mode << ISC00);
      EIMSK |= (1 << INT0);
      break;
    case 1:
      EICRA = (EICRA & ~((1 << ISC10) | (1 << ISC11))) | (mode << ISC10);
      EIMSK |= (1 << INT1);
      break;
    case 2:
      EICRA = (EICRA & ~((1 << ISC20) | (1 << ISC21))) | (mode << ISC20);
      EIMSK |= (1 << INT2);
      break;
    case 3:
      EICRA = (EICRA & ~((1 << ISC30) | (1 << ISC31))) | (mode << ISC30);
      EIMSK |= (1 << INT3);
      break;
    case 4:
      EICRB = (EICRB & ~((1 << ISC40) | (1 << ISC41))) | (mode << ISC40);
      EIMSK |= (1 << INT4);
      break;
    case 5:
      EICRB = (EICRB & ~((1 << ISC50) | (1 << ISC51))) | (mode << ISC50);
      EIMSK |= (1 << INT5);
      break;
    case 6:
      EICRB = (EICRB & ~((1 << ISC60) | (1 << ISC61))) | (mode << ISC60);
      EIMSK |= (1 << INT6);
      break;
    case 7:
      EICRB = (EICRB & ~((1 << ISC70) | (1 << ISC71))) | (mode << ISC70);
      EIMSK |= (1 << INT7);
      break;
```

and

``` c++
#elif defined(EICRA) && defined(EICRB) && defined(EIMSK)
    case 0:
      EIMSK &= ~(1 << INT0);
      break;
    case 1:
      EIMSK &= ~(1 << INT1);
      break;
    case 2:
      EIMSK &= ~(1 << INT2);
      break;
    case 3:
      EIMSK &= ~(1 << INT3);
      break;
    case 4:
      EIMSK &= ~(1 << INT4);
      break;
    case 5:
      EIMSK &= ~(1 << INT5);
      break;
    case 6:
      EIMSK &= ~(1 << INT6);
      break;
    case 7:
      EIMSK &= ~(1 << INT7);
      break;
```

Then there won't be any workarounds 😉

closed 2016-07-14 by MCUdude