MCUdude/MegaCore #14: SPI

wothke, 2016-07-04:
` 
#elif defined(EICRA) && defined(EICRB) && defined(EIMSK)
  #define SPI_INT0_MASK  (1<<INT4)
  #define SPI_INT1_MASK  (1<<INT5)
  #define SPI_INT2_MASK  (1<<INT0)
  #define SPI_INT3_MASK  (1<<INT1)
  #define SPI_INT4_MASK  (1<<INT2)
  #define SPI_INT5_MASK  (1<<INT3)
  #define SPI_INT6_MASK  (1<<INT6)
  #define SPI_INT7_MASK  (1<<INT7)
`

MCUdude, 2016-07-04:
Shouldn't thus work with the ATmega128?

wothke, 2016-07-04:
I still have the problem with my SPI based transceiver (it successfully sends but it does not seem to receive anything..- it works fine with my AT328 based board) I have no idea what might be the most likely root cause for a respective behavior.. I've double checked my wires and they all seem to be properly connected, i.e. SS, SCK, MOSI, MISO and INT7. I am just guessing that the problem might in some way be linked to the interrupt handling.. but maybe there is something else where an ATmega128 just reacts differenty as compared to an AT328?

wothke, 2016-07-04:
Example: On the 5V AT328 a 3.3V signal from the transceiver is still considered to be high.. is this maybe handled differently on the ATmega128?

MCUdude, 2016-07-04:
I'm sorry, your notation was all messed up. Try replacing it with:

``` c++
#elif defined(EICRA) && defined(EICRB) && defined(EIMSK)
  #define SPI_INT0_MASK  (1<<INT4)
  #define SPI_INT1_MASK  (1<<INT5)
  #define SPI_INT2_MASK  (1<<INT0)
  #define SPI_INT3_MASK  (1<<INT1)
  #define SPI_INT4_MASK  (1<<INT2)
  #define SPI_INT5_MASK  (1<<INT3)
  #define SPI_INT6_MASK  (1<<INT6)
  #define SPI_INT7_MASK  (1<<INT7)
```

I'm not able to test this my self right now, but I'd love to hear some feedback 😃

wothke, 2016-07-04:
it's only this wonderful edirtor that messed up the formatting of the code that I had copy/pasted.. the code in SPI.cpp looks exactly like the one you just pasted. However I am still curious why you are using that funny mapping: SPI_INT0_MASK using INT4, etc?

MCUdude, 2016-07-04:
Sorry, I forgot to do the actual changes 😆  I've edited the code in the previous post

wothke, 2016-07-04:
just tried it, but unfortunately it makes no difference to my problem..

wothke, 2016-07-04:
however I just measured the voltages on the SPI + IRQ wires to compare the idle-state of the old AT328 based device with the new ATmega128 based one (I've measured just after power-on and I would expect both device versions to bahave similarly at that point..).

there is one striking difference: on the old device (which still works fine) the nIRQ line (which is connected to INT0 on that device) shows 3.15V (which is the voltage that I am using for the transceiver), but on the new (ATmega128 based) device - where the nIRQ is connected to INT7 (to avoid I2C clash) the voltage is 0V.

Any ideas where that difference might come from?

wothke, 2016-07-04:
ARRRGH.. just had a look in the original specs: http://www.atmel.com/images/doc2467.pdf and from what I see your "ATmega64/128 standard pinout" graph is just wrong: INT4 is on PE4 (not PE3), etc. So your original "digitalPinToInterrupt" impl was actually correct. Its just that the PE6 pin that I am using is not INT7 but INT6!  I'll revert the earlier "fixes" and try what happens when I just use INT6 in my code..

MCUdude, 2016-07-05:
Ah, that's another bug that have sneaked in.. I've fixed the digitalPinToInterrupt macro and the pinout pic 😃 So you got the SPI library to work now? What interrupt masking did work out for you?

wothke, 2016-07-05:
I am using the original interrupt masking (see 1st post above) as well as your original digitalPinToInterrupt macro.. SPI seems to work now (transceiver is sending and receiving).

MCUdude, 2016-07-05:
Great! No need to do any changes in the SPI library then. Thanks again! 😃

closed 2016-07-05 by MCUdude

spa-sam, 2017-01-28:
Use the rfm69. Also a problem with interrupts. I have used pin4. Put the number of interrupts 4, but nothing transmits. What could be the problem?

wothke, 2017-01-31:
sorry, no idea..