MCUdude/MegaCore #15: Eclipse

wothke, 2016-07-10:
I just tried to use the MegaCore in http://eclipse.baeyens.it/ but it seems that either the respective Eclipse plugin just sucks or there is some incompatibility/problem with the MegaCore.

At first it seems as if the Eclipse/Arduino plugin correctly detects the MegaCore config. It presents the right processor selection and generates a respective "ATmega128" project. Unfortunately the generated "Includes" list is incomplete and it only contains the "core" and "variant" folders but none of the standard includes that SHOULD also be in that list.

Any ideas if this might rather be caused by the MegaCore config or by the Eclipse plugin?

MCUdude, 2016-07-10:
I have no experience with Eclipse, but I think that the Arduino IDE is a bit short when it comes to functionality. I've tried using PlatformIO for my projects, but low level settings such as bootloader burning and fuse settings is just painful. 

I decided to install Eclipse/Arduino on my mac, and the MegaCore works perfectly when adding the boards manager URL. Please post your error or a screenshot so I might be able to help you out.

Oh, while I'm at it; would mind posting a picture of your ATmega128 board? I'm just curious 😃 At the moment mine looks like this:
[image]

[image]

wothke, 2016-07-11:
first of all, here is what my board looks like.. [image]

I agree that the Arduino IDE isn't really an IDE and that's why I'd like to find something that is more than just notepad+.. that's how I came to try that Sloeber IDE.

Unfortunately I meanwhile have the impression that Sloeber also has quite a few quality issues..
The problem that I have mentioned above is this: When you create a new project for some of the built-in microcontrollers, then the IDE will automatically setup a usable environment where your sketch code can use the various stuff from Arduino.h. However when you create the new project based on MegaCore (e.g. for the ATmega128) then the generated environment lacks all the stuff that should be included via Arduino.h, e.g. if you wrote "uint8_t x=0;" then the IDE will mark uint8_t as a syntax error because it doesn't know about the required include location. I had to manually patch in these missing include folders (see below):

[image]

I think either it is a bug of the IDE that the automatic include folder generation doesn't work correctly for the add-on cores - or if that is something that is actually controlled from come MegaCore config file, then there might be an issue in there..

Another annoying thing is, that eventhough the project is specifically targeted to a specific core - i.e. ATmega128 -  the source code editor is not aware of the respective define __AVR_ATmega128__ and consequently the syntax/error highlighting is utterly useless.

MCUdude, 2016-07-11:
Hmm. Your settings window looks completely different than mine:
[image]

Mine also includes the correct core files. Are you running the latest version of ArduinoEclipse?
[image]

wothke, 2016-07-11:
I am using the stable win32 version of V3 from the download page (http://eclipse.baeyens.it/stable.php?OS=Windows).. the "core" files are correctly included (see 2 top entries) - but its the standard Arduino stuff that's not.. "Installation Details" gives me Version: 3.0.0.201603312133 for "Arduino Eclipse Extensions".

MCUdude, 2016-07-11:
Does regular arduino boards such as the Arduino UNO works?

wothke, 2016-07-11:
yes, that's exactly the point.. the projects generated for regular boards seem to be OK but not the ones for the add-on boards.

wothke, 2016-07-11:
when looking at the generated project settings then the most striking differences between a "standard" file and a "MegaCore" one is this:

MegaCore
`environment/project/it.baeyens.arduino.core.toolChain.release.1023913873/A.RUNTIME.PLATFORM.PATH/value=C\:/Documents and Settings/wothke/Arduino/hardware/MegaCore/hardware/avr/1.0.2`

standard
`environment/project/it.baeyens.arduino.core.toolChain.release.1023913873/A.RUNTIME.TOOLS.AVR-GCC-4.8.1-ARDUINO5.PATH/delimiter=;
environment/project/it.baeyens.arduino.core.toolChain.release.1023913873/A.RUNTIME.TOOLS.AVR-GCC-4.8.1-ARDUINO5.PATH/operation=replace
environment/project/it.baeyens.arduino.core.toolChain.release.1023913873/A.RUNTIME.TOOLS.AVR-GCC-4.8.1-ARDUINO5.PATH/value=N\:\\Program Files\\ArduinoWindows\\eclipseArduino\\arduinoPlugin\\tools\\arduino\\avr-gcc\\4.8.1-arduino5
environment/project/it.baeyens.arduino.core.toolChain.release.1023913873/A.RUNTIME.TOOLS.AVR-GCC.PATH/delimiter=;
environment/project/it.baeyens.arduino.core.toolChain.release.1023913873/A.RUNTIME.TOOLS.AVR-GCC.PATH/operation=replace
environment/project/it.baeyens.arduino.core.toolChain.release.1023913873/A.RUNTIME.TOOLS.AVR-GCC.PATH/value=N\:\\Program Files\\ArduinoWindows\\eclipseArduino\\arduinoPlugin\\tools\\arduino\\avr-gcc\\4.8.1-arduino5
environment/project/it.baeyens.arduino.core.toolChain.release.1023913873/A.RUNTIME.TOOLS.AVR-GCC4.8.1-ARDUINO5.PATH/delimiter=;
environment/project/it.baeyens.arduino.core.toolChain.release.1023913873/A.RUNTIME.TOOLS.AVR-GCC4.8.1-ARDUINO5.PATH/operation=replace
environment/project/it.baeyens.arduino.core.toolChain.release.1023913873/A.RUNTIME.TOOLS.AVR-GCC4.8.1-ARDUINO5.PATH/value=N\:\\Program Files\\ArduinoWindows\\eclipseArduino\\arduinoPlugin\\tools\\arduino\\avr-gcc\\4.8.1-arduino5
environment/project/it.baeyens.arduino.core.toolChain.release.1023913873/A.RUNTIME.TOOLS.AVRDUDE-6.0.1-ARDUINO5.PATH/delimiter=;
environment/project/it.baeyens.arduino.core.toolChain.release.1023913873/A.RUNTIME.TOOLS.AVRDUDE-6.0.1-ARDUINO5.PATH/operation=replace
environment/project/it.baeyens.arduino.core.toolChain.release.1023913873/A.RUNTIME.TOOLS.AVRDUDE-6.0.1-ARDUINO5.PATH/value=N\:\\Program Files\\ArduinoWindows\\eclipseArduino\\arduinoPlugin\\tools\\arduino\\avrdude\\6.0.1-arduino5
environment/project/it.baeyens.arduino.core.toolChain.release.1023913873/A.RUNTIME.TOOLS.AVRDUDE.PATH/delimiter=;
environment/project/it.baeyens.arduino.core.toolChain.release.1023913873/A.RUNTIME.TOOLS.AVRDUDE.PATH/operation=replace
environment/project/it.baeyens.arduino.core.toolChain.release.1023913873/A.RUNTIME.TOOLS.AVRDUDE.PATH/value=N\:\\Program Files\\ArduinoWindows\\eclipseArduino\\arduinoPlugin\\tools\\arduino\\avrdude\\6.0.1-arduino5
environment/project/it.baeyens.arduino.core.toolChain.release.1023913873/A.RUNTIME.TOOLS.AVRDUDE6.0.1-ARDUINO5.PATH/delimiter=;
environment/project/it.baeyens.arduino.core.toolChain.release.1023913873/A.RUNTIME.TOOLS.AVRDUDE6.0.1-ARDUINO5.PATH/operation=replace
environment/project/it.baeyens.arduino.core.toolChain.release.1023913873/A.RUNTIME.TOOLS.AVRDUDE6.0.1-ARDUINO5.PATH/value=N\:\\Program Files\\ArduinoWindows\\eclipseArduino\\arduinoPlugin\\tools\\arduino\\avrdude\\6.0.1-arduino5
`

wothke, 2016-07-11:
no idea what those different AVR versions might be causing... what do you think?

MCUdude, 2016-07-11:
I have no idea.. Have you tried with another computer?

wothke, 2016-07-11:
no, I only have this one copmuter.. btw: when originally installing MegaCore in the Arduino IDE I had used the https://mcudude.github.io/MegaCore/package_MCUdude_MegaCore_index.json approach.. but for Eclipse I had just copied the respective MegaCore folder from the Arduino IDE (see C:\...\Application Data\Arduino15\packages\) I would have thought that's equivalent to the "manual approach" described on your page - is isn't it?

MCUdude, 2016-07-11:
Try deleting the local folder and use the Jon link instead.  Worked for me!

wothke, 2016-07-11:
my mistake.. just deleted the folder and reinstalled using JSON. There still are some SPI related compile issued but it now looks much better.

MCUdude, 2016-07-11:
I'll draft a new release soon that fixes all the SPI errors. I think I'll add some info about Eclipse in the Readme file

wothke, 2016-07-11:
thanks for the support :-)

MCUdude, 2016-07-11:
A version is released! Please update to MegaCore v1.0.3 in ArduinoEclipse IDE

wothke, 2016-07-11:
new JSON URL?  maybe Eclipse is caching the old file.. but I still only see versions 1.0.0, 1.0.1 and 1.0.2

MCUdude, 2016-07-11:
Oops! Now v1.0.3 is available

wothke, 2016-07-11:
Installing Arduino Board Platforms
Failed to extract tar.bz2. N:/Program Files/ArduinoWindows/eclipseArduino/arduinoPlugin/downloads/MegaCore-1.0.3.tar.gz
Can't extract file N:\Program Files\ArduinoWindows\eclipseArduino\arduinoPlugin\packages\MegaCore\hardware\avr\1.0.3\avrdude.conf, file already exists!

wothke, 2016-07-11:
now I am getting these errors. I guess the problem are the duplicate SPI source files in the SPI and the SPI/src folders.
`"N:\Program` Files\ArduinoWindows\eclipseArduino\arduinoPlugin\tools\arduino\avr-gcc\4.8.1-arduino5/bin/avr-gcc" -Os -Wl,--gc-sections -mmcu=atmega128  -o "N:/Program Files/ArduinoWindows/eclipseWorkspace/RainMaker/Release/RainMaker.elf"    ./CoreEngine.cpp.o ./Display.cpp.o ./EepromStub.cpp.o ./PersistentState.cpp.o ./RainMaker.cpp.o ./ReceiptTracker.cpp.o ./SensorHub.cpp.o ./Transceiver.cpp.o ./ValveDriver.cpp.o  ./libraries/extEEPROM/extEEPROM.cpp.o  ./libraries/Wire/utility/twi.c.o  ./libraries/Wire/Wire.cpp.o  ./libraries/Time-master/Time.cpp.o  ./libraries/SPI/src/SPI.cpp.o  ./libraries/SPI/SPI.cpp.o  ./libraries/RadioHead_RF22/tools/simMain.cpp.o  ./libraries/RadioHead_RF22/STM32ArduinoCompat/HardwareSPI.cpp.o ./libraries/RadioHead_RF22/STM32ArduinoCompat/HardwareSerial.cpp.o ./libraries/RadioHead_RF22/STM32ArduinoCompat/wirish.cpp.o  ./libraries/RadioHead_RF22/RHutil/HardwareSerial.cpp.o ./libraries/RadioHead_RF22/RHutil/RasPi.cpp.o  ./libraries/RadioHead_RF22/RHCRC.cpp.o ./libraries/RadioHead_RF22/RHDatagram.cpp.o ./libraries/RadioHead_RF22/RHGenericDriver.cpp.o ./libraries/RadioHead_RF22/RHGenericSPI.cpp.o ./libraries/RadioHead_RF22/RHHardwareSPI.cpp.o ./libraries/RadioHead_RF22/RHSPIDriver.cpp.o ./libraries/RadioHead_RF22/RHSoftwareSPI.cpp.o ./libraries/RadioHead_RF22/RH_ASK.cpp.o ./libraries/RadioHead_RF22/RH_RF22.cpp.o ./libraries/RadioHead_RF22/RH_Serial.cpp.o  ./libraries/MyComm/BridgeMsg.cpp.o ./libraries/MyComm/CommUtils.cpp.o ./libraries/MyComm/Msg.cpp.o ./libraries/MyComm/RainMakerMsg.cpp.o ./libraries/MyComm/ReceiptMsg.cpp.o ./libraries/MyComm/SensorMsg.cpp.o  ./libraries/LiquidCrystal+/unused/LiquidCrystal_SR.cpp.o ./libraries/LiquidCrystal+/unused/LiquidCrystal_SR2W.cpp.o ./libraries/LiquidCrystal+/unused/LiquidCrystal_SR3W.cpp.o  ./libraries/LiquidCrystal+/FastIO.cpp.o ./libraries/LiquidCrystal+/I2CIO.cpp.o ./libraries/LiquidCrystal+/LCD.cpp.o ./libraries/LiquidCrystal+/LiquidCrystal.cpp.o ./libraries/LiquidCrystal+/LiquidCrystal_I2C.cpp.o  ./libraries/I2C_eeprom/I2C_eeprom.cpp.o  ./libraries/DS1307RTC/DS1307RTC.cpp.o  ./libraries/DHT/DHT.cpp.o   arduino.ar     "-LN:/Program Files/ArduinoWindows/eclipseWorkspace/RainMaker/Release" -lm
./libraries/SPI/SPI.cpp.o: In function `SPIClass::begin()':
N:/Program Files/ArduinoWindows/eclipseArduino/arduinoPlugin/packages/MegaCore/hardware/avr/1.0.3/libraries/SPI/SPI.cpp:27: multiple definition of `SPIClass::begin()'
./libraries/SPI/src/SPI.cpp.o:N:/Program Files/ArduinoWindows/eclipseArduino/arduinoPlugin/packages/MegaCore/hardware/avr/1.0.3/libraries/SPI/src/SPI.cpp:27: first defined here
./libraries/SPI/SPI.cpp.o: In function `SPIClass::begin()':
N:/Program Files/ArduinoWindows/eclipseArduino/arduinoPlugin/packages/MegaCore/hardware/avr/1.0.3/libraries/SPI/SPI.cpp:27: multiple definition of `SPIClass::initialized'
./libraries/SPI/src/SPI.cpp.o:N:/Program Files/ArduinoWindows/eclipseArduino/arduinoPlugin/packages/MegaCore/hardware/avr/1.0.3/libraries/SPI/src/SPI.cpp:27: first defined here
./libraries/SPI/SPI.cpp.o: In function `SPIClass::begin()':
N:/Program Files/ArduinoWindows/eclipseArduino/arduinoPlugin/packages/MegaCore/hardware/avr/1.0.3/libraries/SPI/SPI.cpp:27: multiple definition of `SPIClass::end()'
./libraries/SPI/src/SPI.cpp.o:N:/Program Files/ArduinoWindows/eclipseArduino/arduinoPlugin/packages/MegaCore/hardware/avr/1.0.3/libraries/SPI/src/SPI.cpp:27: first defined here
./libraries/SPI/SPI.cpp.o: In function `SPIClass::begin()':
N:/Program Files/ArduinoWindows/eclipseArduino/arduinoPlugin/packages/MegaCore/hardware/avr/1.0.3/libraries/SPI/SPI.cpp:27: multiple definition of `SPIClass::interruptMode'
makefile:98: recipe for target 'RainMaker.elf' failed
./libraries/SPI/src/SPI.cpp.o:N:/Program Files/ArduinoWindows/eclipseArduino/arduinoPlugin/packages/MegaCore/hardware/avr/1.0.3/libraries/SPI/src/SPI.cpp:27: first defined here
./libraries/SPI/SPI.cpp.o: In function `SPIClass::begin()':
N:/Program Files/ArduinoWindows/eclipseArduino/arduinoPlugin/packages/MegaCore/hardware/avr/1.0.3/libraries/SPI/SPI.cpp:27: multiple definition of `SPIClass::usingInterrupt(unsigned char)'
./libraries/SPI/src/SPI.cpp.o:N:/Program Files/ArduinoWindows/eclipseArduino/arduinoPlugin/packages/MegaCore/hardware/avr/1.0.3/libraries/SPI/src/SPI.cpp:27: first defined here
./libraries/SPI/SPI.cpp.o: In function `SPIClass::begin()':
N:/Program Files/ArduinoWindows/eclipseArduino/arduinoPlugin/packages/MegaCore/hardware/avr/1.0.3/libraries/SPI/SPI.cpp:27: multiple definition of `SPIClass::interruptMask'
./libraries/SPI/src/SPI.cpp.o:N:/Program Files/ArduinoWindows/eclipseArduino/arduinoPlugin/packages/MegaCore/hardware/avr/1.0.3/libraries/SPI/src/SPI.cpp:27: first defined here
./libraries/SPI/SPI.cpp.o: In function `SPIClass::begin()':
N:/Program Files/ArduinoWindows/eclipseArduino/arduinoPlugin/packages/MegaCore/hardware/avr/1.0.3/libraries/SPI/SPI.cpp:27: multiple definition of `SPIClass::notUsingInterrupt(unsigned char)'
./libraries/SPI/src/SPI.cpp.o:N:/Program Files/ArduinoWindows/eclipseArduino/arduinoPlugin/packages/MegaCore/hardware/avr/1.0.3/libraries/SPI/src/SPI.cpp:27: first defined here
./libraries/SPI/SPI.cpp.o: In function `SPIClass::begin()':
N:/Program Files/ArduinoWindows/eclipseArduino/arduinoPlugin/packages/MegaCore/hardware/avr/1.0.3/libraries/SPI/SPI.cpp:27: multiple definition of `SPIClass::interruptSave'
[... cut ...] I.cpp:27: first defined here
./libraries/SPI/SPI.cpp.o: In function `SPIClass::begin()':
N:/Program Files/ArduinoWindows/eclipseArduino/arduinoPlugin/packages/MegaCore/hardware/avr/1.0.3/libraries/SPI/SPI.cpp:27: multiple definition of `SPI'
./libraries/SPI/src/SPI.cpp.o:N:/Program Files/ArduinoWindows/eclipseArduino/arduinoPlugin/packages/MegaCore/hardware/avr/1.0.3/libraries/SPI/src/SPI.cpp:27: first defined here
collect2.exe: error: ld returned 1 exit status
make: *** [RainMaker.elf] Error 1`

MCUdude, 2016-07-11:
Really weird! The update went flawless here.. I uninstalled v1.0.2 and installed v1.0.3. I also needed to update my projects to the latest version, if not I get tons of errors about missing files

wothke, 2016-07-11:
after I deleted the SPI/src folder and removed respective references in the project the build completes successfully..

wothke, 2016-07-12:
I just installed the "latest" nightly version of the IDE from scratch and then installed the MegaCore 1,0.3. The SPI/src folder still needs to be deleted before the code builds..

wothke, 2016-07-14:
Another thing: when creating a new project for one of the "built-in" board types, Sloeber (Eclipse Arduino) provides a selection of the available "Upload Protocols" (e.g. AVRISP MKII, etc) - among others also the one that I need: "USBasp".

However, as soon as I switch (within that same dialog) to the boards.txt for the MegaCore, all the "Upload Protocol" options disappear and the only option left is "Default" (which doesn't work). 

Is there anything within the MegaCore configuration that might explain this behaviour or is this just a Sloeber bug with regard to the handling of non-built-in boards?

MCUdude, 2016-07-14:
MegaCore doesn't originally got its own set of programmers, because the built in ones (The ones that comes with the IDE) works fine. Does these programmers disappear if you try MightyCore (which have its own programmers?)

MCUdude, 2016-07-14:
Where in the Slobber IDE should the programmers and the burn bootloader option be?

wothke, 2016-07-14:
no idea whether or not it's possible to burn bootloader.. when using the "latest" Nightly build version the respective "Upload Protocol" dropdown is also shown in the "New Sketch" funtionality..

wothke, 2016-07-14:
and yes, for MightyCore there is a bunch of programmers listed in the respective dropdown - they are all marked as "(MightyCore)"...

MCUdude, 2016-07-14:
There's no problem adding extra programmers so that they will appear in Eclipse when using MegaCore. The problem is that this completely mess up Arduino IDE. Theres duplicates everywhere 😞 

[image]

wothke, 2016-07-14:
In case you don't want to mess up the ArduinoIDE just yet.. could you get me a copy of the MegaCore config that you used for the above sceenshot so that I can use it for Eclipse?

MCUdude, 2016-07-14:
Here's the programmers.txt file:
[programmers.txt](https://github.com/MCUdude/MegaCore/files/364798/programmers.txt)

The Arduino IDE way of handling the programmers is just terrible! I hope there will be a better solution soon. A solution may be to create an Eclipse branch, I'll need to think about that. I already got two other cores to keep up to date, and the only difference between this branch and the master is the programmers.txt file

wothke, 2016-07-14:
works like a charm.. thanks a lot! :-)

jantje, 2016-07-15:
I could support a localprogrammers.txt or something like that.
That file would be ignored by the arduinno ide.

MCUdude, 2016-07-15:
That would be great! How about externalprogrammers.txt?

Oh, while you're here; will there be possible to burn bootloader using Sloeber IDE in the future? That's the only missing feature left! 😃

jantje, 2016-07-15:
externalprogrammers.txt is fine for me.
Burning bootloaders? 
I just added the programmers and more people p*** me off with it than made me happy. But I agree that it is the only missing feature left! And now that I'm no longer pointing to the Arduino IDE the only reason to install the Arduino IDE next to sloeber.

MCUdude, 2016-07-15:
externalprogrammers.txt it is! I'll push the new files this evening.

<br/>
> I just added the programmers and more people p*** me off with it than made me happy.

I didn't quite get that; people are complaining instead of being happy about this feature?

jantje, 2016-07-15:
>I didn't quite get that; people are complaining instead of being happy about this feature?

yes, I did my first github block ever yesterday.

MCUdude, 2016-07-15:
> yes, I did my first github block ever yesterday.

I'm not picking sides, and there's definitely at least two sides of this case. From the owners side it's annoying to constantly get questions where <b>a)</b> the user didn't follow the instructions or read the documentation or <b>b)</b> it's not your issue. You're doing this voluntarily, and it's open source and free, so no one can or shouldn't expect anything. People will always report "false" issues and ask questions, but no one is forcing you to help or answer them. Have a look at AVRdudes issue list; it's long, most issues are unanswered but the latest version (6.3) was just released 😉 

From the users side there should be a place where questions can be asked. And there will be questions when documentation is poor. I'll happily donate money to an author if the software is well written, proper support is available and my projects are depending on it.

MCUdude, 2017-04-06:
I'm closing this issue as it's not really my job to make sure it works with Eclipse. My job is to make it work seamlessly with Arduino IDE, and jantjes job is to make sure seamless IDE adding work with Eclipse as well 😉

closed 2017-04-06 by MCUdude