MCUdude/MegaCore #17: delay() function

leventesen, 2016-07-21:
When I call delay(100000UL) function it waits only 1 second.
_delay_ms() function does not have problem.
This issue is not tested in real hardware enviroment, it occurs in Proteus simulation.

MCUdude, 2016-07-21:
Try with real hardware, and try removing the UL suffix. If the problem still occurs you should create an issue at [Arduino's repository](https://github.com/arduino/Arduino), as it's they code (i haven't done any modifications to the delay fiction)

leventesen, 2016-07-22:
I changed the implementation of delay(ms) function and for now it's working. I also created a topic in Arduino mailing group about this issue and waiting for approve. I will inform you about this.
Thanks.

closed 2016-07-29 by MCUdude