MCUdude/MegaCore #19: Verification Error on Burning Bootloader

delphiassistant, 2016-07-28:
Hi.
I'm getting following Error on burning bootloader using arduino ide:
`avrdude: verifying ...
avrdude: verification error, first mismatch at byte 0xfd00
         0xff != 0x09
avrdude: verification error; content mismatch`

I get this error on both Atmega64a and Atmega128a using any clock config and any programmer.

MCUdude, 2016-07-28:
Please turn on verbose upload in the IDE settings and post your full error message. What programmer are you using? This may be a lockbit issue, but I can't tell before the full error message is posted 😉

delphiassistant, 2016-07-29:
`C:\Arduino\hardware\tools\avr/bin/avrdude -CC:\Users\Mehdi\AppData\Local\Arduino15\packages\MegaCore\hardware\avr\1.0.3/avrdude.conf -v -patmega64 -cusbasp -Pusb -e -Ulock:w:0x3f:m -Uefuse:w:0xff:m -Uhfuse:w:0xc6:m -Ulfuse:w:0xbf:m 

avrdude: Version 6.0.1, compiled on Apr 15 2015 at 19:59:58
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2009 Joerg Wunsch

         System wide configuration file is "C:\Users\Mehdi\AppData\Local\Arduino15\packages\MegaCore\hardware\avr\1.0.3/avrdude.conf"

         Using Port                    : usb
         Using Programmer              : usbasp
         AVR Part                      : ATmega64
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PA0
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom         4    20    64    0 no       2048    8      0  9000  9000 0xff 0xff
           flash         33     6   128    0 yes     65536  256    256  4500  4500 0xff 0xff
           lfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           lock           0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           calibration    0     0     0    0 no          4    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : usbasp
         Description     : USBasp, http://www.fischl.de/usbasp/

avrdude: auto set sck period (because given equals null)
avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude: Device signature = 0x1e9602
avrdude: erasing chip
avrdude: auto set sck period (because given equals null)
avrdude: reading input file "0x3f"
avrdude: writing lock (1 bytes):

C:\Arduino\hardware\tools\avr/bin/avrdude -CC:\Users\Mehdi\AppData\Local\Arduino15\packages\MegaCore\hardware\avr\1.0.3/avrdude.conf -v -patmega64 -cusbasp -Pusb -Uflash:w:C:\Users\Mehdi\AppData\Local\Arduino15\packages\MegaCore\hardware\avr\1.0.3/bootloaders/optiboot_flash/atmega64/optiboot_flash_atmega64_115200_16MHz.hex:i -Ulock:w:0x0f:m 

avrdude: Version 6.0.1, compiled on Apr 15 2015 at 19:59:58
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2009 Joerg Wunsch

         System wide configuration file is "C:\Users\Mehdi\AppData\Local\Arduino15\packages\MegaCore\hardware\avr\1.0.3/avrdude.conf"

         Using Port                    : usb
         Using Programmer              : usbasp
Writing | ################################################## | 100% 0.00s

avrdude: 1 bytes of lock written
avrdude: verifying lock memory against 0x3f:
avrdude: load data lock data from input file 0x3f:
avrdude: input file 0x3f contains 1 bytes
avrdude: reading on-chip lock data:
[... 91 lines cut ...]
avrdude: NOTE: "flash" memory has been specified, an erase cycle will be performed
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: auto set sck period (because given equals null)
avrdude: reading input file "C:\Users\Mehdi\AppData\Local\Arduino15\packages\MegaCore\hardware\avr\1.0.3/bootloaders/optiboot_flash/atmega64/optiboot_flash_atmega64_115200_16MHz.hex"
avrdude: writing flash (65536 bytes):

Writing | ################################################## | 100% 0.00s

avrdude: 65536 bytes of flash written
avrdude: verifying flash memory against C:\Users\Mehdi\AppData\Local\Arduino15\packages\MegaCore\hardware\avr\1.0.3/bootloaders/optiboot_flash/atmega64/optiboot_flash_atmega64_115200_16MHz.hex:
avrdude: load data flash data from input file C:\Users\Mehdi\AppData\Local\Arduino15\packages\MegaCore\hardware\avr\1.0.3/bootloaders/optiboot_flash/atmega64/optiboot_flash_atmega64_115200_16MHz.hex:
avrdude: input file C:\Users\Mehdi\AppData\Local\Arduino15\packages\MegaCore\hardware\avr\1.0.3/bootloaders/optiboot_flash/atmega64/optiboot_flash_atmega64_115200_16MHz.hex contains 65536 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% -0.00s

avrdude: verifying ...
avrdude: verification error, first mismatch at byte 0xfd00
         0xff != 0x09
avrdude: verification error; content mismatch

avrdude done.  Thank you.

Error while burning bootloader.`

MCUdude, 2016-07-29:
Have you tried using an other programmer? I've experienced some issues with the USBasp.  The USBtinyISP works just fine

MCUdude, 2016-07-29:
Please try the 1MHz option also. Does this option return the same error?

delphiassistant, 2016-07-29:
Yes, I tried 1Mhz option, and returns same error.

MCUdude, 2016-07-29:
> Have you tried using an other programmer? I've experienced some issues with the USBasp. The USBtinyISP works just fine

MCUdude, 2016-07-29:
I'm actually able to reproduce this exact error using my own USBasp. Personally I prefer to use the [USBtinyISP (~3$)](http://www.ebay.com/itm/161947116048) instead. It's ALWAYS working, and I'll never experienced any issues with it. I've had several issues with the USBasp, and that's why I rarely use it when burning fuses, uploading bootloader or using AVRdude in general 😉

closed 2016-07-29 by MCUdude

delphiassistant, 2016-07-30:
**I have same error with USBTinyISP:**

C:\Arduino\hardware\tools\avr/bin/avrdude -CC:\Users\Mehdi.BARNAMENEVIS.000\AppData\Local\Arduino15\packages\MegaCore\hardware\avr\1.0.3/avrdude.conf -v -patmega128 -cusbtiny -e -Ulock:w:0x3f:m -Uefuse:w:0xff:m -Uhfuse:w:0xc6:m -Ulfuse:w:0xbf:m 

avrdude: Version 6.0.1, compiled on Apr 15 2015 at 19:59:58
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2009 Joerg Wunsch

         System wide configuration file is "C:\Users\Mehdi.BARNAMENEVIS.000\AppData\Local\Arduino15\packages\MegaCore\hardware\avr\1.0.3/avrdude.conf"

         Using Port                    : usb
         Using Programmer              : usbtiny
avrdude: usbdev_open(): Found USBtinyISP, bus:device: bus-0:\\.\libusb0-0001--0x1781-0x0c9f
         AVR Part                      : ATmega128
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PA0
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom         4    12    64    0 no       4096    8      0  9000  9000 0xff 0xff
           flash         33     6   128    0 yes    131072  256    512  4500  4500 0xff 0xff
           lfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           lock           0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           calibration    0     0     0    0 no          4    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : USBtiny
         Description     : USBtiny simple USB programmer, http://www.ladyada.net/make/usbtinyisp/
avrdude: programmer operation not supported

avrdude: Using SCK period of 10 usec
avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude: Device signature = 0x1e9702
avrdude: erasing chip
avrdude: Using SCK period of 10 usec
avrdude: reading input file "0x3f"
avrdude: writing lock (1 bytes):

C:\Arduino\hardware\tools\avr/bin/avrdude -CC:\Users\Mehdi.BARNAMENEVIS.000\AppData\Local\Arduino15\packages\MegaCore\hardware\avr\1.0.3/avrdude.conf -v -patmega128 -cusbtiny -Uflash:w:C:\Users\Mehdi.BARNAMENEVIS.000\AppData\Local\Arduino15\packages\MegaCore\hardware\avr\1.0.3/bootloaders/optiboot_flash/atmega128/optiboot_flash_atmega128_115200_16MHz.hex:i -Ulock:w:0x0f:m 

avrdude: Version 6.0.1, compiled on Apr 15 2015 at 19:59:58
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2009 Joerg Wunsch

         System wide configuration file is "C:\Users\Mehdi.BARNAMENEVIS.000\AppData\Local\Arduino15\packages\MegaCore\hardware\avr\1.0.3/avrdude.conf"

         Using Port                    : usb
         Using Programmer              : usbtiny
avrdude: usbdev_open(): Found USBtinyISP, bus:device: bus-0:\\.\libusb0-0001--0x1781-0x0c9f
         AVR Part                      : ATmega128
         Chip Erase delay              : 9000 us
[... 97 lines cut ...]
avrdude: NOTE: "flash" memory has been specified, an erase cycle will be performed
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: Using SCK period of 10 usec
avrdude: reading input file "C:\Users\Mehdi.BARNAMENEVIS.000\AppData\Local\Arduino15\packages\MegaCore\hardware\avr\1.0.3/bootloaders/optiboot_flash/atmega128/optiboot_flash_atmega128_115200_16MHz.hex"
avrdude: writing flash (131072 bytes):

Writing | ################################################## | 100% 0.00s

avrdude: 131072 bytes of flash written
avrdude: verifying flash memory against C:\Users\Mehdi.BARNAMENEVIS.000\AppData\Local\Arduino15\packages\MegaCore\hardware\avr\1.0.3/bootloaders/optiboot_flash/atmega128/optiboot_flash_atmega128_115200_16MHz.hex:
avrdude: load data flash data from input file C:\Users\Mehdi.BARNAMENEVIS.000\AppData\Local\Arduino15\packages\MegaCore\hardware\avr\1.0.3/bootloaders/optiboot_flash/atmega128/optiboot_flash_atmega128_115200_16MHz.hex:
avrdude: input file C:\Users\Mehdi.BARNAMENEVIS.000\AppData\Local\Arduino15\packages\MegaCore\hardware\avr\1.0.3/bootloaders/optiboot_flash/atmega128/optiboot_flash_atmega128_115200_16MHz.hex contains 131072 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 0.00s

avrdude: verifying ...
avrdude: verification error, first mismatch at byte 0x1fc00
         0xff != 0x01
avrdude: verification error; content mismatch

avrdude done.  Thank you.

Error while burning bootloader.

delphiassistant, 2016-07-30:
And here is schematic of my board, if it helps.
[image]

MCUdude, 2016-07-30:
Yes, the USBtinyISP doesn't officially support chips larger than 64kB. I suggested that one because you used the ATmega64 in your previous post. I get the exact same error as you with the USBtinyISP and the ATmega128, but the bootloader is still working. If you attach an LED to PB5 you'll see that it's flashing twice every seconds or so, indicating that the bootloader is running

javierpelaez, 2016-08-11:
I have exactly the same error with an Atmega644PA-AU programming with a USBASP. The strange thing is I have tried to load the bootloader in other unit of the same microcontroller and there is no problem with it. So, it could be a problem with the microcontroller or with something in the board.
In the board which there is no problem I have an external through hole 8MHz crystal with two capacitors, in the board which brings the error I have a Murata SMD crystal (whith internal capacitors). It does not seem to be much more differences.
We have tried several times with exactly the same results, always perfect with the second board and always the same error in the first board.

JWahaus, 2019-01-17:
I'm having the same problem with my USBtinyISP.  It worked great for about a year and then one day it started giving verification errors on every board that I own.  I tried a new ICSP cable, a different PC, several Arduino boards and it consistently corrupts flash (verification errors!).   The board seems to work fine as far as my PC is concerned.  I've ordered another USBtinyISP, hopefully this next one will last longer than the first.

zozo6015, 2020-03-10:
I just installed the bootloader with the USBtinyISP. It gave me the error but then uploading the firmware via USB worked. That mean it has installed the bootloader properly.