MCUdude/MegaCore #21: Facing problems in uploading bootloader to Atmega128A

shaktidhar, 2016-09-25:
{{ I already opened a thread at https://forum.arduino.cc/index.php?topic=425844.0 but then realized this might be the better place...  }}  

Hi, 
I followed the exact circuit as noted [https://camo.githubusercontent.com/b6046ccf5b261d6accdf030c2158ddb119f898d8/687474703a2f2f692e696d6775722e636f6d2f5549696e6f38492e706e67](url) 

But I get errors (or simply time out) while uploading. 
I tried uploading the bootloader using a genuine Arduino Uno as SPI programmer. And just to experiment, I tried uploading bootloader using a USB to TTL serial connection. 

No luck so far. 

Is there any specific programmer that is known to work? There are conflicting notes on USBTiny. Seems it is better than USBasp but only works for Atmega64... Or I am just reading too much and not putting things in correct order? 

Any inputs will be greatly appreciated. 

Thanks.

MCUdude, 2016-09-25:
I have no issues burning bootloader using the latest Arduino IDE 1.6.12, so it's not a bug in the core. I need to see a picture of your actual setup to be able to figure out what might be wrong.

You also need to turn on verbose output to get the full error information. Open Arduino IDE, goto settings and turn on all verbose option. Then post the error message from avrdude when you're burning the bootloader

shaktidhar, 2016-09-25:
I am using the basic circuit : 
https://camo.githubusercontent.com/b6046ccf5b261d6accdf030c2158ddb119f898d8/687474703a2f2f692e696d6775722e636f6d2f5549696e6f38492e706e67 

I am sure there is no bug in the core, but apparently I should be using the correct programmer. Using an Arduino is not working for me. What else could I try to make it work? 
This is the output when uploading : 
`
Arduino: 1.6.9 (Mac OS X), Board: "ATmega128, 16MHz external (BOD 2.7v)"

/Users/macowner/Library/Arduino15/packages/arduino/tools/avrdude/6.3.0-arduino6/bin/avrdude -C/Users/macowner/Library/Arduino15/packages/MegaCore/hardware/avr/1.0.3/avrdude.conf -v -patmega128 -cstk500v1 -P/dev/cu.usbmodem1411 -b19200 -e -Ulock:w:0x3f:m -Uefuse:w:0xff:m -Uhfuse:w:0xc6:m -Ulfuse:w:0xbf:m

avrdude: Version 6.3, compiled on Sep 12 2016 at 17:22:25
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "/Users/macowner/Library/Arduino15/packages/MegaCore/hardware/avr/1.0.3/avrdude.conf"
         User configuration file is "/Users/macowner/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : /dev/cu.usbmodem1411
         Using Programmer              : stk500v1
         Overriding Baud Rate          : 19200
avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x15
         AVR Part                      : ATmega128
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PA0
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom         4    12    64    0 no       4096    8      0  9000  9000 0xff 0xff
           flash         33     6   128    0 yes    131072  256    512  4500  4500 0xff 0xff
           lfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           lock           0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           calibration    0     0     0    0 no          4    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : STK500
         Description     : Atmel STK500 Version 1.x firmware

avrdude: stk500_getparm(): (a) protocol error, expect=0x14, resp=0x14

avrdude: stk500_getparm(): (a) protocol error, expect=0x14, resp=0x02

avrdude: stk500_getparm(): (a) protocol error, expect=0x14, resp=0x10
         Hardware Version: -1559310084
         Firmware Version: -1073745416.440209
         Topcard         : STK502
         Vtarget         : 1.8 V
         Varef           : 0.0 V
         Oscillator      : Off
         SCK period      : 0.1 us

avrdude: stk500_initialize(): (b) protocol error, expect=0x10, resp=0x01
avrdude: initialization failed, rc=-1
         Double check connections and try again, or use -F to override
         this check.

avrdude: stk500_disable(): protocol error, expect=0x14, resp=0x10

avrdude done.  Thank you.

Problem uploading to board.  See http://www.arduino.cc/en/Guide/Troubleshooting#upload for suggestions.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

`

MCUdude, 2016-09-25:
I don't have very good experience with the Arduino as ISP programmer, and prefer to use the USBtinyISP, which _always_ works. It seems like an issue with your programmer, so you should try another programmer.

The reason why I'm asking for a picture of your sets is because people tends to screw up without even knowing. It's impossible for me to spot HW errors without a picture

shaktidhar, 2016-09-25:
I don't have USBtinyISP at hand. I read two different notes on it (one that suggested this works more reliably compared to Usbasp, and other that said USBTinyISP has been tested only for Atmega64 ). So, I didn't get one. 
I will order one now and will update on the progress. 

I will upload a hardware picture too. My soldering skills are a little embarrassing so showing off is not my instinctive action. I have a little suspicion on the crystal as well so will try replacing it and see if that can help.

MCUdude, 2016-09-25:
You may try to burn the bootloader with the 1 MHz internal oscillator instead. If there's an issue with the crystal, that should solve it. If you want to change the clock frequency back to 16 MHz again, you just re-burn the bootloader

shaktidhar, 2016-09-25:
Tried. 

`/Users/macowner/Library/Arduino15/packages/arduino/tools/avrdude/6.3.0-arduino6/bin/avrdude -C/Users/macowner/Library/Arduino15/packages/MegaCore/hardware/avr/1.0.3/avrdude.conf -v -patmega128 -cstk500v1 -P/dev/cu.usbmodem1411 -b19200 -e -Ulock:w:0x3f:m -Uefuse:w:0xff:m -Uhfuse:w:0xd6:m -Ulfuse:w:0xa1:m 

avrdude: Version 6.3, compiled on Sep 12 2016 at 17:22:25
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "/Users/macowner/Library/Arduino15/packages/MegaCore/hardware/avr/1.0.3/avrdude.conf"
         User configuration file is "/Users/macowner/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : /dev/cu.usbmodem1411
         Using Programmer              : stk500v1
         Overriding Baud Rate          : 19200
avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x15
         AVR Part                      : ATmega128
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PA0
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom         4    12    64    0 no       4096    8      0  9000  9000 0xff 0xff
           flash         33     6   128    0 yes    131072  256    512  4500  4500 0xff 0xff
           lfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           lock           0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           calibration    0     0     0    0 no          4    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : STK500
         Description     : Atmel STK500 Version 1.x firmware

avrdude: stk500_getparm(): (a) protocol error, expect=0x14, resp=0x14

avrdude: stk500_getparm(): (a) protocol error, expect=0x14, resp=0x02

avrdude: stk500_getparm(): (a) protocol error, expect=0x14, resp=0x10
         Hardware Version: -1559310084
         Firmware Version: -1073745416.440209
         Topcard         : STK502
         Vtarget         : 1.8 V
         Varef           : 0.0 V
         Oscillator      : Off
         SCK period      : 0.1 us

avrdude: stk500_initialize(): (b) protocol error, expect=0x10, resp=0x01
avrdude: initialization failed, rc=-1
         Double check connections and try again, or use -F to override
         this check.

avrdude: stk500_disable(): protocol error, expect=0x14, resp=0x10

avrdude done.  Thank you.

Problem uploading to board.  See http://www.arduino.cc/en/Guide/Troubleshooting#upload for suggestions.` 

I guess that leaves me with no other choice but get UsbTinyISP.

MCUdude, 2016-09-25:
There's a forum post discussing the exact problem with Arduino as ISP as you have:
[https://forum.arduino.cc/index.php?topic=354674.0](https://forum.arduino.cc/index.php?topic=354674.0)

If you're using an Arduino UNO; have you connected a 10 uF capacitor between the reset pin and ground (on the UNO)?

shaktidhar, 2016-09-25:
Thanks. 
Tried with capacitor connected. 
Can't see any difference. 

USBTinyISP ordered.  

One query. The led on the circuit,  connected to pin 15, is it supposed to be on when connected? (Mine is not, I am trying to rule out any of my typical mistakes.)

MCUdude, 2016-09-25:
The LED is connected to pin PB5, and it should flash twice every second after the bootloader have been burnt. It isn't strictly necessary, but I've added it to the minimal setup schematic to help users debug potensial issues like this

shaktidhar, 2016-09-25:
:-) Ok. 
That means till the bootloader is up, I can presume it's ok that it doesn't blink...

yo8tww, 2017-02-09:
Hi..I have the same issue with the USBtinyISP v3.0  I had just bought...I can't burn the bootloader on 128A. With another programmer ( AVRISP-MKII) and Arduino IDE 1.06 and chip-45 bootloader, it works but with higher IDE versions, with Mega core I can't burn it...
I have this error :

avrdude: initialization failed, rc=-1
Double check connections and try again, or use -F to override
this check

The same error appears when I'm using the Pololu USBavrISP v3, with only this 128-64 core. The rest of the avr's I can burn them whitout a problem..

MCUdude, 2017-02-09:
Please turn on verbose upload so I can see the entire output from Avrdude.

yo8tww, 2017-02-09:
Hi again..This is what I get on screen
C:\arduino-1.6.11\hardware\tools\avr/bin/avrdude -CC:\Users\Costin\AppData\Local\Arduino15\packages\MegaCore\hardware\avr\1.0.5/avrdude.conf -v -patmega128 -cusbtiny -e -Ulock:w:0x3f:m -Uefuse:w:0xff:m -Uhfuse:w:0xc6:m -Ulfuse:w:0b10111111:m 

avrdude: Version 6.0.1, compiled on Apr 15 2015 at 19:59:58
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2009 Joerg Wunsch

         System wide configuration file is "C:\Users\Costin\AppData\Local\Arduino15\packages\MegaCore\hardware\avr\1.0.5/avrdude.conf"

         Using Port                    : usb
         Using Programmer              : usbtiny
avrdude: usbdev_open(): Found USBtinyISP, bus:device: bus-0:\\.\libusb0-0001--0x1781-0x0c9f
         AVR Part                      : ATmega128
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PA0
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom         4    12    64    0 no       4096    8      0  9000  9000 0xff 0xff
           flash         33     6   128    0 yes    131072  256    512  4500  4500 0xff 0xff
           lfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           lock           0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           calibration    0     0     0    0 no          4    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : USBtiny
         Description     : USBtiny simple USB programmer, http://www.ladyada.net/make/usbtinyisp/
avrdude: programmer operation not supported

avrdude: Using SCK period of 10 usec
avrdude: initialization failed, rc=-1
         Double check connections and try again, or use -F to override
         this check.

avrdude done.  Thank you.

Error while burning bootloader.

yo8tww, 2017-02-09:
Hi..just discovered another thing....The Atmega that don't initialize it's an Atmega128A AU...This can be written only with avrispmkII. Why ? I don't know...On an other board I have two avr chips, Atmega128 16AU 
which can be bootloaded, programmed very well with the USBtinyISP...Weird, isn't it ?

shaktidhar, 2017-02-10:
I am getting similar errors. I had kind of given up on this. But if it helps. 

`/Users/mac1/Library/Arduino15/packages/arduino/tools/avrdude/6.3.0-arduino8/bin/avrdude -C/Users/mac1/Library/Arduino15/packages/MegaCore/hardware/avr/1.0.5/avrdude.conf -v -patmega128 -cusbtiny -e -Ulock:w:0x3f:m -Uefuse:w:0xff:m -Uhfuse:w:0xc6:m -Ulfuse:w:0b10111111:m 

avrdude: Version 6.3, compiled on Dec 16 2016 at 13:32:25
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "/Users/mac1/Library/Arduino15/packages/MegaCore/hardware/avr/1.0.5/avrdude.conf"
         User configuration file is "/Users/mac1/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : usb
         Using Programmer              : usbtiny
avrdude: usbdev_open(): Found USBtinyISP, bus:device: 020:008
         AVR Part                      : ATmega128
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PA0
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom         4    12    64    0 no       4096    8      0  9000  9000 0xff 0xff
           flash         33     6   128    0 yes    131072  256    512  4500  4500 0xff 0xff
           lfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           lock           0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           calibration    0     0     0    0 no          4    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : USBtiny
         Description     : USBtiny simple USB programmer, http://www.ladyada.net/make/usbtinyisp/
avrdude: programmer operation not supported

avrdude: Using SCK period of 10 usec
avrdude: initialization failed, rc=-1
         Double check connections and try again, or use -F to override
         this check.

avrdude done.  Thank you.

Error while burning bootloader.`

yo8tww, 2017-02-11:
Hi again..Well, it seems that usbtiny is very sensitive to loads on the PE0,PE1 pins and the avrispmkII seems to be more "tanky".Put a 1k resistor in series with PE0, solved the problem with avr device initialisation..Everything work fine now, despite those "content mismatch" & "programmer unsuported" messages. Thanks Hans for all this core...

shaktidhar, 2017-02-11:
Ok, so I also added the 1K resistor. 
And seems it did proceed further after that. 

`/Users/mac1/Library/Arduino15/packages/arduino/tools/avrdude/6.3.0-arduino8/bin/avrdude -C/Users/mac1/Library/Arduino15/packages/MegaCore/hardware/avr/1.0.5/avrdude.conf -v -patmega128 -cusbtiny -e -Ulock:w:0x3f:m -Uefuse:w:0xff:m -Uhfuse:w:0xc6:m -Ulfuse:w:0b10111111:m 

avrdude: Version 6.3, compiled on Dec 16 2016 at 13:32:25
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "/Users/mac1/Library/Arduino15/packages/MegaCore/hardware/avr/1.0.5/avrdude.conf"
         User configuration file is "/Users/mac1/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : usb
         Using Programmer              : usbtiny
avrdude: usbdev_open(): Found USBtinyISP, bus:device: 020:010
         AVR Part                      : ATmega128
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PA0
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom         4    12    64    0 no       4096    8      0  9000  9000 0xff 0xff
           flash         33     6   128    0 yes    131072  256    512  4500  4500 0xff 0xff
           lfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           lock           0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           calibration    0     0     0    0 no          4    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : USBtiny
         Description     : USBtiny simple USB programmer, http://www.ladyada.net/make/usbtinyisp/
avrdude: programmer operation not supported

avrdude: Using SCK period of 10 usec
avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e9702 (probably m128)
avrdude: erasing chip
avrdude: Using SCK period of 10 usec
avrdude: reading input file "0x3f"
avrdude: writing lock (1 bytes):

Writing | ################################################## | 100% 0.00s

avrdude: 1 bytes of lock written
avrdude: verifying lock memory against 0x3f:
avrdude: load data lock data from input file 0x3f:
avrdude: input file 0x3f contains 1 bytes
avrdude: reading on-chip lock data:

Reading | ################################################## | 100% 0.00s

[... 105 lines cut ...]
avrdude: Using SCK period of 10 usec
avrdude: reading input file "/Users/mac1/Library/Arduino15/packages/MegaCore/hardware/avr/1.0.5/bootloaders/optiboot_flash/atmega128/optiboot_flash_atmega128_115200_16000000L.hex"
avrdude: writing flash (131072 bytes):

Writing | ################################################## | 100% 0.00s

avrdude: 131072 bytes of flash written
avrdude: verifying flash memory against /Users/mac1/Library/Arduino15/packages/MegaCore/hardware/avr/1.0.5/bootloaders/optiboot_flash/atmega128/optiboot_flash_atmega128_115200_16000000L.hex:
avrdude: load data flash data from input file /Users/mac1/Library/Arduino15/packages/MegaCore/hardware/avr/1.0.5/bootloaders/optiboot_flash/atmega128/optiboot_flash_atmega128_115200_16000000L.hex:
avrdude: input file /Users/mac1/Library/Arduino15/packages/MegaCore/hardware/avr/1.0.5/bootloaders/optiboot_flash/atmega128/optiboot_flash_atmega128_115200_16000000L.hex contains 131072 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 0.00s

avrdude: verifying ...
avrdude: verification error, first mismatch at byte 0x1fc00
         0xff != 0x01
avrdude: verification error; content mismatch

avrdude done.  Thank you.

Error while burning bootloader.` 

It throws an error about burning bootloader. 
Is there a quick way to validate whether it went through or not ?

MCUdude, 2017-02-11:
Woah, thats great! the last error (*avrdude: verification error; content mismatch*) is a  USBtinyISP related, and is nothing to worry about. The USBtinyISP doesn't _actually_ support devices with more than 64kB, but works fine on the ATmega128 despite the error.

Your ATmega128 should now have a boatloader, and the LED connected to PB5 should flash twice every second or so.

shaktidhar, 2017-02-11:
One more success story here. The led is happy blinking... (It seems I did something early which led to the led getting burnt out.) 
I noticed there is an instruction for adding 1K resistor. 
Happy day and happy end to the week. 

More power to you Hans.

closed 2017-02-11 by shaktidhar

yo8tww, 2017-02-11:
I can confirm the bootloader on 128A is ok, and it can be programmed succesfully via an FTDI   ( in my case, an FTDI 232RL onboard mounted).Have a nice weekend, mates..

ntsang-lab, 2020-07-31:
hi MCUdude
I have problems uploading the bootloader for atmega128A.
can you help me
How can I contact you
Give me your email or Facebook address.
Please help me.
My email address: Ntsangbg98@gmail.com
Thank you