MCUdude/MegaCore #27: Bootloader boot time faster than expected

hackscribble, 2017-01-03:
This is more a question than a problem.

I previously used MightyCore with Atmega324 and 16MHz external oscillator.  After burning bootloader, LED flashed twice per second as per readme.  Time taken by bootloader between reset and start of program execution was 1.5s (consistent with an Arduino).

Now I am using MegaCore with Atmega64A and 16MHz external oscillator.  After burning the bootloader, I saw the LED flash much faster and (I think) it was not regular.  Bootloader works OK, programs loaded via serial run OK and millis() based timing is correct.

However, I have measured how long the bootloader takes between reset and start of program.  It is much shorter, under 240ms.

Is this an expected difference between MegaCore and MightCore?

Thanks

MCUdude, 2017-01-03:
I've actually never thought of this. The first releases of MegaCore I used another port of Optiboot, without the "Write to flash" ability. If you want to test the other version of Optiboot, try installing v1.0.1. To be honest I don't know why it faster to start up. As far as I remember the port was a fairly small job. I'm a bit busy at the moment, but if you want to dig deeper into this you can always use a text editor or Github to compare the MegaCore and the MightyCore bootloader!

Quick question. Where did you first hear of MegaCore? Was it Google, or the forums? Just curious 😃

hackscribble, 2017-01-03:
Thanks for the quick reply.  I'll compare the two bootloaders.

When I ran out of GPIO on the 324, I looked at the Atmel family, spotted the 64 and googled for it and Arduino.  That took me to this page:  http://playground.arduino.cc/Main/ArduinoOnOtherAtmelChips

MCUdude, 2017-02-11:
Any updates on this? Maybe our bootloader guru @majekw knows whats going on? 😉

majekw, 2017-02-12:
My version of Optiboot have enabled 'fast start' feature. It means that bootloader checks for reset cause and if external reset is triggered (by manually pressing reset button or by line change by serial programmer), it waits 2 seconds for code upload. But in almost all other cases (for example normal power on) it skips all bootloader code and instantly starts application.
So, in most cases booting should be instant, without any delay. But you are still able to enter bootloader using external reset.

More info about this feature you could find here: https://github.com/Optiboot/optiboot/issues/140#issuecomment-107694603

Just curious @hackscribble : how do you measure this 240ms? It's from power-up, from applying reset? Now precise is your measure? I ask because I can't assign this time to anything during bootup. In 'fast mode', there is startup time set by fuses (SUT) which should be about 65ms (by default, @MCUdude: does it also apply to your fuses for Atmega64?) and almost nothing more (only few instructions in bootloader). But triggered by external reset it should be about 2s (enter bootloader and then timeout within it).

hackscribble, 2017-02-13:
@majekw I loaded the board under test with a program that raised an output pin at program start.  Then connected an Arduino with a test program that applied and released reset to the board under test, then measured the time until the output was raised.  This gave the 240ms figure, which was consistent with what I saw when the board under test had a simple blink program and I watched how long it took between reset and blinking to start.  I ran the tests several times and the figure was the same.

I don't that original test board any more, so can't repeat the tests on it.  A second board that I have just completed does **not** show the same symptoms, i.e. the bootloader period is 1.5s and it flashes the LED at the expected rate.  So I'm putting it down maybe to a problem with the way I burned the bootloader onto the original board.

alshorin, 2017-03-01:
I'm tired of fighting with this file optiboot (original) bootloader and modified it so it was constantly in download mode until you removed the jumper between PD7 and GND.
Insert in file optiboot_flash.c (484) before appStart(ch);
if (PIND & (1<<PIND7)) 
and you will be happy...

closed 2017-04-05 by hackscribble