MCUdude/MegaCore #30: Can't find Mega2561 after installation of MegaCore,json

PingCube, 2017-02-09:
Hi nice work but i can't finde the mega2561 or the mega1281 in Boardmanager after installation. I use the version 1.0.5

regards

elektronikjogi

[image]

per1234, 2017-02-10:
The problem is there hasn't been a release of MegaCore since support for ATmega2561 and ATmega1281 was added. To use those boards you will need do a manual instruction following these instructions:
https://github.com/MCUdude/MegaCore#manual-installation
It's probably best to remove the Boards Manager installation of MegaCore because it might conflict with the manual installation.

PingCube, 2017-02-11:
Thanks for your help

Best regards

elektronikjogi

MCUdude, 2017-02-11:
The latest boards manager version (v1.0.6) is now available, with Atmega1281/2561 support!

closed 2017-02-11 by MCUdude