MCUdude/MegaCore #34: How about full atmega640,1280,2560 support?

wa2mze, 2017-04-02:
I know that two of these are 'supported' as the arduino mega board.  However, this board ignores over a dozen i/o pins, and there is no support for the atmega640 part.  It should be easy enough to add an i/o map that enables full use of all pins on these parts, and to add support for the '640 variant.

I'll look into this myself eventually, but thought I'd mention it incase someone else wants to pick up this gauntlet.

MCUdude, 2017-04-02:
Hi! I'm sorry but you're 5 days too late 😉 allready working on that. Check out the [avr-100-pin] ( https://github.com/MCUdude/MegaCore/tree/avr-100-pin) branch! I'd love some help if you'd like to participate. I need help with testing, and finishing the "AVR pinout" for these puppies.

wa2mze, 2017-04-02:
Looks like all that is up there now is a place holder, cloned from the existing megacore branch.
Let me know when you have something ready to test, or if you need any help with this.

per1234, 2017-04-02:
It's a lot more than that. See:
https://github.com/MCUdude/MegaCore/compare/master...avr-100-pin

wa2mze, 2017-04-02:
I had done a quick read on the hub.  I'll try downloading from the 100 pin branch.

MCUdude, 2017-04-07:
The pinouts for the 100 pin chips should now be working. I'd appreciate if someone would like to help me out by testing IO pins, interrupts, analogRead and stuff like that with the new pinout.

> BTW, I was thinking, how about swapping the serial0 and serial1 lines for the 64 pin parts so that the ICP lines don't get shared with the serial port USB adapter? IE: us the tx1/rx1 lines for the serial object, and the tx0,rx0 lines for the serial1 object?

Sorry, but this core have been public for quite a while now, and a lot of people have built HW that corresponds to this exact pinout. If you want to use UART1 for uploading, it's as easy as modifying the _makeall_ file and execute it

wa2mze, 2017-04-08:
I guess there could be a variant with the serial lines swapped, I know it's real easy to do.

MCUdude, 2017-05-25:
Now that support for ATmega640/1280/2560 improperly implemented, we can now close this issue.

closed 2017-05-25 by MCUdude