MCUdude/MegaCore #35: Interrupt not working

jrozner, 2017-04-04:
Have some code that I prototyped on an Arduino Uno and Mega2560. It works fine on those platforms but when I move the code over it doesn't work. The issue is that an interrupt attached to pin 4 never fires.

The hardware is setup to pull the pin high and pressing a button will bring it low. I've verified this is actually happening by attaching an oscilloscope and checking the waveform. I'm also able to switch the code to do a digitalRead instead and that works correctly, it's only the interrupt that doesn't fire. The relevant code is:
```arduino
#define TRIGGER_PIN 4

pinMode(TRIGGER_PIN, INPUT);
attachInterrupt(digitalPinToInterrupt(TRIGGER_PIN), triggerPress, LOW);
```

MCUdude, 2017-04-04:
Thanks for reporting the issue, and actually validating your circuit with an oscilloscope. I'm on vacation right now, but I'll have a look at it next week 🌞

MCUdude, 2017-04-04:
While you're at it, could you please test your code without using the digitalPinToInterrupt macro, but rather insert the _real_ interrupt number? You can find the corresponding interrupt number in the pinout pic in the Readme file.

jrozner, 2017-04-04:
Gave removing digitalPinToInterrupt a try and it was no different. It's connected to pin 4 which corresponds to interrupt 4 and given the macro should end up the exact same value if I'm reading it correctly. I also tried swapping around some of the other settings than LOW like RISING and FALLING and none of those changed the behavior. I can try cutting the trace and soldering it to a different pin to try out some of the other interrupts but I don't have an easy way to test all of them as this is on a custom PCB and not a demo board.

MCUdude, 2017-04-04:
You don't have to cut traces if you don't have to, I have a dev board I can test on. One thing, could your try to add sei(); at the very end of the setup function? This enables global interrupts, so nothing will happend if this is not ran (which it should be by default). 

I haven't seen your entire code, but I assure you're stripped away everything to isolate this issue as much as possible?

jrozner, 2017-04-04:
I tried adding sei() at the end of setup and saw no change. Here is the isolated code that I've tested against. I've also removed the debounce logic as well with no effect.

```arduino
#define TRIGGER_PIN 4
#define RED_TEAM_PIN 6

volatile boolean shouldFire = false;
const int debounceDelay = 1000;
volatile long lastDebounceTime = 0;

void setup() {
  pinMode(RED_TEAM_PIN, OUTPUT);
  pinMode(TRIGGER_PIN, INPUT);
  digitalWrite(RED_TEAM_PIN, HIGH);
  attachInterrupt(digitalPinToInterrupt(TRIGGER_PIN), triggerPress, LOW);
}

void loop() {
  if (shouldFire == true) {
    shoot();
    shouldFire = false;
  }
}

void triggerPress() {
  long now = millis();
  if ((now - lastDebounceTime) > debounceDelay) {
    shouldFire = true;
    lastDebounceTime = now;
  }
}

void shoot() {
  digitalWrite(RED_TEAM_PIN, LOW);
  delay(1000);
  digitalWrite(RED_TEAM_PIN, HIGH);
}
```

MCUdude, 2017-04-07:
I found the issue. Turned out that [these lines](https://github.com/MCUdude/MegaCore/commit/601194f69b21d457a67caf14a522f76af3285ee8#diff-6dd1f07b84035a357edbabc7be7a593bL300) didn't match the current pinout.

jrozner, 2017-04-07:
Thanks! In fixing one issue on the board I was testing this on I broke something else. Still able to program the mcu but it doesn't look like it's actually executing anything. Gonna try swapping for a new one and I can give this a shot.

MCUdude, 2017-04-08:
Stupid late-night coding! All interrupts are working except for INT4. Try using the [avr-100-pin](https://github.com/MCUdude/MegaCore/tree/avr-100-pin) fork, which has this issue resolved. I'll merge this branch into the master as soon as it's a little more well tested

jrozner, 2017-04-12:
Gave this a try and it's working now. Thanks!

MCUdude, 2017-04-13:
Great! Thanks for reporting 😃

closed 2017-04-13 by MCUdude