MCUdude/MegaCore #36: analogReference(EXTERNAL)

czghzc, 2017-04-05:
analogReference(EXTERNAL);

However, the "EXTERNAL" is not what we understand the external reference voltage, but AVCC. 
My circuit uses an external reference voltage, and it is so set, but I found that my reference voltage is wrong, but I can not find my circuit and components have any problems, this problem plagued me a long time, and finally found The result of this setting is to connect the AVCC to the VREF Pin, so there will be such a problem.

[image]

Also, the reference voltage on the Mega128 DataSheet is set such that the REFS register is set to "11" to be the internal reference, but it is "10" in your file, but it also seems to work properly.

I recommend: 
     DEFAULT -> AVCC
     INTERNAL -> Internal 2.56V Voltage Reference
     EXTERNAL -> True external connection for reference voltage
Do you see if this is feasible?

MCUdude, 2017-04-05:
You're absolutely correct, I can see now that I messed up this in one of the recent commits. You can find these designs in the Arduino.h file. I'm on vacation right now, but I'd gladly merge a pull request if you want to contribute!

czghzc, 2017-04-05:
Oh, yes, now it's all right, thank you!

MCUdude, 2017-04-05:
Great, thanks for reporting!

closed 2017-04-05 by MCUdude