MCUdude/MegaCore #37: building bootstraps fail for branch 100-pin

wa2mze, 2017-04-08:
I tried executing ./makeall on optiboot_flash_256kb directory and got the following output:
[errors.txt](https://github.com/MCUdude/MegaCore/files/907542/errors.txt)

I tried building a single bootstrap with just "make atmega2561" selecting the defaults.  Which seems to woirk:
avr-gcc (GCC) 4.8.2
Copyright (C) 2013 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

BAUD RATE CHECK: Desired: 115200, Real: 117647, UBRRL = 16, Error=2.1%
avr-gcc -g -Wall -Os -fno-split-wide-types -mrelax -mmcu=atmega2561 -DF_CPU=16000000L  -DBAUD_RATE=115200 -DLED_START_FLASHES=3      -DBIGBOOT    -c -o optiboot_flash.o optiboot_flash.c
optiboot_flash.c:288:6: warning: #warning BAUD_RATE error greater than 2% [-Wcpp]
     #warning BAUD_RATE error greater than 2%
      ^
avr-gcc -g -Wall -Os -fno-split-wide-types -mrelax -mmcu=atmega2561 -DF_CPU=16000000L  -DBAUD_RATE=115200 -DLED_START_FLASHES=3      -DBIGBOOT  -Wl,--section-start=.text=0x3fc00 -Wl,--section-start=.version=0x3fffe -Wl,--relax -nostartfiles -nostdlib -o atmega2561/optiboot_flash_atmega2561__.elf optiboot_flash.o -lc
avr-size atmega2561/optiboot_flash_atmega2561__.elf
   text    data     bss     dec     hex filename
    666       0       0     666     29a atmega2561/optiboot_flash_atmega2561__.elf
avr-objcopy -j .text -j .data -j .version --set-section-flags .version=alloc,load -O ihex atmega2561/optiboot_flash_atmega2561__.elf atmega2561/optiboot_flash_atmega2561__.hex
avr-objdump -h -S atmega2561/optiboot_flash_atmega2561__.elf > atmega2561

But produces truncated files:

[optiboot_flash_atmega2561__.hex.txt](https://github.com/MCUdude/MegaCore/files/907546/optiboot_flash_atmega2561__.hex.txt)
[optiboot_flash_atmega2561__.lst.txt](https://github.com/MCUdude/MegaCore/files/907547/optiboot_flash_atmega2561__.lst.txt)

MCUdude, 2017-04-08:
First I recommend using avr-gcc 4.9.3 instead. At least I do it, and it works for me ;)

try executing this command in your terminal window.
`make atmega2560 AVR_FREQ=16000000L BAUD_RATE=115200 UART=0 LED=B5 LED_START_FLASHES=2 -BIGBOOT
`

does this fails as well? It seems like there's an issue with the passed parameters.. the baud rate error is at -100%..

wa2mze, 2017-04-15:
I found the problem.
override LDFLAGS       = $(LDSECTIONS) -Wl,--relax -nostartfiles -nostdlib
Need to remove the -nostlib option from LDFLAGS line in Makefile.  I also removed the comments on the three lines that enable using the arduio supplied avr tools.

MCUdude, 2017-04-16:
Great, thanks for reporting back! Will keep this in mind if someone else get this problem as well.

Don't hesitate to open an new issue if you find any other bugs, especially related to the 100 pin branch 👍

closed 2017-04-16 by MCUdude