MCUdude/MegaCore #38: Macro errors with IRLib2

jrozner, 2017-04-13:
Not sure if this is an error in IRLib2's ifdefs or in the MegaCore support but when trying to even include in the library for the ATmega64a I run into these errors. I've gone through the IRLibe2 code and checked around for ifdefs that optionally include code for specific avr chips but didn't see anything that didn't have an else block that would skip including these. The define macros listed here all seem to be part of the Arduino framework so I'm confused as to why they wouldn't be included.

```
In file included from /Users/joe/Documents/Arduino/libraries/IRLibProtocols/IRLibSendBase.cpp:14:0:
/Users/joe/Documents/Arduino/libraries/IRLibProtocols/IRLibSendBase.cpp: In member function 'void IRsendBase::enableIROut(uint8_t)':
/Users/joe/Documents/Arduino/libraries/IRLibProtocols/IRLibHardware.h:286:34: error: 'TIMSK2' was not declared in this scope
  #define IR_RECV_DISABLE_INTR   (TIMSK2 = 0)
                                  ^
/Users/joe/Documents/Arduino/libraries/IRLibProtocols/IRLibSendBase.cpp:77:2: note: in expansion of macro 'IR_RECV_DISABLE_INTR'
  IR_RECV_DISABLE_INTR; //Timer2 Overflow Interrupt    
  ^
/Users/joe/Documents/Arduino/libraries/IRLibProtocols/IRLibHardware.h:193:3: error: 'TCCR2A' was not declared in this scope
   TCCR2A = _BV(WGM20);  TCCR2B = _BV(WGM22) | _BV(CS20); \
   ^
/Users/joe/Documents/Arduino/libraries/IRLibProtocols/IRLibSendBase.cpp:80:2: note: in expansion of macro 'IR_SEND_CONFIG_KHZ'
  IR_SEND_CONFIG_KHZ(khz);
  ^
/Users/joe/Documents/Arduino/libraries/IRLibProtocols/IRLibHardware.h:193:25: error: 'TCCR2B' was not declared in this scope
   TCCR2A = _BV(WGM20);  TCCR2B = _BV(WGM22) | _BV(CS20); \
                         ^
/Users/joe/Documents/Arduino/libraries/IRLibProtocols/IRLibSendBase.cpp:80:2: note: in expansion of macro 'IR_SEND_CONFIG_KHZ'
  IR_SEND_CONFIG_KHZ(khz);
  ^
In file included from /Users/joe/Library/Arduino15/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.4-arduino2/avr/include/avr/io.h:99:0,
                 from /Users/joe/Library/Arduino15/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.4-arduino2/avr/include/avr/pgmspace.h:90,
                 from /Users/joe/Documents/Arduino/hardware/MegaCore/avr/cores/MegaCore/Arduino.h:28,
                 from /Users/joe/Documents/Arduino/libraries/IRLibProtocols/IRLibProtocols.h:12,
                 from /Users/joe/Documents/Arduino/libraries/IRLibProtocols/IRLibSendBase.h:16,
                 from /Users/joe/Documents/Arduino/libraries/IRLibProtocols/IRLibSendBase.cpp:13:
/Users/joe/Documents/Arduino/libraries/IRLibProtocols/IRLibHardware.h:193:38: error: 'WGM22' was not declared in this scope
   TCCR2A = _BV(WGM20);  TCCR2B = _BV(WGM22) | _BV(CS20); \
                                      ^
/Users/joe/Documents/Arduino/libraries/IRLibProtocols/IRLibSendBase.cpp:80:2: note: in expansion of macro 'IR_SEND_CONFIG_KHZ'
  IR_SEND_CONFIG_KHZ(khz);
  ^
In file included from /Users/joe/Documents/Arduino/libraries/IRLibProtocols/IRLibSendBase.cpp:14:0:
/Users/joe/Documents/Arduino/libraries/IRLibProtocols/IRLibHardware.h:194:3: error: 'OCR2A' was not declared in this scope
   OCR2A = pwmval; OCR2B = pwmval / 3; })
   ^
/Users/joe/Documents/Arduino/libraries/IRLibProtocols/IRLibSendBase.cpp:80:2: note: in expansion of macro 'IR_SEND_CONFIG_KHZ'
  IR_SEND_CONFIG_KHZ(khz);
  ^
/Users/joe/Documents/Arduino/libraries/IRLibProtocols/IRLibHardware.h:194:19: error: 'OCR2B' was not declared in this scope
   OCR2A = pwmval; OCR2B = pwmval / 3; })
                   ^
/Users/joe/Documents/Arduino/libraries/IRLibProtocols/IRLibSendBase.cpp:80:2: note: in expansion of macro 'IR_SEND_CONFIG_KHZ'
  IR_SEND_CONFIG_KHZ(khz);
  ^
/Users/joe/Documents/Arduino/libraries/IRLibProtocols/IRLibSendBase.cpp: In member function 'void IRsendBase::mark(uint16_t)':
/Users/joe/Documents/Arduino/libraries/IRLibProtocols/IRLibHardware.h:188:33: error: 'TCCR2A' was not declared in this scope
  #define IR_SEND_PWM_START     (TCCR2A |= _BV(COM2B1))
                                 ^
/Users/joe/Documents/Arduino/libraries/IRLibProtocols/IRLibSendBase.cpp:95:2: note: in expansion of macro 'IR_SEND_PWM_START'
  IR_SEND_PWM_START;
  ^
In file included from /Users/joe/Library/Arduino15/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.4-arduino2/avr/include/avr/io.h:99:0,
                 from /Users/joe/Library/Arduino15/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.4-arduino2/avr/include/avr/pgmspace.h:90,
                 from /Users/joe/Documents/Arduino/hardware/MegaCore/avr/cores/MegaCore/Arduino.h:28,
                 from /Users/joe/Documents/Arduino/libraries/IRLibProtocols/IRLibProtocols.h:12,
                 from /Users/joe/Documents/Arduino/libraries/IRLibProtocols/IRLibSendBase.h:16,
                 from /Users/joe/Documents/Arduino/libraries/IRLibProtocols/IRLibSendBase.cpp:13:
/Users/joe/Documents/Arduino/libraries/IRLibProtocols/IRLibHardware.h:188:47: error: 'COM2B1' was not declared in this scope
  #define IR_SEND_PWM_START     (TCCR2A |= _BV(COM2B1))
                                               ^
/Users/joe/Documents/Arduino/libraries/IRLibProtocols/IRLibSendBase.cpp:95:2: note: in expansion of macro 'IR_SEND_PWM_START'
  IR_SEND_PWM_START;
  ^
In file included from /Users/joe/Documents/Arduino/libraries/IRLibProtocols/IRLibSendBase.cpp:14:0:
/Users/joe/Documents/Arduino/libraries/IRLibProtocols/IRLibSendBase.cpp: In member function 'void IRsendBase::space(uint16_t)':
/Users/joe/Documents/Arduino/libraries/IRLibProtocols/IRLibHardware.h:190:31: error: 'TCCR2A' was not declared in this scope
  #define IR_SEND_PWM_STOP    (TCCR2A &= ~(_BV(COM2B1)))
                               ^
/Users/joe/Documents/Arduino/libraries/IRLibProtocols/IRLibSendBase.cpp:101:2: note: in expansion of macro 'IR_SEND_PWM_STOP'
  IR_SEND_PWM_STOP;
  ^
In file included from /Users/joe/Library/Arduino15/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.4-arduino2/avr/include/avr/io.h:99:0,
                 from /Users/joe/Library/Arduino15/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.4-arduino2/avr/include/avr/pgmspace.h:90,
                 from /Users/joe/Documents/Arduino/hardware/MegaCore/avr/cores/MegaCore/Arduino.h:28,
                 from /Users/joe/Documents/Arduino/libraries/IRLibProtocols/IRLibProtocols.h:12,
                 from /Users/joe/Documents/Arduino/libraries/IRLibProtocols/IRLibSendBase.h:16,
                 from /Users/joe/Documents/Arduino/libraries/IRLibProtocols/IRLibSendBase.cpp:13:
/Users/joe/Documents/Arduino/libraries/IRLibProtocols/IRLibHardware.h:190:47: error: 'COM2B1' was not declared in this scope
  #define IR_SEND_PWM_STOP    (TCCR2A &= ~(_BV(COM2B1)))
                                               ^
/Users/joe/Documents/Arduino/libraries/IRLibProtocols/IRLibSendBase.cpp:101:2: note: in expansion of macro 'IR_SEND_PWM_STOP'
  IR_SEND_PWM_STOP;
  ^
Using library LiquidCrystal at version 1.0.5 in folder: /Applications/Arduino.app/Contents/Java/libraries/LiquidCrystal 
Using library EEPROM at version 2.0 in folder: /Users/joe/Documents/Arduino/hardware/MegaCore/avr/libraries/EEPROM 
Using library IRLibRecvPCI in folder: /Users/joe/Documents/Arduino/libraries/IRLibRecvPCI (legacy)
Using library IRLib2 in folder: /Users/joe/Documents/Arduino/libraries/IRLib2 (legacy)
Using library IRLibProtocols in folder: /Users/joe/Documents/Arduino/libraries/IRLibProtocols (legacy)
exit status 1
Error compiling for board ATmega64.
```

MCUdude, 2017-04-13:
The reason why you're getting an error message is because there's hardware incompatibility.  There's no such thing as timer2b on the ATmega64. If you're sending or decoding IR signals, I suggest using [Arduino IRremote](https://github.com/z3t0/Arduino-IRremote) instead. Much more popular, and [dedicated support](https://github.com/z3t0/Arduino-IRremote/blob/e0c2649b9f6681cd6ffb89514e94955af5634225/boarddefs.h#L119) for ATmega64/128.

jrozner, 2017-04-17:
I'll take a look at IRemote and see if I can get it working in a similar way. When I looked at it in the past the reason I chose to use IRLib2 was the support for interrupt based receiving since the actual event loop the mcu is running is potentially a bit long and didn't want to miss transmissions. Thanks for explaining why it's not working though, that makes sense.

closed 2017-04-28 by jrozner