MCUdude/MegaCore #40: Sketch destroing Bootloader.

DallySon, 2017-04-26:
Dear Sirs,

Thanks for great Job.
But I have one problem. 
I using UNO as ISP for burning Bootloader. All works great.
But After first uploading of any sketch by USB-adapter(FTDI) - I have no more bootloader on Atmega128.
I use ARDUINO_USB_LIGHT - adapter. I have cap. 0.1 beetween RESETS (adapter-atmega). For my projects on Atmega328 - a have't this problems.
How I can resolve this situation.

Hope on Your help.
Thanks.

MCUdude, 2017-04-28:
If you're able to upload your code only the first time after you're burned the bootloader, there's somethig wrong with your auto reset circuity. Connect an LED to PB7. Does this flash when you reset the microcontroller? If it does, the bootloader is still present

DallySon, 2017-05-03:
Thanks so much! I'll try in a couple days and wiill inform about result.

DallySon, 2017-05-07:
I connected a LED to PB7 (and to PB5 like on your schematic) but it doesn't flash.
So, I think maybe something wrong with my board (I use board from our oldest project 2007). Give me some more days (i ordered PCB for  tqfp64  and NEW USB adapter  (cp2001) with DTR but it may take 5-7 days). I'll do new circuit exact on your schematic and we can see the real problem. Or may be no any problem at all :). Thnks!

DallySon, 2017-05-12:
Now I builded a new circuit board and all thing started running very well!!!
Thank You so much for your help and great job!!!
Take care!!

DallySon, 2017-05-12:
Sorry, one more question. I used library for deep sleepeng CPU - SnoozeLib before. Now it doesn't work.
I tried to use sleep()  (with SLEEP_POWER_DOWN) - but I can't to wakeUp my board after sleep().  How I can to do it? If it possible - example( or link) please!

MCUdude, 2017-05-12:
Great to hear you figured out the issue. Thanks for using MegaCore!

It seems like the snoozelib is based on `avr/sleep.h`. If you're having issues, you should create an issue over at the library repository 🙂

closed 2017-05-12 by MCUdude