MCUdude/MegaCore #46: Installation of MegaCore

gratefulfrog, 2017-05-18:
I wanted to do the manual install but I don't have a ~/Documents/Arduino directory.

I do have:
~/.arduino
~/.arduino15/packages/arduino/hardware/
~/.arduino-create/

Any suggestions?

Thanks,
Bob

per1234, 2017-05-18:
I'm guessing that `~/Documents/Arduino` is the default sketchbook location on macOS. You should put it in `{sketchbook folder}/hardware`. You can find the location of your sketchbook folder in the Arduino IDE at **File > Preferences > Sketchbook location**.

@MCUdude maybe the manual installation instructions should be updated to be more generic rather than assuming the sketchbook folder is at a specific location.

gratefulfrog, 2017-05-18:
@per1234 
Thanks! on my linux system, the default is ~/sketchbook/hardware. When I extract the zip archive to that directory, I get the new tools boards MegaCore section and it seems to work as expected!
@MCUdude 
Indeed, if the instructions were updated to say, "extract the zip archive to the hardware subdirectory of the user sketchbook directory." it would be much more clear.

Thanks!
Bob

closed 2017-05-18 by MCUdude