MCUdude/MegaCore #57: Problems with intrrupts on atmega128

WD24, 2017-06-26:
Hi 
Firstly Thank you for you hardwork supplying this repository to everyone.

Im having problems compiling GRBL for the ATmega128. and I am wondering if you can help.

I can compile GRBL on ATmega2561, ATmega128, ATmega2560, Atmega1280 and ATmega640 but when I try to compile it on the ATmega128 (and on the ATmega64) I keep getting the same error.

I'm using GRBL1.1f (https://github.com/gnea/grbl) but I have also tried using a few other versions. I'm using arduino 1.8.1 on windows 10 and arduino 1.8.3 on Linux mint and keep getting the same error. 

I think the issue is relating to the intrrupt in the core rather than the GRBL software. I've spent several days now googling for and trying to get an answer but still to no luck.

Any help would be greatfully receaved. Thank you

The output from the arduino IDE is:

In file included from C:\Users\X\Documents\Arduino\libraries\grbl\grbl.h:47:0,
                 from C:\Users\X\Documents\Arduino\libraries\grbl\limits.c:22:
C:\Users\X\Documents\Arduino\libraries\grbl\limits.c: In function 'limits_init':
C:\Users\X\Documents\Arduino\libraries\grbl\cpu_map.h:73:28: error: 'PCMSK0' undeclared (first use in this function)
   #define LIMIT_PCMSK      PCMSK0 // Pin change interrupt register
                            ^
C:\Users\X\Documents\Arduino\libraries\grbl\limits.c:44:5: note: in expansion of macro 'LIMIT_PCMSK'
     LIMIT_PCMSK |= LIMIT_MASK; // Enable specific pins of the Pin Change Interrupt
     ^
C:\Users\X\Documents\Arduino\libraries\grbl\cpu_map.h:73:28: note: each undeclared identifier is reported only once for each function it appears in
   #define LIMIT_PCMSK      PCMSK0 // Pin change interrupt register
                            ^
C:\Users\X\Documents\Arduino\libraries\grbl\limits.c:44:5: note: in expansion of macro 'LIMIT_PCMSK'
     LIMIT_PCMSK |= LIMIT_MASK; // Enable specific pins of the Pin Change Interrupt
     ^
C:\Users\X\Documents\Arduino\libraries\grbl\limits.c:45:5: error: 'PCICR' undeclared (first use in this function)
     PCICR |= (1 << LIMIT_INT); // Enable Pin Change Interrupt
     ^
In file included from C:\Users\X\Documents\Arduino\libraries\grbl\grbl.h:47:0,
                 from C:\Users\X\Documents\Arduino\libraries\grbl\limits.c:22:
C:\Users\X\Documents\Arduino\libraries\grbl\cpu_map.h:71:28: error: 'PCIE0' undeclared (first use in this function)
   #define LIMIT_INT        PCIE0  // Pin change interrupt enable pin
                            ^
C:\Users\X\Documents\Arduino\libraries\grbl\limits.c:45:20: note: in expansion of macro 'LIMIT_INT'
     PCICR |= (1 << LIMIT_INT); // Enable Pin Change Interrupt
                    ^
C:\Users\X\Documents\Arduino\libraries\grbl\limits.c: In function 'limits_disable':
C:\Users\X\Documents\Arduino\libraries\grbl\cpu_map.h:73:28: error: 'PCMSK0' undeclared (first use in this function)
   #define LIMIT_PCMSK      PCMSK0 // Pin change interrupt register
                            ^
C:\Users\X\Documents\Arduino\libraries\grbl\limits.c:61:3: note: in expansion of macro 'LIMIT_PCMSK'
   LIMIT_PCMSK &= ~LIMIT_MASK;  // Disable specific pins of the Pin Change Interrupt
   ^
C:\Users\X\Documents\Arduino\libraries\grbl\limits.c:62:3: error: 'PCICR' undeclared (first use in this function)
   PCICR &= ~(1 << LIMIT_INT);  // Disable Pin Change Interrupt
   ^
In file included from C:\Users\X\Documents\Arduino\libraries\grbl\grbl.h:47:0,
                 from C:\Users\X\Documents\Arduino\libraries\grbl\limits.c:22:
C:\Users\X\Documents\Arduino\libraries\grbl\cpu_map.h:71:28: error: 'PCIE0' undeclared (first use in this function)
   #define LIMIT_INT        PCIE0  // Pin change interrupt enable pin
                            ^
C:\Users\X\Documents\Arduino\libraries\grbl\limits.c:62:19: note: in expansion of macro 'LIMIT_INT'
   PCICR &= ~(1 << LIMIT_INT);  // Disable Pin Change Interrupt
                   ^
exit status 1
Error compiling for board ATmega128.

MCUdude, 2017-06-27:
Not strange at all. The ATmega64 and 128 is quite old and doesn't have PCINTs, only regular interrupts. If you want support for these microcontrollers you'll have to create an issue over at the grbl repo 🙂 

It's a hardware limitation, and there's nothing i can do about it

closed 2017-06-27 by MCUdude

WD24, 2017-06-30:
Thank you.
I've just bought some ATmega2561's which should be pin compattible, I'll swap them out.