MCUdude/MegaCore #58: Unable to see any output in the Serial monitor

shaktidhar, 2017-07-01:
Hi, 
I have been able to successfully burn the boot loader onto Atmega128 and then upload sketch to it as well. 
Sketch works, as one would expect it to. 
I am using USBTinyISP as the programmer. 
But when I try to check the serial output, I am unable to. 

(It's possible I am missing something here, Serial doesn't go really with ISP, no TX, RX pins there ). 

Thanks.

MCUdude, 2017-07-03:
The hardware serial works just fine,, I've used it many times. It's **a)** something wrong with your hardware setup, og **b)** something wrong with your code.

Connect your USB to serial adapter as shown in the minimal setup in the README. I think you'll have to unplug your programmer, since the hardware UART and the programming lines share the same pins (PE0 and PE1).

Try uploading a simple sketch like tha **AnalogReadSerial**. Remember that the TX pin on the microcontroller goes to the RX pin on the USB to serial adapter and vice versa

shaktidhar, 2017-07-03:
I knew I am missing out something here. 

So, I can upload programs via TinyISP but to check serial monitor, I will have to use the other port. 

Got it. 
Sorry about the false flag.

closed 2017-07-03 by shaktidhar