MCUdude/MegaCore #63: Atmega2560V, trying to burn bootloaders

beingmachine, 2017-08-09:
Hello,
I am trying to burn MegaCore bootloaders to it, I have setup my custom PCB  with Atmega2560V-8AU and 8Mhz Crystal + Load Caps all assembled on using PCBA process.
I am trying to burn MegaCore bootloaders to it, I am using Arduino UNO as ISP Programmer. My connections are all correct, It shows different Device Signatures (see below) whenever I hit Burn Bootloaders....

> 0x0000ff
> 0x00003f
> 0x0000ff
> 0x00ff00
> 0x001000
> 0xff0000
> 0xfc0000
> 0x00fe00

This is complete output with different Device Signatures

> Arduino: 1.8.3 (Windows 10), Board: "ATmega2560, Arduino MEGA pinout, 2.7v, Disabled (default), 8 MHz external"
> 
> C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avrdude -CC:\Users\mudas\OneDrive\Documents\Arduino\hardware\MegaCore-avr-100-pin\avr/avrdude.conf -v -patmega2560 -cstk500v1 -PCOM4 -b19200 -e -Ulock:w:0x3f:m -Uefuse:w:0xfd:m -Uhfuse:w:0xd6:m -Ulfuse:w:0xf7:m 
> 
> avrdude: Version 6.3, compiled on Jan 17 2017 at 12:00:53
>          Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
>          Copyright (c) 2007-2014 Joerg Wunsch
> 
>          System wide configuration file is "C:\Users\mudas\OneDrive\Documents\Arduino\hardware\MegaCore-avr-100-pin\avr/avrdude.conf"
> 
>          Using Port                    : COM4
>          Using Programmer              : stk500v1
>          Overriding Baud Rate          : 19200
>          AVR Part                      : ATmega2560
>          Chip Erase delay              : 9000 us
>          PAGEL                         : PD7
>          BS2                           : PA0
>          RESET disposition             : dedicated
>          RETRY pulse                   : SCK
>          serial program mode           : yes
>          parallel program mode         : yes
>          Timeout                       : 200
>          StabDelay                     : 100
>          CmdexeDelay                   : 25
>          SyncLoops                     : 32
>          ByteDelay                     : 0
>          PollIndex                     : 3
>          PollValue                     : 0x53
>          Memory Detail                 :
> 
>                                   Block Poll               Page                       Polled
>            Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
>            ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
>            eeprom        65    10     8    0 no       4096    8      0  9000  9000 0x00 0x00
>            flash         65    10   256    0 yes    262144  256   1024  4500  4500 0x00 0x00
>            lfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
>            hfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
>            efuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
>            lock           0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
>            calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
>            signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00
> 
>          Programmer Type : STK500
>          Description     : Atmel STK500 Version 1.x firmware
>          Hardware Version: 2
>          Firmware Version: 1.18
>          Topcard         : Unknown
>          Vtarget         : 0.0 V
>          Varef           : 0.0 V
>          Oscillator      : Off
>          SCK period      : 0.1 us
> 
> avrdude: AVR device initialized and ready to accept instructions
> 
> Error while burning bootloader.
> Reading | ################################################## | 100% 0.02s
> 
> avrdude: Device signature = 0x00fe00
> avrdude: Expected signature for ATmega2560 is 1E 98 01
>          Double check chip, or use -F to override this check.
> 
> avrdude done.  Thank you.
> 
> This report would have more information with
> "Show verbose output during compilation"
> option enabled in File -> Preferences.

What should be the problem.??? How to solve this.?

MCUdude, 2017-08-09:
Do you have a schematic and a picture of your hardware you can post? I suspect there might be a bad connection somewhere. Does the device signature always reported to be 0x00fe00 or does it varies?

beingmachine, 2017-08-09:
Hello,
Schematics and pictures of the board are attached to this comment. 
Device Signature is reporting to 0x000000 now before it was my mistake, Reset pin was floating.
[vt2930---SCH FILE.pdf](https://github.com/MCUdude/MegaCore/files/1212923/vt2930---SCH.FILE.pdf)
[image]

I have used the same connection with my Atmega2560-16AU with 16Mhz crystal on 5V, I have used Nick Gammon Arduino Bootloaders to burn and it works.

MCUdude, 2017-08-09:
It's most likely a bad connection or an error on your board. Make sure the microcontroller has 3.3v on its Vcc pins. Make sure the reset pin is pulled up to 3.3v with a resistor; typically 10k. I also don't see any capacitors from the two crystal pins to ground (should be 22pF). Do you have bypass capacitors close to the Vcc and god pins on the microcontroller (~100nF)? This is very important

beingmachine, 2017-08-09:
> Make sure the reset pin is pulled up to 3.3v with a resistor; typically 10k..?
Which reset pin.? Target micro controllers or Programmers.?

 

> I also don't see any capacitors from the two crystal pins to ground (should be 22pF). 
I am using cstce8m00g55-r0, which I hope it has 22pF caps built-in.?

> Do you have bypass capacitors close to the Vcc and gnd pins on the microcontroller (~100nF)?
I have tried it, But no joy :(

is there any way to factory reset this chip, I mean to set fuses back to default.? or any other way to detect chip and its connection.? Because I have tried this PCB with my Atmega2560-16AU with 16Mhz Crystal and 22pF load caps on 5V.
Thank you

MCUdude, 2017-08-09:
The Murata CSTE resonator he built in caps. Just make sure the middle pin is grounded. I cannot see any decoupling capacitors close to the microcontroller. This might also be the reason why you're experiencing problems. 

When you first soldered the microcontroller to the board; where you able to communicate with it and change the fuse settings? If you're enabled the use of an external oscillator but you're having issues with it, the microcontroller will appear dead like yours.

beingmachine, 2017-08-09:
> Just make sure the middle pin is grounded.

Yes! the middle pin is grounded.

> I cannot see any decoupling capacitors close to the microcontroller.
see U10 in my schematics which is AMS117, 3.3V regulator, I have soldered 100nF like a stack C27 1uF capacitor.. 

> When you first soldered the microcontroller to the board; where you able to communicate with it and change the fuse settings?
These boards have come up with the PCBA with all the components on it.

To be honest, I am suspecting incorrect Fuse settings, Because these chips has come in a tray, some one might have played with.. How to correct it.? I have Atmel Studio installed on my computer..

MCUdude, 2017-08-09:
The caps close to the 1117 regulator isn't enough. They'll have to be placed as close as possible, ideally just a few millimeters from the microcontroller. 

The simplest way to recover potentially wrong fuse settings is to apply a clock signal to the XTAL1 pin, to force the processor to run. A megahertz will do it. BTW do you have an oscilloscope? If you do you can check if the oscillator is running by probing on one of the XTAL pins. 

Alternatively your can recover it using a high voltage parallel programmer. It requires lots of wires, and is almost never worth the trouble.

beingmachine, 2017-08-09:
BTW! now again, it's giving me different Device Signatures like 

> 0x0000ff
0x00ff00

etc

Why it's so.?

beingmachine, 2017-08-09:
And also I have tried using settings with internal clock,it also didn't
worked either. :(

beingmachine, 2017-08-09:
BTW! Which reset pin should I need to pull up? Target microcontroller or Arduino Uno?
Thanks 

_Sent from my Xiaomi Redmi Note 4 using [FastHub](https://play.google.com/store/apps/details?id=com.fastaccess.github)_

MCUdude, 2017-08-10:
> Why it's so.?

No idea..

> And also I have tried using settings with internal clock,it also didn't
worked either. :(

You're not able to change the fuse settings unless you can comunicate with the target. In this case, you can't.

When it comes to the reset pin, you need to add a 10k pullup resistor to the reset pin on the ATmega2560. According to the [ArduinoISP tutorial](https://www.arduino.cc/en/Tutorial/ArduinoISP) you need to add a 10uF capacitor between the reset pin and GND on your Arduino UNO.

This is what you should do:
* Add a pullup resistor on the ATmega2560 reset line
* Add a 10uF capacitor between reset and GND on the Arduino UNO
* **Add some decoupling capacitor CLOSE to the ATmega2560 Vcc and GND pins. THIS IS IMPORTANT, you should not ignore it. Google will tell you why**
* Connect an oscilloscope to one of the XTAL pins on the ATmega2560, and make sure that you can measure an 8 MHz sinewave
* you may add an external clock signal ( 1-8 MHz 3.3v sinewave or squarewave) to the XTAL1 pin to force the microcontroller to run (if the external oscillator is anabled in the fuse settings)
* Get a proper programmer, such as the USBasp or USBtinyISP

beingmachine, 2017-08-10:
> The simplest way to recover potentially wrong fuse settings is to apply a clock signal to the XTAL1 pin, to force the processor to run. A megahertz will do it.

Please see my test results here : [http://www.avrfreaks.net/forum/factory-reset-mega2560-and-mega2560v#comment-2241156](url)

What should be problem now.?

MCUdude, 2017-08-10:
You now got a device signature that looks very much like the ATmega2560s signature. Do you always get 0x1e9803, or does it vary? Honesty I don't know why the device signature doesn't match. I think the guys over at the AVR freaks forum knows this better than I do.

beingmachine, 2017-08-10:
> Do you always get 0x1e9803, or does it vary?

I am getting the same device signature repeatedly, it's not changing BTW! I have connected that 1Mhz to pin 34 of Atmega2560V..

Is that because I am powering Atmega2560V with 3.3V and Arduino UNO and Ardunio Mega at 5V

beingmachine, 2017-08-10:
I have tried with Nick Gammon chip detector sketch and here is the output : 

> Atmega chip detector.
Written by Nick Gammon.
Version 1.20
Compiled on Aug 11 2017 at 04:43:44 with Arduino IDE 10803.
Attempting to enter ICSP programming mode ...
Entered programming mode OK.
Signature = 0x1E 0x98 0x03 
Unrecogized signature.
LFuse = 0xC0 
HFuse = 0xD9 
EFuse = 0xFF 
Lock byte = 0xFF 
Clock calibration = 0xFF 

First 256 bytes of program memory:

00: 0xFF 0xFF 0x00 0x00 0xFF 0xFF 0x00 0x00 0xFF 0xFF 0x00 0x00 0xFF 0xFF 0x00 0x00 
10: 0xFF 0xFF 0x00 0x00 0xFF 0xFF 0x00 0x00 0xFF 0xFF 0x00 0x00 0xFF 0xFF 0x00 0x00 
20: 0xFF 0xFF 0x00 0x00 0xFF 0xFF 0x00 0x00 0xFF 0xFF 0x00 0x00 0xFF 0xFF 0x00 0x00 
30: 0xFF 0xFF 0x00 0x00 0xFF 0xFF 0x00 0x00 0xFF 0xFF 0x00 0x00 0xFF 0xFF 0x00 0x00 
40: 0xFF 0xFF 0x00 0x00 0xFF 0xFF 0x00 0x00 0xFF 0xFF 0x00 0x00 0xFF 0xFF 0x00 0x00 
50: 0xFF 0xFF 0x00 0x00 0xFF 0xFF 0x00 0x00 0xFF 0xFF 0x00 0x00 0xFF 0xFF 0x00 0x00 
60: 0xFF 0xFF 0x00 0x00 0xFF 0xFF 0x00 0x00 0xFF 0xFF 0x00 0x00 0xFF 0xFF 0x00 0x00 
70: 0xFF 0xFF 0x00 0x00 0xFF 0xFF 0x00 0x00 0xFF 0xFF 0x00 0x00 0xFF 0xFF 0x00 0x00 
80: 0xFF 0xFF 0x00 0x00 0xFF 0xFF 0x00 0x00 0xFF 0xFF 0x00 0x00 0xFF 0xFF 0x00 0x00 
90: 0xFF 0xFF 0x00 0x00 0xFF 0xFF 0x00 0x00 0xFF 0xFF 0x00 0x00 0xFF 0xFF 0x00 0x00 
A0: 0xFF 0xFF 0x00 0x00 0xFF 0xFF 0x00 0x00 0xFF 0xFF 0x00 0x00 0xFF 0xFF 0x00 0x00 
B0: 0xFF 0xFF 0x00 0x00 0xFF 0xFF 0x00 0x00 0xFF 0xFF 0x00 0x00 0xFF 0xFF 0x00 0x00 
C0: 0xFF 0xFF 0x00 0x00 0xFF 0xFF 0x00 0x00 0xFF 0xFF 0x00 0x00 0xFF 0xFF 0x00 0x00 
D0: 0xFF 0xFF 0x00 0x00 0xFF 0xFF 0x00 0x00 0xFF 0xFF 0x00 0x00 0xFF 0xFF 0x00 0x00 
E0: 0xFF 0xFF 0x00 0x00 0xFF 0xFF 0x00 0x00 0xFF 0xFF 0x00 0x00 0xFF 0xFF 0x00 0x00 
F0: 0xFF 0xFF 0x00 0x00 0xFF 0xFF 0x00 0x00 0xFF 0xFF 0x00 0x00 0xFF 0xFF 0x00 0x00 

Programming mode off.

beingmachine, 2017-08-11:
is there any way to set Device Signature at avrdude to 0x1e9803 from 0x1e9801.?

beingmachine, 2017-08-11:
I am out of Ideas now :(

MCUdude, 2017-08-11:
You can't change a device signature, but you can make avrdude ignore the signature by adding `-F` to the command. It's not considered safe but in your case you got nothing to loose.

Add `-F` to line 106, 113, 117 and 121 in the [platform.txt file](https://github.com/MCUdude/MegaCore/blob/master/avr/platform.txt#L106)

beingmachine, 2017-08-11:
[platform.txt](https://github.com/MCUdude/MegaCore/files/1218561/platform.txt)

is this correct.? I have added -F in the end

beingmachine, 2017-08-11:
I have changed as you have said and replaced with the existence in MegaCore folder in my documents.

The bootloader is getting in soo fast.?

 

> C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avrdude -F"-CC:\Program Files (x86)\Arduino\hardware\tools\avr/etc/avrdude.conf" -v -patmega2560 -cstk500v1 -PCOM4 -b19200 -e -Ulock:w:0x3f:m -Uefuse:w:0xfd:m -Uhfuse:w:0xd6:m -Ulfuse:w:0xf7:m 
C:\Program Files (x86)\Arduino\hardware\tools\avr\bin\avrdude: unknown option -- "
Usage: avrdude [options]
Options:
  -p <partno>                Required. Specify AVR device.
  -b <baudrate>              Override RS-232 baud rate.
  -B <bitclock>              Specify JTAG/STK500v2 bit clock period (us).
  -C <config-file>           Specify location of configuration file.
  -c <programmer>            Specify programmer type.
  -D                         Disable auto erase for flash memory
  -i <delay>                 ISP Clock Delay [in microseconds]
  -P <port>                  Specify connection port.
  -F                         Override invalid signature check.
  -e                         Perform a chip erase.
  -O                         Perform RC oscillator calibration (see AVR053). 
  -U <memtype>:r|w|v:<filename>[:format]
                             Memory operation specification.
                             Multiple -U options are allowed, each request
                             is performed in the order specified.
  -n                         Do not write anything to the device.
  -V                         Do not verify.
  -u                         Disable safemode, default when running from a script.
  -s                         Silent safemode operation, will not ask you if
                             fuses should be changed back.
  -t                         Enter terminal mode.
  -E <exitspec>[,<exitspec>] List programmer exit specifications.
  -x <extended_param>        Pass <extended_param> to programmer.
  -y                         Count # erase cycles in EEPROM.
  -Y <number>                Initialize erase cycle # in EEPROM.
  -v                         Verbose output. -v -v for more.
  -q                         Quell progress output. -q -q for less.
  -l logfile                 Use logfile rather than stderr for diagnostics.
  -?                         Display this usage.

avrdude version 6.3, URL: <http://savannah.nongnu.org/projects/avrdude/>
C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avrdude -F -CC:\Program Files (x86)\Arduino\hardware\tools\avr/etc/avrdude.conf -v -patmega2560 -cstk500v1 -PCOM4 -b19200 -Uflash:w:C:\Users\mudas\OneDrive\Documents\Arduino\hardware\MegaCore-master\avr/bootloaders/optiboot_flash/atmega2560/optiboot_flash_atmega2560_38400_8000000L.hex:i -Ulock:w:0x0f:m 

avrdude: Version 6.3, compiled on Jan 17 2017 at 12:00:53
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Program Files (x86)\Arduino\hardware\tools\avr/etc/avrdude.conf"

         Using Port                    : COM4
         Using Programmer              : stk500v1
         Overriding Baud Rate          : 19200
         AVR Part                      : ATmega2560
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PA0
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
[... 37 lines cut ...]
avrdude: load data flash data from input file C:\Users\mudas\OneDrive\Documents\Arduino\hardware\MegaCore-master\avr/bootloaders/optiboot_flash/atmega2560/optiboot_flash_atmega2560_38400_8000000L.hex:
avrdude: input file C:\Users\mudas\OneDrive\Documents\Arduino\hardware\MegaCore-master\avr/bootloaders/optiboot_flash/atmega2560/optiboot_flash_atmega2560_38400_8000000L.hex contains 262144 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 0.00s

avrdude: verifying ...
avrdude: 262144 bytes of flash verified
avrdude: reading input file "0x0f"
avrdude: writing lock (1 bytes):

Writing | ################################################## | 100% 0.02s

avrdude: 1 bytes of lock written
avrdude: verifying lock memory against 0x0f:
avrdude: load data lock data from input file 0x0f:
avrdude: input file 0x0f contains 1 bytes
avrdude: reading on-chip lock data:

Reading | ################################################## | 100% 0.01s

avrdude: verifying ...
avrdude: 1 bytes of lock verified

avrdude done.  Thank you.

MCUdude, 2017-08-11:
Seems like everything went fine? Does it work now? You can try attaching an LED to pin PB5. If the bootloader is running, it will flash twice every second.

per1234, 2017-08-11:
```
C:\Program Files (x86)\Arduino\hardware\tools\avr\bin\avrdude: unknown option -- "
```
I think this warning is because you don't have a space between `-F` and the " in this command:
```
C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avrdude -F"-CC:\Program Files (x86)\Arduino\hardware\tools\avr/etc/avrdude.conf" -v -patmega2560 -cstk500v1 -PCOM4 -b19200 -e -Ulock:w:0x3f:m -Uefuse:w:0xfd:m -Uhfuse:w:0xd6:m -Ulfuse:w:0xf7:m
```

beingmachine, 2017-08-12:
Hello, 
I have override using -F and burnt bootloaders to it using settings external 8Mhz Crystal, then i soldered 8Mhz crystal - upload blink sketch and it works.. and the Device signature is back to 0x1e9801 

Thanks a lot, thanks a TON! :)

closed 2017-08-12 by beingmachine