MCUdude/MegaCore #64: 1 Byte of Lock verified..

beingmachine, 2017-08-13:
Hello sorry to disturb you again, I was trying to boot load my another Atmega2560V, supplied 1Mhz to the XTAL1 and used -F to override the signature check, Now I can upload bootloaders to the chip.

But the output shows 1Byte of Lock verified.

Here is the complete output.

> C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avrdude -F -CC:\Users\mudas\OneDrive\Documents\Arduino\hardware\MegaCore-master\avr/avrdude.conf -v -patmega2560 -cstk500v1 -PCOM4 -b19200 -e -Ulock:w:0x3f:m -Uefuse:w:0xfd:m -Uhfuse:w:0xd6:m -Ulfuse:w:0xe2:m 

avrdude: Version 6.3, compiled on Jan 17 2017 at 12:00:53
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Users\mudas\OneDrive\Documents\Arduino\hardware\MegaCore-master\avr/avrdude.conf"

         Using Port                    : COM4
         Using Programmer              : stk500v1
         Overriding Baud Rate          : 19200
         AVR Part                      : ATmega2560
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PA0
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    10     8    0 no       4096    8      0  9000  9000 0x00 0x00
           flash         65    10   256    0 yes    262144  256   1024  4500  4500 0x00 0x00
           lfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           lock           0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : STK500
         Description     : Atmel STK500 Version 1.x firmware
         Hardware Version: 2
         Firmware Version: 1.18
         Topcard         : Unknown
         Vtarget         : 0.0 V
         Varef           : 0.0 V
         Oscillator      : Off
         SCK period      : 0.1 us

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.02s

avrdude: Device signature = 0x1e9803
avrdude: Expected signature for ATmega2560 is 1E 98 01
avrdude: erasing chip
avrdude: reading input file "0x3f"
avrdude: writing lock (1 bytes):

Writing | ################################################## | 100% 0.01s

avrdude: 1 bytes of lock written
avrdude: verifying lock memory against 0x3f:
[... 127 lines cut ...]
avrdude: verifying ...
avrdude: 262144 bytes of flash verified
avrdude: reading input file "0x0f"
avrdude: writing lock (1 bytes):

Writing | ################################################## | 100% 0.02s

avrdude: 1 bytes of lock written
avrdude: verifying lock memory against 0x0f:
avrdude: load data lock data from input file 0x0f:
avrdude: input file 0x0f contains 1 bytes
avrdude: reading on-chip lock data:

Reading | ################################################## | 100% 0.01s

avrdude: verifying ...
avrdude: 1 bytes of lock verified

avrdude done.  Thank you.

I am really sorry to disturb you again.

I hope you reply.

Thank you

MCUdude, 2017-08-13:
You're able to upload and run your program to the new ATmega2560V? Lockbits are there to prevent someone from dumping/"stealing" the content of the microcontroller. MegaCore automatically load the lockouts when burning the bootloader

beingmachine, 2017-08-13:
No I am unable to upload sketches, I can only burn bootloaders to the chip using -F or changing the signature to 03 at the end in avrdude..

_Sent from my Xiaomi Redmi Note 4 using [FastHub](https://play.google.com/store/apps/details?id=com.fastaccess.github)_

beingmachine, 2017-08-14:
I can upload sketches using "Upload using Programmer" by holding down the shift key and uploading. I have tested by uploading Blink sketch example and tested the pin with DMM, it's blinking 2 times per second.

when i upload it directly it shows me this error.

> Arduino: 1.8.3 (Windows 10), Board: "ATmega2560, Arduino MEGA pinout, 2.7v, Disabled (default), 8 MHz external"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\mudas\OneDrive\Documents\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\mudas\OneDrive\Documents\Arduino\libraries -fqbn=MegaCore-master:avr:2560:pinout=mega_pinout,BOD=2v7,LTO=Os,clock=8MHz_external -vid-pid=0X2341_0X0043 -ide-version=10803 -build-path C:\Users\mudas\AppData\Local\Temp\arduino_build_664594 -warnings=none -build-cache C:\Users\mudas\AppData\Local\Temp\arduino_cache_254657 -prefs=build.warn_data_percentage=75 -verbose C:\Users\mudas\AppData\Local\Temp\arduino_modified_sketch_764049\Blink.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\mudas\OneDrive\Documents\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\mudas\OneDrive\Documents\Arduino\libraries -fqbn=MegaCore-master:avr:2560:pinout=mega_pinout,BOD=2v7,LTO=Os,clock=8MHz_external -vid-pid=0X2341_0X0043 -ide-version=10803 -build-path C:\Users\mudas\AppData\Local\Temp\arduino_build_664594 -warnings=none -build-cache C:\Users\mudas\AppData\Local\Temp\arduino_cache_254657 -prefs=build.warn_data_percentage=75 -verbose C:\Users\mudas\AppData\Local\Temp\arduino_modified_sketch_764049\Blink.ino
Using board '2560' from platform in folder: C:\Users\mudas\OneDrive\Documents\Arduino\hardware\MegaCore-master\avr
Using core 'MCUdude_corefiles' from platform in folder: C:\Users\mudas\OneDrive\Documents\Arduino\hardware\MegaCore-master\avr
Warning: Board 3.3V_Pro:avr:mega2560v doesn't define a 'build.board' preference. Auto-set to: AVR_MEGA2560V
Detecting libraries used...
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=8000000L -DARDUINO=10803 -DARDUINO_AVR_ATmega2560 -DARDUINO_ARCH_AVR   "-IC:\Users\mudas\OneDrive\Documents\Arduino\hardware\MegaCore-master\avr\cores\MCUdude_corefiles" "-IC:\Users\mudas\OneDrive\Documents\Arduino\hardware\MegaCore-master\avr\variants\100-pin-arduino-mega" "C:\Users\mudas\AppData\Local\Temp\arduino_build_664594\sketch\Blink.ino.cpp" -o "nul"
Generating function prototypes...
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=8000000L -DARDUINO=10803 -DARDUINO_AVR_ATmega2560 -DARDUINO_ARCH_AVR   "-IC:\Users\mudas\OneDrive\Documents\Arduino\hardware\MegaCore-master\avr\cores\MCUdude_corefiles" "-IC:\Users\mudas\OneDrive\Documents\Arduino\hardware\MegaCore-master\avr\variants\100-pin-arduino-mega" "C:\Users\mudas\AppData\Local\Temp\arduino_build_664594\sketch\Blink.ino.cpp" -o "C:\Users\mudas\AppData\Local\Temp\arduino_build_664594\preproc\ctags_target_for_gcc_minus_e.cpp"
"C:\Program Files (x86)\Arduino\tools-builder\ctags\5.8-arduino11/ctags" -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives "C:\Users\mudas\AppData\Local\Temp\arduino_build_664594\preproc\ctags_target_for_gcc_minus_e.cpp"
Compiling sketch...
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega2560 -DF_CPU=8000000L -DARDUINO=10803 -DARDUINO_AVR_ATmega2560 -DARDUINO_ARCH_AVR   "-IC:\Users\mudas\OneDrive\Documents\Arduino\hardware\MegaCore-master\avr\cores\MCUdude_corefiles" "-IC:\Users\mudas\OneDrive\Documents\Arduino\hardware\MegaCore-master\avr\variants\100-pin-arduino-mega" "C:\Users\mudas\AppData\Local\Temp\arduino_build_664594\sketch\Blink.ino.cpp" -o "C:\Users\mudas\AppData\Local\Temp\arduino_build_664594\sketch\Blink.ino.cpp.o"
Compiling libraries...
Compiling core...
Using precompiled core
Linking everything together...
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-gcc" -w -Os -Wl,--gc-sections,--relax -mmcu=atmega2560  -o "C:\Users\mudas\AppData\Local\Temp\arduino_build_664594/Blink.ino.elf" "C:\Users\mudas\AppData\Local\Temp\arduino_build_664594\sketch\Blink.ino.cpp.o" "C:\Users\mudas\AppData\Local\Temp\arduino_build_664594/..\arduino_cache_254657\core\core_MegaCore-master_avr_2560_pinout_mega_pinout,BOD_2v7,LTO_Os,clock_8MHz_external_9d0b093b0b3f80767f466e624dd47b34.a" "-LC:\Users\mudas\AppData\Local\Temp\arduino_build_664594" -lm
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-objcopy" -O ihex -j .eeprom --set-section-flags=.eeprom=alloc,load --no-change-warnings --change-section-lma .eeprom=0  "C:\Users\mudas\AppData\Local\Temp\arduino_build_664594/Blink.ino.elf" "C:\Users\mudas\AppData\Local\Temp\arduino_build_664594/Blin [... cut ...]
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 9 of 10: not in sync: resp=0xc9
avrdude: ser_send(): write error: sorry no info avail
avrdude: ser_recv(): read error: The device does not recognize the command.

avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 10 of 10: not in sync: resp=0xc9
avrdude: ser_drain(): read error: The device does not recognize the command.

avrdude done.  Thank you.

the selected serial port 
 does not exist or your board is not connected

per1234, 2017-08-14:
Keep in mind that when you do an "Upload Using Programmer" the bootloader is erased so after doing that you need to burn the bootloader again before you can do a standard serial upload via USB.

beingmachine, 2017-08-14:
Hello Perth1234,

Yes, I know that - I have flashed bootloaders to my chip and then trying to upload code using Hardware serial pins  PE0 and PE1..

But the Device Signature is still at 0x1e9803, It didn't came up to 0x1e9801 like happened to my previous chip :/

beingmachine, 2017-08-14:
I have accleromter sensor attached to the target ATmega2560V , i uploaded test sketch using "Upload as a programmer" when i see Serial Monitor i see this :

> x2>⸮⸮⸮X ,⸮⸮⸮⸮⸮۲⸮⸮⸮۲⸮⸮⸮۲⸮⸮⸮۲⸮I=⸮_.QQQC ⸮yIX⸮⸮T⸮*QӚ⸮>8 ⸮⸮⸮⸮⸮⸮T0⸮aQѦ⸮⸮S⸮S*⸮⸮⸮ 8*Q⸮WQ⸮⸮w⸮Oʚ⸮q⸮⸮QQ]U⸮⸮*⸮⸮⸮⸮߶⸮⸮۲⸮⸮⸮߶⸮⸮۶⸮⸮O⸮⸮8<⸮ YQQSQQ⸮I9⸮⸮QQQQѺ⸮TQךI⸮Q⸮S⸮W⸮QIQ]YѦ⸮њ⸮Q⸮y⸮S⸮AIQ]U⸮w⸮њ⸮Q⸮y⸮.W⸮QIQ]Qѿ⸮K⸮⸮Q]y⸮S⸮QIQ]Yѻ⸮O⸮⸮Q⸮m⸮.WSQIQ]Qѿ⸮K⸮⸮Q⸮m⸮.W⸮QMQ]Uѻ⸮O⸮⸮Q⸮}⸮.WQEIQ]YѦ⸮њ⸮Q⸮i⸮S⸮QIQ]YѦ⸮њ⸮Q⸮i⸮S⸮QIQ⸮Uѻ⸮O⸮⸮Q⸮W⸮W⸮QMQ⸮Uѿ⸮K⸮⸮Q⸮i⸮S⸮QIQ]Y⸮w⸮њ⸮Q⸮m⸮.WSEIQ]Yѻ⸮O⸮⸮Q⸮}⸮.WSQIQ]Yѿ⸮O⸮⸮Q⸮y⸮S⸮AIQ]YѦ⸮њ⸮Q⸮i⸮S⸮QMQ]Yѿ⸮K⸮⸮Q⸮i⸮.W⸮QMQ]Uѻ⸮O⸮⸮Q⸮y⸮WSQIQ⸮Uѿ⸮O⸮⸮Q]}⸮W⸮QIQ⸮Uѿ⸮O⸮⸮Q⸮S⸮W⸮QMQ]Yѿ⸮O⸮⸮Q]y⸮S⸮QIQ⸮Q⸮w⸮њ⸮Q]W⸮.WSQIQ]Qѿ⸮K⸮⸮Q]W⸮S⸮QIQ]Yѻ⸮O⸮⸮Q⸮m⸮.WSEIQ]Yѿ⸮O⸮⸮Q]W⸮S⸮QIQ]Qѻ⸮O⸮⸮Q⸮}⸮.WQEIQ⸮UѦ⸮њ⸮Q⸮}⸮.WSQIQ]Qѿ⸮O⸮⸮Q⸮}⸮.WQEIQ]Yѿ⸮O⸮⸮Q⸮}⸮.WSEIQ⸮Qѻ⸮O⸮⸮Q⸮m⸮W⸮QMQ⸮Qѻ⸮O⸮⸮Q]y⸮.W⸮QIQ]YѦ⸮њ⸮Q⸮S⸮W⸮QMQ]Yѿ⸮K⸮⸮Q⸮m⸮.W⸮QIQ]Yѿ⸮K⸮⸮Q⸮S⸮S⸮QIQ]QѦ⸮њ⸮Q⸮m⸮.WSQIQ⸮Uѿ⸮O⸮⸮Q⸮i⸮S⸮QIQ⸮Qѻ⸮O⸮⸮Q⸮m⸮W⸮QMQ⸮Qѻ⸮O⸮⸮Q]}⸮.WQEIQ⸮Q⸮w⸮њ⸮Q⸮W⸮W⸮QMQ⸮Uѿ⸮O⸮⸮Q⸮S⸮W⸮QMQ]Yѿ⸮O⸮⸮Q⸮S⸮W⸮AIQ]Yѻ⸮O⸮⸮Q⸮i⸮WSEIQ]Qѿ⸮K⸮⸮Q⸮y⸮S⸮QIQ]Uѻ⸮O⸮⸮Q⸮i⸮WSQMQ⸮Uѿ⸮O⸮⸮Q⸮i⸮W⸮AMQ]Yѿ⸮K⸮⸮Q⸮i⸮S⸮QMQ]Yѿ⸮K⸮⸮Q]y⸮S⸮QIQ]Uѻ⸮O⸮⸮Q⸮y⸮S⸮AIQ]Yѻ⸮O⸮⸮Q⸮i⸮S⸮QIQ⸮Qѻ⸮O⸮⸮Q⸮m⸮.W⸮QMQ]Qѿ⸮O⸮⸮Q⸮i⸮S⸮AIQ]Yѿ⸮K⸮⸮Q⸮m⸮.WSQIQ]Uѻ⸮O⸮⸮Q]}⸮.WSEIQ]Uѿ⸮O⸮⸮Q]W⸮W⸮QMQ⸮Qѿ⸮O⸮⸮Q⸮[⸮W⸮AIQ⸮Qѻ⸮O⸮⸮Q⸮W⸮W⸮QMQ⸮UѦ⸮њ⸮Q]y⸮W⸮QIQ]Q⸮w⸮њ⸮Q⸮i⸮W⸮QMQ]Yѿ⸮K⸮⸮Q⸮m⸮WSQIQ]Uѻ⸮O⸮⸮Q]y⸮S⸮QIQ]Yѻ⸮O⸮⸮Q⸮}⸮.WQEIQ⸮Qѻ⸮O⸮⸮Q⸮y⸮S⸮QIQ⸮Qѻ⸮O⸮⸮Q⸮y⸮W⸮QMQ]YѦ⸮њ⸮Q⸮W⸮W⸮QMQ]Yѿ⸮K⸮⸮Q⸮m⸮.WSQIQ⸮Qѻ⸮O⸮⸮Q⸮i⸮W⸮QIQ⸮Uѻ⸮O⸮⸮Q⸮S⸮W⸮AMQ⸮[Ѧ⸮њ⸮Q⸮i⸮W⸮QMQ]YѦ⸮њ⸮Q⸮i⸮S⸮QIQ]Uѿ⸮K⸮⸮Q⸮y⸮WSEIQ]Q⸮w⸮њ⸮Q]W⸮S⸮QMQ]Qѻ⸮O⸮⸮Q⸮i⸮S⸮EIQ]QѦ⸮њ⸮Q⸮i⸮SSEIQ⸮Uѿ⸮K⸮⸮Q⸮y⸮WSQIQ]Yѻ⸮O⸮⸮Q]y⸮WQEIQ]Yѻ⸮O⸮⸮Q⸮i⸮WSEIQ]Uѻ⸮O⸮⸮Q⸮m⸮.WSEIQ]Q⸮s⸮њ⸮Q⸮i⸮W⸮QMQ]YѦ⸮њ⸮Q⸮W⸮W⸮AMQ⸮Qѿ⸮O⸮⸮Q⸮i⸮WSQIQ]Y⸮s⸮њ⸮Q⸮y⸮S⸮QIQ⸮Qѻ⸮O⸮⸮Q⸮y⸮S⸮QIQ⸮U⸮w⸮њ⸮Q⸮i⸮W⸮AIQ⸮Qѻ⸮O⸮⸮Q⸮m⸮.W⸮QIQ]Yѿ⸮O⸮⸮Q⸮i⸮S⸮QIQ⸮Uѿ⸮O⸮⸮Q]y⸮S⸮AIQ]Yѻ⸮O⸮⸮Q⸮m⸮.WQEIQ⸮Uѻ⸮O⸮⸮Q]y⸮W⸮AMQ⸮Qѿ⸮K⸮⸮Q⸮y⸮WQEIQ]Q⸮s⸮њ⸮Q⸮}⸮.W⸮QMQ]Qѿ⸮K⸮⸮Q⸮S⸮W⸮QMQ]Yѿ⸮K⸮⸮Q⸮i⸮S⸮AMQ]Y⸮w⸮њ⸮Q⸮S⸮W⸮QIQ⸮UѦ⸮њ⸮Q⸮y⸮WSQIQ]Yѻ⸮O⸮⸮Q]y⸮S⸮QIQ]Uѻ⸮O⸮⸮Q⸮S⸮W⸮QIQ]Uѻ⸮O⸮⸮Q]y⸮S⸮AMQ⸮U⸮w⸮њ⸮Q⸮W⸮*W⸮QIQ]Y⸮w⸮њ⸮Q⸮y⸮.WSQIQ]Qѿ⸮K⸮⸮Q⸮m⸮*W⸮QMQ]YѦ⸮њ⸮Q]}⸮.W⸮QIQ]Yѿ⸮O⸮⸮Q⸮S⸮W⸮QMQ⸮Qѿ⸮O⸮⸮Q⸮S⸮W⸮QMQ⸮Qѿ⸮O⸮⸮Q⸮i⸮S⸮AMQ⸮Qѿ⸮O⸮⸮Q⸮W⸮.WSQMQ⸮Qѿ⸮O⸮⸮Q⸮S⸮S⸮AIQ]Q⸮s⸮њ⸮Q]y⸮S⸮AIQ⸮Qѻ⸮O⸮⸮Q⸮}⸮.W⸮QIQ⸮Uѿ⸮K⸮⸮Q]}⸮W⸮QMQ]QѦ⸮њ⸮Q⸮m⸮.WQEIQ]Yѿ⸮K⸮⸮Q⸮S⸮WSQIQ]Yѻ⸮O⸮⸮Q⸮S⸮.WQEIQ⸮Qѻ⸮O⸮⸮Q⸮i⸮S⸮AIQ]Uѻ⸮O⸮⸮Q]S⸮W⸮QMQ]Uѿ⸮K⸮⸮Q]y⸮S⸮QMQ]Yѻ⸮O⸮⸮Q⸮}⸮WSQIQ]Y⸮w⸮њ⸮Q⸮i⸮S⸮AMQ]Yѿ⸮O⸮⸮Q]}⸮S⸮AIQ]Qѻ⸮O⸮⸮Q⸮W⸮.WSQMQ]Yѿ⸮K⸮⸮Q⸮m⸮WSEIQ⸮UѦ⸮њ⸮Q]m⸮.WSEIQ⸮[ѻ⸮O⸮⸮Q]}⸮.WSQIQ]Yѻ⸮O⸮⸮Q⸮W⸮W⸮QMQ⸮UѦ⸮њ⸮Q⸮}⸮.WSQIQ⸮UѦ⸮њ⸮Q]y⸮WSQIQ]Y⸮s⸮њ⸮Q]}⸮.WQEIQ]Yѻ⸮O⸮⸮Q⸮W⸮W⸮QIQ]Y⸮w⸮њ⸮Q⸮S⸮W⸮QIQ]Yѻ⸮O⸮⸮Q]y⸮S⸮QMQ]YѦ⸮њ⸮Q]y⸮WSEIQ]Qѿ⸮K⸮⸮Q]W⸮W⸮QIQ]Q⸮w⸮њ⸮Q⸮i⸮S⸮QMQ]QѦ⸮њ⸮Q]}⸮*W⸮QIQ]QѦ⸮њ⸮Q]W⸮W⸮QMQ⸮Uѿ⸮K⸮⸮Q]W⸮W⸮QMQ]Uѿ⸮O⸮⸮Q⸮i⸮S⸮AIQ⸮Qѿ⸮O⸮⸮Q]W⸮W⸮QIQ]YѦ⸮њ⸮Q⸮y⸮S⸮AMQ⸮Qѻ⸮O⸮⸮Q⸮S⸮WSQIQ⸮Qѻ⸮O⸮⸮Q⸮i⸮S⸮QIQ]Yѻ⸮O⸮⸮Q]}⸮WSAIQ]Yѿ⸮O⸮⸮Q⸮S⸮W⸮QMQ⸮Qѿ⸮K⸮⸮Q⸮S⸮*W⸮QIQ]YѦ⸮њ⸮Q⸮S⸮W⸮QMQ]Yѿ⸮O⸮⸮Q⸮S⸮W⸮QIQ]Uѻ⸮O⸮⸮Q⸮y⸮S⸮QMQ]Yѿ⸮O⸮⸮Q]}⸮.WSEIQ⸮Qѻ⸮O⸮⸮Q]}⸮W⸮QMQ]Yѿ⸮K⸮⸮Q⸮m⸮.WQEIQ⸮Qѿ⸮O⸮⸮Q⸮m⸮.WSQIQ⸮Q⸮s⸮њ⸮Q⸮S⸮S⸮QMQ]Yѻ⸮O⸮⸮Q⸮m⸮WSQIQ]Yѻ⸮O⸮⸮Q⸮i⸮S⸮QIQ⸮Qѻ⸮O⸮⸮Q⸮S⸮.WQQIQ]Uѿ⸮O⸮⸮Q⸮y⸮S⸮QIQ]Yѻ⸮O⸮⸮Q⸮m⸮.WSQIQ]Q⸮w⸮њ⸮Q⸮m⸮W⸮QMQ⸮Uѿ⸮O⸮⸮Q⸮i⸮S⸮QIQ]Yѿ⸮K⸮⸮Q⸮S⸮W⸮QIQ⸮UѦ⸮њ⸮Q]W⸮W⸮QMQ⸮UѦ⸮њ⸮Q⸮i⸮S⸮AIQ⸮Qѻ⸮O⸮⸮Q⸮S⸮*W⸮QIQ]Q⸮w⸮њ⸮Q⸮m⸮WQEIQ⸮Uѿ⸮K⸮⸮Q⸮i⸮SSEIQ⸮Uѻ⸮O⸮⸮Q⸮}⸮.WSQIQ⸮Qѻ⸮O⸮⸮Q⸮m⸮WSQIQ]Qѿ⸮K⸮⸮Q]}⸮.WSQIQ]Yѿ⸮O⸮⸮Q]}⸮.WSQIQ⸮Qѿ⸮K⸮⸮Q]i⸮W⸮QIQ]Uѻ⸮O⸮⸮Q⸮}⸮.WQEIQ]Yѿ⸮K⸮⸮Q⸮i⸮S⸮QMQ]Yѻ⸮O⸮⸮Q]y⸮S⸮AIQ]Uѿ⸮K⸮⸮Q⸮i⸮S⸮QIQ]QѦ⸮њ⸮Q]y⸮S⸮QMQ]Yѿ⸮K⸮⸮Q⸮W⸮WSQIQ]YѦ⸮њ⸮Q]y⸮S⸮QMQ]Uѿ⸮K⸮⸮Q⸮m⸮*W⸮QMQ]Yѿ⸮O⸮⸮Q⸮i⸮S⸮QMQ⸮Qѿ⸮K⸮⸮Q⸮S⸮W⸮QMQ⸮U⸮w⸮њ⸮Q⸮i⸮S⸮QIQ⸮Qѿ⸮K⸮⸮Q⸮y⸮S⸮QIQ]Yѻ⸮O⸮⸮Q⸮m⸮.WSEIQ⸮Qѿ⸮K⸮⸮Q⸮}⸮S⸮AIQ⸮Qѿ⸮O⸮⸮Q]W⸮W⸮QIQ]QѦ⸮њ⸮Q⸮i⸮S⸮QIQ]YѦ⸮њ⸮Q⸮i⸮S⸮QIQ]Yѻ⸮O⸮⸮Q⸮y⸮S⸮AIQ⸮Qѻ⸮O⸮⸮Q⸮i⸮W⸮QMQ]YѦ⸮њ⸮Q⸮W⸮W⸮QMQ]Y⸮w⸮њ⸮Q⸮y⸮S⸮QIQ]Yѿ⸮K⸮⸮Q]W⸮W⸮QMQ]Yѿ⸮K⸮⸮Q]}⸮.W⸮QIQ]Yѿ⸮K⸮⸮Q⸮i⸮.WSEIQ⸮Uѿ⸮O⸮⸮Q⸮i⸮S⸮QIQ⸮Qѿ⸮O⸮⸮Q]W⸮W⸮QIQ⸮U⸮s⸮њ⸮Q⸮m⸮.WQEIQ]Q⸮s⸮њ⸮Q⸮}⸮.WQEIQ]Qѿ⸮O⸮⸮Q⸮y⸮WSQIQ⸮Uѿ⸮K⸮⸮Q⸮m⸮W⸮AMQ⸮Qѿ⸮K⸮⸮Q⸮i⸮WSQIQ⸮Uѿ⸮K⸮⸮Q⸮i⸮S⸮QIQ]YѦ⸮њ⸮Q⸮i⸮W⸮QIQ⸮Qѻ⸮O⸮⸮Q⸮S⸮S⸮QIQ]Y⸮w⸮њ⸮Q⸮i⸮S⸮EIQ]QѦ⸮њ⸮Q⸮W⸮W⸮QMQ⸮Qѻ⸮O⸮⸮Q⸮y⸮WQQMQ⸮Qѿ⸮O⸮⸮Q⸮S⸮S⸮QIQ]Uѻ⸮O⸮⸮Q⸮y⸮S⸮EIQ⸮Q⸮w⸮њ⸮Q⸮W⸮S⸮QIQ⸮Qѻ⸮O⸮⸮Q⸮m⸮W⸮QMQ⸮Qѻ⸮O⸮⸮Q]y⸮S⸮QIQ⸮Qѻ⸮O⸮⸮Q⸮m⸮.WSQIQ⸮Uѻ⸮O⸮⸮Q⸮i⸮WSQIQ]Uѿ⸮O⸮⸮Q⸮i⸮WSEIQ]Yѿ⸮K⸮⸮Q]}⸮.W⸮QIQ⸮Uѿ⸮O⸮⸮Q⸮}⸮.W⸮QMQ⸮Qѻ⸮K⸮⸮Q⸮m⸮WSQIQ]Yѿ⸮K⸮⸮Q]W⸮W⸮QIQ]QѦ⸮њ⸮Q⸮W⸮S⸮QIQ]Yѿ⸮O⸮⸮Q⸮S⸮S⸮QIQ]Yѿ⸮O⸮⸮Q]S⸮Q⸮AI⸮U⸮⸮⸮K⸮⸮Q⸮i⸮S⸮QIQ]Wѻ⸮O⸮⸮Q⸮m⸮U⸮QM⸮QSѿ⸮K⸮⸮Q⸮S⸮U⸮⸮⸮*Y⸮E⸮⸮K⸮⸮Q]W⸮Q⸮QIQ⸮[ѻ⸮O⸮⸮Q]m⸮*U⸮QM⸮⸮Yѻ⸮O⸮⸮Q⸮U⸮.S⸮⸮.Y⸮Q⸮⸮*O⸮⸮

beingmachine, 2017-08-14:
it solved, it was with my crystal.

MCUdude, 2017-08-14:
Great!

closed 2017-08-14 by MCUdude