MCUdude/MegaCore #66: For Atmeg128 the Etnernet2 library does not work The controller in your examples has found a modified Etnernet library, there are changes in the initialization of the SPI. How to implement for Etnernet2?

Xekz, 2017-08-16:


MCUdude, 2017-08-23:
I'm not the author of the Ethernet2 library, so I can't help you there. What's wrong? Does it compile; are there any errors you can post?

Xekz, 2017-08-24:
Errors are not loaded normally, but atmega128 w5500 is not initialized. By connecting to the Arduino nano everything works. In the examples for atmega128 I saw the library of Etnernet and there are differences from the original.

Xekz, 2017-09-21:
what pins to connect the ethernet? in the library of examples provided pins 10, 11, 12, 13 and the picture https://camo.githubusercontent.com/02359aa02f9ff86bba528efac949864febf50d2a/687474703a2f2f692e696d6775722e636f6d2f657a354d7976492e6a7067 other SPI pins 8,9,10,11?

MCUdude, 2017-09-21:
You just connect the ethernet module to the SPI pins on the ATmega128. that should be MISO, MOSI and SCK. However you can't just expect the Ethernet2 library to work with ATmega128 without doing any modifications to the library. Ethernet2 is not designed for ATmega128

Xekz, 2017-09-21:
thank you have dealt with the modification of the library

closed 2017-09-21 by Xekz