MCUdude/MegaCore #67: Error on compiling bootloader with -DBIGBOOT

vanbwodonk, 2017-08-16:
make atmega128
avr-gcc (GCC) 4.9.2
Copyright (C) 2014 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

BAUD RATE CHECK: Desired: 115200, Real: 117647, UBRRL = 16, Error=2.1%
avr-gcc -g -Wall -Os -fno-split-wide-types -mrelax -mmcu=atmega128 -DF_CPU=16000000L  -DBAUD_RATE=115200 -DLED_START_FLASHES=3      -DBIGBOOT 	   -c -o optiboot_flash.o optiboot_flash.c
optiboot_flash.c:311:2: warning: #warning BAUD_RATE error greater than 2% [-Wcpp]
 #warning BAUD_RATE error greater than 2%
  ^
optiboot_flash.c: In function ‘main’:
optiboot_flash.c:502:93: warning: backslash and newline separated by space
 #if defined(__AVR_ATmega8515__) || defined(__AVR_ATmega8535__) || defined(__AVR_ATmega16__) \ 
 ^
avr-gcc -g -Wall -Os -fno-split-wide-types -mrelax -mmcu=atmega128 -DF_CPU=16000000L  -DBAUD_RATE=115200 -DLED_START_FLASHES=3      -DBIGBOOT 	 -Wl,--section-start=.text=0x1fc00 -Wl,--section-start=.version=0x1fffe -Wl,--relax -nostartfiles -nostdlib -o atmega128/optiboot_flash_atmega128__.elf optiboot_flash.o -lc
optiboot_flash.o: In function `appStart':
/home/drw/Software/builder/MegaCore/avr/bootloaders/optiboot_flash/optiboot_flash.c:942: undefined reference to `eeprom_write_byte'
/home/drw/Software/builder/MegaCore/avr/bootloaders/optiboot_flash/optiboot_flash.c:942: undefined reference to `eeprom_read_byte'
collect2: error: ld returned 1 exit status
Makefile:614: recipe for target 'atmega128/optiboot_flash_atmega128__.elf' failed
make: *** [atmega128/optiboot_flash_atmega128__.elf] Error 1

vanbwodonk, 2017-08-16:
This issue related into optiboot issue, https://github.com/Optiboot/optiboot/issues/179

vanbwodonk, 2017-08-16:
commit PR #68 to fix this issue

MCUdude, 2017-08-16:
Thanks! So this error doesn't apply for smaller AVRs such as ATmega328 with BIGBOOT enabled?
the optiboot_flash bootloader is actually hosted on [another repo](https://github.com/MCUdude/optiboot_flash), so you'll have to submit your PR there.

vanbwodonk, 2017-08-16:
> So this error doesn't apply for smaller AVRs such as ATmega328 with BIGBOOT enabled?

This will be produce same error. If BIGBOOT need to enabled, we need add `LIBS` into all avr family.

vanbwodonk, 2017-08-16:
update PR, add `LIBS` to all....

MCUdude, 2018-02-17:
Already applied to the optiboot_flash repo 👍

closed 2018-02-17 by MCUdude