MCUdude/MegaCore #69: Old ATmega128 board blinky doesn't work on arduino compile

flodejr, 2017-08-17:
Hi,

I have this weird problem, when i compile a simple  blinky program using Arduino and Megacore Atmega128.  I can't get the LED to blink at all, however, a simple C program compiled in avr-gcc will make it blink.  Here is my Arduino code.

// the setup function runs once when you press reset or power the board
void setup() {
  // initialize digital pin LED_BUILTIN as an output.
  pinMode(PORTB0, OUTPUT);
}

// the loop function runs over and over again forever
void loop() {
  digitalWrite(PORTB0, HIGH);   // turn the LED on (HIGH is the voltage level)
  delay(1000);                       // wait for a second
  digitalWrite(PORTB0, LOW);    // turn the LED off by making the voltage LOW
  delay(1000);                       // wait for a second
}

and here is my c code.

/*
 * $Id: hw.c,v 1.1 2010/10/19 05:18:47 bsd Exp $
 *
 */

/*
 * HW - Hello World
 *
 * This is a very simple project designed only to familiarize oneself
 * with the use of a Makefile, the compiler, linker, and FLASH
 * download tool (such as AVRDUDE) when working with the MAVRIC /
 * MAVRIC-II board.
 *
 * This program simply blinks the on-board LED on and off, once per
 * second.  It demonstrates setting up a periodic timer interrupt as a
 * time keeping source, the use of declaring and using an interrupt
 * handler with AVR-GCC, and toggling a PORT output line.
 *
 * See: http://www.bdmicro.com/
 *
 * Configuration:
 *
 *   MAVRIC - ensure that DIPSW 9 is "on".
 *
 *   MAVRIC-II - ensure that jumper LEDEN is installed
 *
 *  */

#include <avr/io.h>
#include <avr/interrupt.h>

#include <inttypes.h>

volatile uint16_t ms_count;

/*
 * ms_sleep() - delay for specified number of milliseconds
 */
void ms_sleep(uint16_t ms)
{
  TCNT0  = 0;
  ms_count = 0;
  while (ms_count != ms)
    ;
}

/* 
 * millisecond counter interrupt vector 
 */
SIGNAL(TIMER0_COMP_vect)
{
  ms_count++;
}

/*
 * initialize timer 0 to generate an interrupt every millisecond.
 */
void init_timer(void)
{
  /*
   * Initialize timer0 to generate an output compare interrupt, and
   * set the output compare register so that we get that interrupt
   * every millisecond.
   */
  TIFR  |= _BV(OCIE0);
  TCCR0  = _BV(WGM01)|_BV(CS02)|_BV(CS00); /* CTC, prescale = 128 */
  TCNT0  = 0;
  TIMSK |= _BV(OCIE0);    /* enable output compare interrupt */
  OCR0   = 125;          /* match in 1 ms */
}

int main(void)
{
  init_timer();

  /* enable interrupts */
  sei();

  DDRB = 0x01;  /* enable PORTB 1 as an output */

  while (1) {
    ms_sleep(512);    /* wait 0.5 seconds */
    PORTB ^= 0x01;    /* toggle LED */
  }

}

What do I need to change in the MegaCore for this old board that will compile the code and execute the code correctly?

I am using avrdude + usbasp to upload the code in both instances.

MCUdude, 2017-08-17:
How about replacing PORTB0 with the correct digital pin instead? PB0 is digital pin 0

flodejr, 2017-08-17:
Thanks, using digitalwrite(8,HIGH) in this case works (PB0 is D8).  However, it is strange when I specify the exact port and pin and it doesn't work?

MCUdude, 2017-08-17:
Sorry, I mixed the pinout for MightyCore and MegaCore. It's correct that PB0 is digital pin 8. However, you can't use plain avr-libc macros directly with pinMode, digitalRead and digitalWrite; It's not supported by the [Arduino framework](https://www.arduino.cc/en/Reference/HomePage).

closed 2017-08-17 by MCUdude