MCUdude/MegaCore #71: Verification error after loading code by Serial

Borsoft73, 2017-08-27:
Hi! I'm building my project on Atmega128 chip using your bootloader. Bootloader was load properly by USBASP and I even can see it in FLASH through Avrdude terminal mode from address 0xFC00 (as I remember). Then I faced an issue during loading my program by Serial through USB-UART (CH340 chip). It seems that Arduino IDE connects to Atmega and I see Avrdude writing progress bars but then it fails on verification step:

>  avrdude: verification error, first mismatch at byte 0x0000
>          0xff != 0x0c
> avrdude: verification error; content mismatch

It seems that my sketch code was not recorded to Atmega... If I loading the code through programmer by ICSP - it works well.
Here is the Avrdude log:

> /Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avrdude -C/Users/dmitrij/Library/Arduino15/packages/MegaCore/hardware/avr/2.0.0/avrdude.conf -v -patmega128 -carduino -P/dev/cu.wchusbserial1420 -b115200 -D -Uflash:w:/var/folders/18/60_7xyc92138x6vhc69t25zc0000gn/T/arduino_build_113404/Blink.ino.hex:i 
> 
> avrdude: Version 6.3, compiled on Jan 17 2017 at 12:01:35
>          Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
>          Copyright (c) 2007-2014 Joerg Wunsch
> 
>          System wide configuration file is "/Users/dmitrij/Library/Arduino15/packages/MegaCore/hardware/avr/2.0.0/avrdude.conf"
>          User configuration file is "/Users/dmitrij/.avrduderc"
>          User configuration file does not exist or is not a regular file, skipping
> 
>          Using Port                    : /dev/cu.wchusbserial1420
>          Using Programmer              : arduino
>          Overriding Baud Rate          : 115200
>          AVR Part                      : ATmega128
>          Chip Erase delay              : 9000 us
>          PAGEL                         : PD7
>          BS2                           : PA0
>          RESET disposition             : dedicated
>          RETRY pulse                   : SCK
>          serial program mode           : yes
>          parallel program mode         : yes
>          Timeout                       : 200
>          StabDelay                     : 100
>          CmdexeDelay                   : 25
>          SyncLoops                     : 32
>          ByteDelay                     : 0
>          PollIndex                     : 3
>          PollValue                     : 0x53
>          Memory Detail                 :
> 
>                                   Block Poll               Page                       Polled
>            Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
>            ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
>            eeprom         4    12    64    0 no       4096    8      0  9000  9000 0xff 0xff
>            flash         33     6   128    0 yes    131072  256    512  4500  4500 0xff 0xff
>            lfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
>            hfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
>            efuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
>            lock           0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
>            calibration    0     0     0    0 no          4    0      0     0     0 0x00 0x00
>            signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00
> 
>          Programmer Type : Arduino
>          Description     : Arduino
>          Hardware Version: 3
>          Firmware Version: 6.2
>          Vtarget         : 0.3 V
>          Varef           : 0.3 V
>          Oscillator      : 28.800 kHz
>          SCK period      : 3.3 us
> 
> avrdude: AVR device initialized and ready to accept instructions
> 
> Reading | ################################################## | 100% 0.02s
> 
> avrdude: Device signature = 0x1e9702 (probably m128)
> avrdude: reading input file "/var/folders/18/60_7xyc92138x6vhc69t25zc0000gn/T/arduino_build_113404/Blink.ino.hex"
> avrdude: writing flash (1240 bytes):
> 
> Writing | ################################################## | 100% 0.34s
> 
> avrdude: 1240 bytes of flash written
> avrdude: verifying flash memory against /var/folders/18/60_7xyc92138x6vhc69t25zc0000gn/T/arduino_build_113404/Blink.ino.hex:
> avrdude: load data flash data from input file /var/folders/18/60_7xyc92138x6vhc69t25zc0000gn/T/arduino_build_113404/Blink.ino.hex:
> avrdude: input file /var/folders/18/60_7xyc92138x6vhc69t25zc0000gn/T/arduino_build_113404/Blink.ino.hex contains 1240 bytes
> avrdude: reading on-chip flash data:
> 
> Reading | ########################################Произошла ошибка при загрузке скетча
> ########## | 100% 0.34s
> 
> avrdude: verifying ...
> avrdude: verification error, first mismatch at byte 0x0000
>          0xff != 0x0c
> avrdude: verification error; content mismatch
> 
> avrdude done.  Thank you.

Do you have any thoughts on it?

MCUdude, 2017-08-27:
Hi!
I've never experienced any issues while uploading using a USB to serial adapter. Do you have another chip/board to test with? You might want to double check that the fuses is correct

Borsoft73, 2017-08-31:
I'm stuck, still no progress...:( Bootloader don't want to write the code into the chip.
- I tried 3 another Atmega128A chips and 1 another PC.
- LED on PB5 is flashing so bootloader is in place and working. 
- USB-UART - I tried CH340 and CP2102 based devices. They work well with Atmega328 boards.
- UART connection seems also working as avrdude starting upload and can read chip signature
- 2x100nF caps Vcc-Gnd are in place, Reset pull-up 10k (I tried also 47k as mentioned in datasheet), 100nF cap on DTR to Reset line
- Fuses: 
       LFUSE - 0x3F
       HFUSE - 0xC6 
       EFUSE - 0xFF
       LOCK - 0x0F
- I tried with HFUSE 0xC4 (I found that bootloader starts from 0xFC00) and LOCK 0x3F (fully unlocked) - no result.
- I noted that upload starts even without DTR signal - usually other boards waiting for LOW on DTR line or manual Reset to start upload - looks strange

Here is the schematics I use: [https://easyeda.com/normal/Schematic-daa7aab244404e378c8103818bf43c53](url)

Any other ideas?

Borsoft73, 2017-09-05:
Do I have any chance to get some feedback?

MCUdude, 2017-09-05:
I'm sorry, but I've been on business trip for a whole week. I'll dig into it in a couple of days

MCUdude, 2017-09-07:
I'm not able to view the EasyEDA schematic. All I see is the logo.. please post a picture the schematic instead. A photo of your actual board would also be nice 🙂 

Like I've already mentioned; I've never seen this before. Using USB to serial uploading have always worked perfectly without any errors.

MCUdude, 2017-09-08:
hmm. I don't see any obvious hardware issues. Your schematic looks correct. Can you get avrdude to verify the current fuse settings, both low, high, extended and lockfuse?

Borsoft73, 2017-09-08:
Here are the fuses which I found in the chip after MegaCore installation: Low = 0xBF, High = 0xC6, Ext = 0xFF, Lock = 0x0F.
Bootloader settings I choose in Arduino IDE: Atmega128, BOD = 2.7V, Clock = Ext. 16MHz, LTO disabled.
LED on PB5 is flashing.
Serial itself is working well, I checked it with AnalogReadSerial example sketch.
But the same behavior during sketch upload by Serial:

> Reading | ################################################## | 100% 0.68s
> 
> avrdude: verifying ...
> avrdude: verification error, first mismatch at byte 0x0000
>          0xff != 0x0c
> avrdude: verification error; content mismatch
> 
> avrdude done.  Thank you.
> 
> the selected serial port 
>  does not exist or your board is not connected

Borsoft73, 2017-09-18:
And what about DTR behaviour? Is it normal if bootloader starting to receive data even without DTR line?

MCUdude, 2017-09-18:
Optiboot is able to start receiving data without the DTR line, but only once. After a program is received, it will never stay in bootloader mode forever, like it does when you have just burned the bootloader.

MCUdude, 2017-09-18:
If you want you can try to use the older version of optiboot to see if it works better. This bootloader was "shipped" with MegaCore v1.0.1

Borsoft73, 2017-12-14:
I'm here back again! The same with 1.0.1 and all other versions.
You can see how it looks here: [https://youtu.be/W3Efb8uDyqI](url)

Also I found similar behaviour with your SerialReadWrite sketch. It runs, serial connection is working, but it can't write anything to the Flash. Here is how it works: [https://youtu.be/45vlQO1n-Yg](url), and here is how the Flash dump looks after: [https://youtu.be/F0FbI3GkcaA](url). 

What do you think?

MCUdude, 2017-12-14:
You're not able to write to the flash memory without the bootloader installed. I have no idea why uploading using Serial doesn't work. @majekw do you have any idea why this happen?

Borsoft73, 2017-12-18:
Tried SerialReadWrite sketch compiled with bootloader - the same result - it can not write to the flash

Borsoft73, 2018-01-05:
Hi @majekw, may be you can help me with my issue? Still no progress and I will appreciate any ideas…

Borsoft73, 2018-01-09:
SOLVED!
Just switched from USBISP to AVRISP Mk2 programmer.

closed 2018-01-09 by Borsoft73

MCUdude, 2018-01-09:
Really?!

MCUdude, 2018-01-09:
You need to re-program you USBasp and check for sure. I use the USBasp as one of my main programmers, and it has never failed me.

X3msnake, 2018-01-10:
What usbisp are you using. Is it the chinese version. If so you should take a look at this git i fixed mine with it. https://github.com/aleh/usbisp

majekw, 2018-01-10:
Better late than never :-)
I also heard about some USBasp which have problems with devices with more flash. I'm glad that this is not bootloader problem, because everything from description looked fine.

Borsoft73, 2018-01-10:
Yes, probably it has old firmware. But I was confused as it looked like both bootloader and USBasp works properly - LED on PB5 flashed and I had no problems with loading my software (~70kB) through this USBasp... So anyway thanks to everybody for support!

prajju7, 2021-05-14:
I had the same problem getting that Verification error and content mismatch all the time for code above 50 bytes (Blink code works fine), I m using USBasp programmer, So should I buy the new one?

Thank You.

Borsoft73, 2021-05-14:
> I had the same problem getting that Verification error and content mismatch all the time for code above 50 bytes (Blink code works fine), I m using USBasp programmer, So should I buy the new one?
> 
> 
> 
> Thank You.
I switched to AVRIsp mkII programmer - works fine for me, no issues.