MCUdude/MegaCore #72: verification error, first mismatch at byte 0x3fc00

beingmachine, 2017-09-16:
Hello there,
I am trying to burn bootloaders to my Atmega2560V-8AU but came up with this error, what is this suppose to be.? how to solve this one .

> Arduino: 1.6.13 (Mac OS X), Board: "ATmega2560, Arduino MEGA pinout, 2.7v, Disabled (default), 8 MHz internal"
> 
> /Users/mudassir/Library/Arduino15/packages/arduino/tools/avrdude/6.3.0-arduino8/bin/avrdude -C/Users/mudassir/Documents/Arduino/hardware/MegaCore-avr-100-pin 2/avr/avrdude.conf -v -patmega2560 -cstk500v1 -P/dev/cu.usbmodem1421 -b19200 -e -Ulock:w:0x3f:m -Uefuse:w:0xfd:m -Uhfuse:w:0xd6:m -Ulfuse:w:0xe2:m 
> 
> avrdude: Version 6.3, compiled on Dec 16 2016 at 13:32:25
>          Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
>          Copyright (c) 2007-2014 Joerg Wunsch
> 
>          System wide configuration file is "/Users/mudassir/Documents/Arduino/hardware/MegaCore-avr-100-pin 2/avr/avrdude.conf"
>          User configuration file is "/Users/mudassir/.avrduderc"
>          User configuration file does not exist or is not a regular file, skipping
> 
>          Using Port                    : /dev/cu.usbmodem1421
>          Using Programmer              : stk500v1
>          Overriding Baud Rate          : 19200
>          AVR Part                      : ATmega2560
>          Chip Erase delay              : 9000 us
>          PAGEL                         : PD7
>          BS2                           : PA0
>          RESET disposition             : dedicated
>          RETRY pulse                   : SCK
>          serial program mode           : yes
>          parallel program mode         : yes
>          Timeout                       : 200
>          StabDelay                     : 100
>          CmdexeDelay                   : 25
>          SyncLoops                     : 32
>          ByteDelay                     : 0
>          PollIndex                     : 3
>          PollValue                     : 0x53
>          Memory Detail                 :
> 
>                                   Block Poll               Page                       Polled
>            Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
>            ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
>            eeprom        65    10     8    0 no       4096    8      0  9000  9000 0x00 0x00
>            flash         65    10   256    0 yes    262144  256   1024  4500  4500 0x00 0x00
>            lfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
>            hfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
>            efuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
>            lock           0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
>            calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
>            signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00
> 
>          Programmer Type : STK500
>          Description     : Atmel STK500 Version 1.x firmware
>          Hardware Version: 2
>          Firmware Version: 1.18
>          Topcard         : Unknown
>          Vtarget         : 0.0 V
>          Varef           : 0.0 V
>          Oscillator      : Off
>          SCK period      : 0.1 us
> 
> avrdude: AVR device initialized and ready to accept instructions
> 
> Reading | ################################################## | 100% 0.02s
> 
> avrdude: Device signature = 0x1e9801 (probably m2560)
> avrdude: erasing chip
> avrdude: reading input file "0x3f"
> avrdude: writing lock (1 bytes):
> 
> /Users/mudassir/Library/Arduino15/packages/arduino/tools/avrdude/6.3.0-arduino8/bin/avrdude -C/Users/mudassir/Documents/Arduino/hardware/MegaCore-avr-100-pin 2/avr/avrdude.conf -v -patmega2560 -cstk500v1 -P/dev/cu.usbmodem1421 -b19200 -Uflash:w:/Users/mudassir/Documents/Arduino/hardware/MegaCore-avr-100-pin 2/avr/bootloaders/optiboot_flash/atmega2560/optiboot_flash_atmega2560_38400_8000000L.hex:i -Ulock:w:0x0f:m 
> 
> avrdude: Version 6.3, compiled on Dec 16 2016 at 13:32:25
>          Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
[... 112 lines cut ...]
> avrdude: Device signature = 0x1e9801 (probably m2560)
> avrdude: NOTE: "flash" memory has been specified, an erase cycle will be performed
>          To disable this feature, specify the -D option.
> avrdude: erasing chip
> avrdude: reading input file "/Users/mudassir/Documents/Arduino/hardware/MegaCore-avr-100-pin 2/avr/bootloaders/optiboot_flash/atmega2560/optiboot_flash_atmega2560_38400_8000000L.hex"
> avrdude: writing flash (262144 bytes):
> 
> Writing | ################################################## | 100% 0.00s
> 
> avrdude: 262144 bytes of flash written
> avrdude: verifying flash memory against /Users/mudassir/Documents/Arduino/hardware/MegaCore-avr-100-pin 2/avr/bootloaders/optiboot_flash/atmega2560/optiboot_flash_atmega2560_38400_8000000L.hex:
> avrdude: load data flash data from input file /Users/mudassir/Documents/Arduino/hardware/MegaCore-avr-100-pin 2/avr/bootloaders/optiboot_flash/atmega2560/optiboot_flash_atmega2560_38400_8000000L.hex:
> avrdude: input file /Users/mudassir/Documents/Arduino/hardware/MegaCore-avr-100-pin 2/avr/bootloaders/optiboot_flash/atmega2560/optiboot_flash_atmega2560_38400_8000000L.hex contains 262144 bytes
> avrdude: reading on-chip flash data:
> 
> Reading | ################################################## | 100% 0.00s
> 
> avrdude: verifying ...
> avrdude: verification error, first mismatch at byte 0x3fc00
>          0x00 != 0x01
> avrdude: verification error; content mismatch
> 
> avrdude done.  Thank you.
> 
> Error while burning bootloader.

closed 2017-09-16 by beingmachine

jegjessing, 2019-12-16:
You closed this? Why? Did you find a sollution?
I have the same problem on 2 out of 5 boards. Nothing seems wrong on any of them, but the two of them still fail :-(

DeshmukhMalhar, 2019-12-23:
I think that this is a arduino as ISP, and this is a known error for it when verifying flash of mega 2560