MCUdude/MegaCore #74: TIMER1_A_PIN and TIMSK1 not declared in this scope

hapra2000, 2017-10-27:
I make DMD sketch with #include DMD.h, TimerOne.h, SPI.h, Wire.h, RTClib.h and DHT.h, but the error messages appears.
What should I do ?
 -----------------------------------------------------------------

Arduino: 1.8.4 (Windows 7), Board: "ATmega128, 2.7v, Disabled (default), 8 MHz external"

In file included from C:\Users\myfolder\Documents\Arduino\Dot Matrix\test128\test128.ino:10:0:
C:\Users\myfolder\Documents\Arduino\libraries\TimerOne-master/TimerOne.h: In member function 'void TimerOne::setPwmDuty(char, unsigned int)':
C:\Users\myfolder\Documents\Arduino\libraries\TimerOne-master/TimerOne.h:236:13: error: 'TIMER1_A_PIN' was not declared in this scope
  if (pin == TIMER1_A_PIN) OCR1A = dutyCycle;
             ^
C:\Users\myfolder\Documents\Arduino\libraries\TimerOne-master/TimerOne.h: In member function 'void TimerOne::pwm(char, unsigned int)':
C:\Users\myfolder\Documents\Arduino\libraries\TimerOne-master/TimerOne.h:245:13: error: 'TIMER1_A_PIN' was not declared in this scope
  if (pin == TIMER1_A_PIN) { pinMode(TIMER1_A_PIN, OUTPUT); TCCR1A |= _BV(COM1A1); }
             ^
C:\Users\myfolder\Documents\Arduino\libraries\TimerOne-master/TimerOne.h: In member function 'void TimerOne::disablePwm(char)':
C:\Users\myfolder\Documents\Arduino\libraries\TimerOne-master/TimerOne.h:260:13: error: 'TIMER1_A_PIN' was not declared in this scope
  if (pin == TIMER1_A_PIN) TCCR1A &= ~_BV(COM1A1);
             ^
C:\Users\myfolder\Documents\Arduino\libraries\TimerOne-master/TimerOne.h: In member function 'void TimerOne::attachInterrupt(void (*)())':
C:\Users\myfolder\Documents\Arduino\libraries\TimerOne-master/TimerOne.h:274:2: error: 'TIMSK1' was not declared in this scope
  TIMSK1 = _BV(TOIE1); //atmega128 only TIMSK
  ^
C:\Users\myfolder\Documents\Arduino\libraries\TimerOne-master/TimerOne.h: In member function 'void TimerOne::detachInterrupt()':
C:\Users\myfolder\Documents\Arduino\libraries\TimerOne-master/TimerOne.h:281:2: error: 'TIMSK1' was not declared in this scope
  TIMSK1 = 0; //atmega128 only TIMSK
  ^
exit status 1
Error compiling for board ATmega128.

This report would have more information with "Show verbose output during compilation" option enabled in File -> Preferences.

MCUdude, 2017-10-27:
ATmega128 is not supported by the TimerOne library by default. You need to add ATmega128 to the configuration file [known_16bit_timers.h](https://github.com/PaulStoffregen/TimerOne/blob/master/config/known_16bit_timers.h).

hapra2000, 2017-10-27:
Hi MCUdude,
I have read Atmega128 datasheet and related AVR header files about pins, but it seems not so easy for me to add it into TimerOne library.
Firstly, I must understand correlation between Arduino pins and Atmega128 pins physically, and how they work out. 
BTW, thank you for suggestion.

MCUdude, 2017-10-27:
It's quite easy, you need to look at the [pinout picture](https://github.com/MCUdude/MegaCore#pinout) I've provided.
OC1A is timer pin 1A, OC1B is timer pin 1B and OC1C is timer pin 1C 😉

MCUdude, 2017-11-10:
I'm closing this issue as this isn't a MegaCore related bug. If you can't figure it out you should probably create an issue over at the [TimerOne](https://github.com/PaulStoffregen/TimerOne) repo.

closed 2017-11-10 by MCUdude