MCUdude/MegaCore #77: Incorrect prescaler values in Tone.cpp

wolfbert, 2018-01-06:
Hi, first of of all, thank you for providing this software and instruction, after quite a bit of searching and frustration it has helped me to get my ATmega128 board to run with Arduino.

I realize tone() is not on your list of supported functions, but just in case anyone tries, here's what I've found. In file Tone.cpp, lines 267 ff. the timer prescaler value is determined. Now, on ATmega128, only timer 0 has prescalers 32 and 128, while the other three timers have only 1/8/64/256/1024 (see datasheet). 
So the code needs to be rewritten, in particular:
- Remove distinction between 8- and 16-bit timers (line 268), following code should work for both.
- Lines 277 and 288 should be changed to `if (_timer == 0...` to account for the additional values.
- The values in the conditional expressions in lines 286, 297 and 302 are switched around.

I've tried this with timer 2 (which is assigned by default) and the toneMelody example works as expected (it didn't before). Making the above changes may break the code for other processors (I've no idea how many different prescalers exist), so a general solution would require a bit of research in the datasheets.

Thanks and best regards
Wolfgang

MCUdude, 2018-01-06:
Thanks for reporting!
The core files for MegaCore is actually a separate repo and should work with MightyCore, MiniCore and MajorCore compatible microcontrollers as well. The repo can be found here:
[https://github.com/MCUdude/MCUdude_corefiles](https://github.com/MCUdude/MCUdude_corefiles).

Are you able to figure out a way to only affect the ATmega64/128 without throwing in a bunch of ugly macros? IMO the tone library has always been a mess though..

wolfbert, 2018-01-07:
> Are you able to figure out a way to only affect the ATmega64/128 without throwing in a bunch of ugly macros?

I'll look into it. At first sight, it seems the timers work the same. The challenge will be to integrate this in the existing code. I'll post an update when I know more.

wolfbert, 2018-01-07:
Looking into this some more, I found that

- ATmega64 and ATmega128 have the same timer prescaler settings, which are different from the "standard settings" in other ATxxxx in that timers 0 and 2 have been switched
- The code in Tone.cpp has quite a few problems and areas for improvement as indicated in the Arduino issues log, and not much seems to be happening about it

I believe I have a workable fix that addresses the issue of  the switched timers with minimal changes. It does require a new macro, though, see the attached file (changes marked with `// Proposed fix`).

[Tone_fixed.zip](https://github.com/MCUdude/MegaCore/files/1609945/Tone_fixed.zip)

Regards, Wolfgang

MCUdude, 2018-01-08:
Wow, great work! I'll test it out on various hardware this evening. Thank you for contributing! 😃

closed 2018-01-15 by wolfbert