MCUdude/MegaCore #78: Wrong high-fuses

ZayaV, 2018-01-07:
I think that the high-fuses have an incorrect value (0xc6) for ATmega128 in file boards.txt. Correct value is 0xc4, because the bootloader is downloaded to address 0xfc00.

MCUdude, 2018-01-07:
The bootloader "space" is determined by the the high fuse setting. How can the bootloader be loaded into address 0xfc00 -> 0xffff when the bootloader space is already set to 0xfe00 -> 0xffff with high fuse 0xC6?

ZayaV, 2018-01-07:
In file optiboot_flash_atmega128_115200_16000000L.hex (it is the bootloader) we can see the following lines:
:020000021000EC
:10**FC00**0001C0F5C01F92CDB7DEB711248FEF90E190

MCUdude, 2018-01-07:
I'm not the creator of Optiboot, so I'm not quite sure how to change the address. We don't need more than than 1024B (512 words) allocated for the bootloader space, so I'v rather edit the bootloader source and re-compile than edit the high fuse.

MCUdude, 2018-01-07:
.. And how would the bootloader even work if the memory location is wrong?

MCUdude, 2018-01-07:
OK, I've been searching around, but can't figure this out:

I've always thought that 1 word is the same as 2 bytes. The size of the compiled Optiboot bootloader for the ATmega128 is 1024 bytes, which is the same as 512 words (size of bootloader binary file). Why does the hex file uses 0xFC00 - 0xFFFF when the bootloader would fit into 0xFE00 - 0xFFFF? Do you know @westfw?

ZayaV, 2018-01-07:
Excuse me. I understood my mistake. I counted the addresses in bytes, and they are counted in words.

closed 2018-01-07 by ZayaV