MCUdude/MegaCore #79: Blink_using_Timer example not updated?

wolfbert, 2018-01-07:
Hi, I just ran into the incorrect bit value assignment in this example, which you fixed sometime in November. However, I only installed Arduino a few days ago, and the problem is still there. Is it possible that the fix somehow didn't make it from github to the automatic installation?

Regards, Wolfgang

MCUdude, 2018-01-07:
Hi!
It's true that the issue was fixed in November. However, I've not done a "Boards manager release" on a while, so this fix is not available yet. If you want the latest and greatest, you'll need to to a manual install.
I guess I'll have to do a new release soon? Maybe after the tone issue is solved? 🙂

wolfbert, 2018-01-07:
Ok, I see, no worries.

> Maybe after the tone issue is solved?

Done (see #77) .

closed 2018-01-15 by wolfbert