MCUdude/MegaCore #84: Interrupts not working when in sleep mode

theshadow124, 2018-03-07:
Hello,

Essentially the problem I am having is my interrupts are not working to wake a 2560 from sleep using the AVR core pinout, here is my test code:

```
#define alarm_pin 4
void blink()
{
  noSleep(); //comment out for no-sleep example
  alarm_flag = 1;
  Serial.println("Alarm?");
}
void setup()
{

  Serial.begin(250000);
  Serial.println("Start");

  pinMode(alarm_pin, INPUT);           // set pin to input
  digitalWrite(alarm_pin, HIGH);       // turn on pullup resistors
}

void loop()
{

  Serial.println("Loop");
  sleepMode(SLEEP_POWER_DOWN); //comment out for no-sleep example

  attachInterrupt(alarm_pin, blink, FALLING);
  Serial.flush(); //comment out for no-sleep example
  sleep(); //comment out for no-sleep example
  noSleep(); //comment out for no-sleep example

  delay(1000);
  if (alarm_flag == 1) {
    clr_alarm();
  }

void clr_alarm()
{
  detachInterrupt(alarm_pin);
  Serial.print("blink!\r\n");

  rtc.clearAlarm();
  delay(1000);
  alarm_flag = 0;
    attachInterrupt(alarm_pin, blink, FALLING);
}
```

The interrupt works if I remove the sleep lines, but won't wake the micro from sleep. I've tried interrupts INT4 and INT7, the board I am currently working on does not have any other INT pins broken out.

MCUdude, 2018-03-08:
Does it work if you use the regular Arduino Mega 2560 as your target? We need to figure out if this is a MegaCore related issue or not

theshadow124, 2018-03-08:
After quite a bit of fiddling with both using avr/sleep.h and the sleep options through this board functions, I have narrowed it down to be specifically the FALLING condition.

Works fine when using LOW but not FALLING. Both using your MegaCore and the standard 2560 board. I can make that work.

So I can say conclusively it is not limited to your board lib, where should I file a bug report on this for presumably the default lib? Unless my understanding of the chip is wrong, but this FALLING edge interrupt should still work in sleep, but I may be wrong.

MCUdude, 2018-03-09:
I think the best place to report/discuss this may be over at the [avrfreaks forum](https://www.avrfreaks.net). Thats where all the big guys hang out 😉

There _might_ be a silicon bug inside the ATmega2560, so post the date code of the chip along with your question.

closed 2018-03-09 by MCUdude