MCUdude/MegaCore #89: ATMega128 Serial

bonekost, 2018-06-03:
Hey @MCUdude,
I managed to burn bootloader with USBtinyISP,
and I also uploaded sketch with "_Upload Using programmer_"
I am trying to flash sketch over USB now, but I can't upload sketch through USB (USB dongle to Serial with CH340).
I got also problem with debug through Serial.

It looks like I can flash over USBtiny but I can't from USB to serial dongle.
I useed this [scheme](https://camo.githubusercontent.com/9407f77bb6316f8ea42b45d72acb6212b8baf726/68747470733a2f2f692e696d6775722e636f6d2f426b4a664957432e706e67).

I also tried switch Tx Rx but no luck.

Thanks

bonekost, 2018-06-04:
when I try upload sketch with USB I got error:
```
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x00
avrdude: ser_recv(): read error: Device not configured
```
[image]

MCUdude, 2018-06-04:
I looked up the RobotDyn adapter you're uaing, and it seems like it has a pullup resistor and a capacitor in series with the reset line. Connect the RST pin on your adapter to the RST pin on the ATmega128. It's also a good idea to try the 1 MHz internal oscillator instead just to get it to work. The programming baudrate is much slower (9600 instead of 57600 baud), so it's not so likely to fail due to poor wiring.

bonekost, 2018-06-05:
Hey @MCUdude thanks for advice but connecting RST pin with atmega didn't help.
Still getting error:
```
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x00
```

VinokurovSE, 2018-07-02:
@bonekost try to reset your board manually (connect and disconnect RTC to GND) before upload.
@MCUdude, thank you very much for your work!
When designing the board, I did not notice that there are no some pins in Arduino Mega. And the board did not want to work. And with your package without any problems.
Sorry for the offtopic.

bonekost, 2018-07-25:
Hey @MCUdude  I also tried classic parallel port for flash from Arduino IDE but still no luck. I noticed one weird thing when I push AVR example through USBtiny delays are twice long as should be (means 1000ms = 2sec in real)

MCUdude, 2018-07-25:
Did you choose 16 MHz external crystal in Arduino IDE?

bonekost, 2018-07-25:
yes

MCUdude, 2018-07-25:
Then there you have it! The crystal on your PCB is a 8 MHz one. select **8 MHz external** and re-burn the bootloader. I'm sure serial upload would work fine now 👍

MCUdude, 2018-08-01:
So, how did it go? 🙂

bonekost, 2018-08-02:
hey, it didn't help :( still getting `avrdude: stk500_recv(): programmer is not responding`

MCUdude, 2018-08-02:
As I said it may be an issue with the clock. Do you have an LED on the board you can blink? Try uploading the **Blink** example sketch using the programmer. Does the LED toggle every second, every 500ms or every 2s?

Are you able to use Serial.print in your code to print readable messages to your computer?

bonekost, 2018-08-02:
Yes, I have led on board and blink example now works as it should.
1sec = 1sec 
So changing to 8 MHz external helped, but only for time issue.
I still can't upload through USB or serial.
I also tried upload code for print to serial but it's weird cause I don't see any output on `dev/ttyS0`

MCUdude, 2018-08-02:
What happens if you use **Serial1**.print instead? You're sure you didn't mixed RX and TX?

bonekost, 2018-09-30:
Hey @MCUdude I tried to re-upload code through USBtinyISP with `Serial.print` and it works! I can see now output on serial monitor.

But still when I tried to upload some code through parallel port or USB-TTL no luck. It just uploading and no error message.

bonekost, 2019-01-01:
I used another `USB -> Serial (CH340G)` and I tried to kept RESET at GND connected until Arduino IDE starts uploading. And it works.
Now I can upload normally through USB.

closed 2019-01-01 by bonekost