MCUdude/MegaCore #91: Update Travis build script

MCUdude, 2018-07-08:
Maybe it's time to update the Travis build script @per1234?

I'm not 100% up to date on what features are missing, but at least it would be great if travis stuck check if any tabs were present.

per1234, 2018-07-08:
Sure thing. I've been adding some new features to arduino-ci-script over the last couple weeks and should be making a release with those and a couple minor improvements to existing code within a few days.

Here is a list of other available enhancements for the continuous integration process:
- Check for files starting with a blank line
- Check for trailing whitespace
- Check for non-Unix line endings
- Check for non-Unix line endings
- Check for files that don't end in a newline
- Use [grep-typos](https://github.com/ss18/grep-typos) to check for commonly misspelled words.
- Use [Travis Buddy](https://github.com/bluzi/travis-buddy) to automatically comment on pull requests that fail CI. I like this because otherwise you have to remember to come back to your PR later to see how the CI build went.
- Check for common problems with structure of libraries and their example sketches
- Check for common problems with keywords.txt files
- Check for common problems with library.properties files

Let me know if you would like any of those in addition to the tabs check.

I'm doing all except for the last three in all my own repositories and they're working fine.

The last three are the new features added to arduino-ci-script so I haven't added them to my repos yet but they do have good unit test coverage and are being **extensively** tested by running them on all 8736 libraries in my automatically generated [list](https://github.com/per1234/inoliblist).

per1234, 2019-01-19:
@MCUdude let me know if you still want me to work on this. I haven't proceeded because I was waiting for a reply to my previous comment regarding whether you're interested in any of the additional enhancements. The last three have now been thoroughly tested and all the former have been in use for quite some time in my own repos.

MCUdude, 2019-01-19:
Sorry, I totally forgot to reply to your previous message. As you may already know I don't know much about Travis build scripts, but I'll gladly merge an improved version of your build script!

closed 2019-04-12 by MCUdude