MCUdude/MegaCore #92: Can't upload bootloader and sketch

tamasgyugyei, 2018-08-27:
I bought Atmega128A chip, and I connected to RX, TX, RST, SCK, VCC, GND , CRY1, CRY2 (16MHz) but not burning bootloader, and not uploading sketches.
I try CH340G USB TTL, FT232RL, CP2102
and Arduino UNO ISP -> ICSP port.
and USBAsp, and USBAsp 2.0
but not working.... I tried several method, but nothing.
I saw other people have similar problem.
Maybe the MCUdude , MegaCore is fake?

I get not syncronize or invalid device sugnature error...

alshorin, 2018-08-27:
MegaCore for part no atmega128a I have works great. I didn't notice the problems you mentioned.

MCUdude, 2018-08-27:
First of all, if you believe MegaCore is fake just because you messed up somewhere, you don't deserve any help at all.

With that being said, you have provided very little (and unclear) information.

* Start out by using the **USBasp (slow)** programmer option instead. The USBasp may be too fast for a stock ATmega128. I've seen this many times before. MegaCore does not bring you this option, so you may install [MicroCore](https://github.com/MCUdude/MicroCore) to get the slow programmers.
* Were you actually able to burn bootloader? You have to to this to get the fuse bits right.
* Did you try adding series resistors to the data lines on the programmers, explained in the README file?
* Did you add proper bypass capacitors close to the Vcc and GND pins?
* please post a picture of your setup. It makes it so much easier for me to troubleshoot.

Before you even try uploading using a USB to serial adapter you should atlas get the bootloader running.

closed 2019-05-21 by MCUdude