MCUdude/MegaCore #94: atmega 64a

anandspathak, 2018-09-18:
Sir MCUdude,
Is there any difference between atmega 64 and 64a?
Do the above have different bootloader respectively?
I m new to this so pls help..
Regards
Anand Pathak

MCUdude, 2018-09-18:
Hi!
No, there's no difference between the ATmega64 and the ATmega64A in terms of code compatibility.

If you want to read about the electrical differences you can read the [AVR524](http://ww1.microchip.com/downloads/en/AppNotes/doc8165.pdf) document.

closed 2018-09-18 by MCUdude

anandspathak, 2018-09-18:
thanks sir, i uploaded bootloader successfully to atmega 64a... having some issue to blink an led,

anandspathak, 2018-09-18:
update.....blinking done...experimenting ahead

anandspathak, 2018-09-20:
Thanks for this wonderful core...its up and running on atmega 64a(once I left hope of which)