MCUdude/MegaCore #95: atmega 64a crystal issue

anandspathak, 2018-10-14:
Sir MCUdude,
i have atmega 64a with megacore running excellent at 8mhz crystal external and internal both...
but 16 mhz and 20 mhz crystals are not working and getting error as below upon bootloading...
can u suggest something...
Regards
Anand Pathak

***************************************************************
Arduino: 1.8.7 (Windows 7), Board: "ATmega64, 2.7v, Disabled (default), 16 MHz external"

C:\Program Files\Arduino\hardware\tools\avr/bin/avrdude -CC:\Users\vrindavan\AppData\Local\Arduino15\packages\MegaCore\hardware\avr\2.0.1/avrdude.conf -v -patmega64 -cstk500v1 -PCOM3 -b19200 -e -Ulock:w:0x3f:m -Uefuse:w:0xff:m -Uhfuse:w:0xc6:m -Ulfuse:w:0b10111111:m 

avrdude: Version 6.3-20171130
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Users\vrindavan\AppData\Local\Arduino15\packages\MegaCore\hardware\avr\2.0.1/avrdude.conf"

         Using Port                    : COM3
         Using Programmer              : stk500v1
         Overriding Baud Rate          : 19200
         AVR Part                      : ATmega64
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PA0
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom         4    20    64    0 no       2048    8      0  9000  9000 0xff 0xff
           flash         33     6   128    0 yes     65536  256    256  4500  4500 0xff 0xff
           lfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           lock           0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           calibration    0     0     0    0 no          4    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : STK500
         Description     : Atmel STK500 Version 1.x firmware
         Hardware Version: 2
         Firmware Version: 1.18
         Topcard         : Unknown
         Vtarget         : 0.0 V
         Varef           : 0.0 V
         Oscillator      : Off
         SCK period      : 0.1 us

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.02s

avrdude: Device signature = 0x000000 (retrying)

Reading | ################################################## | 100% 0.02s

avrdude: Device signature = 0x000000 (retrying)

Error while burning bootloader.
Reading | ################################################## | 100% 0.02s

avrdude: Device signature = 0x000000
avrdude: Yikes!  Invalid device signature.
         Double check connections and try again, or use -F to override
         this check.

avrdude done.  Thank you.

MCUdude, 2018-10-30:
Hi!
It's very likely a hardware issue. A device signature of 0x000000 indicated no communication with the microcontroller. Try using another crystal, try replacing the capacitors that's connected to the crystal with some other value. You can try 10pF, 18pF, 22pF and 27pF. Also try to move the crystal closer to the microcontroller itself. Long crystal traces is often a problem

anandspathak, 2018-10-30:
Thanks for reply...
I'll do as per your suggestion and revert back...
Regards

anandspathak, 2018-10-30:
Sir is this atmega64a designed to work only at 8 mhz not more than that? 
I am using breadboard...using 8mhz crystal I can do everything. ..when I replace higher frequency crystal leaving everything as it is, the above error occurs...just clearing...

MCUdude, 2018-10-30:
Sorry, I though you used an ATmega128. Where did you buy your ATmega64A? I've actually receved microcontrollers from china where the microcontroller actually is refurbished. The "new" text did say ATmega32A, but they didn't sand the chip properly. If you had the light correct I could barely spot "ATmega32L". This _might_ have happend to you as well I guess.

MCUdude, 2018-10-30:
The ATmega64A should infact support 16MHz. Buy how are you able to breadboard this? Do you have some kind of breakout board?

anandspathak, 2018-10-30:
Sir I made a simple pcb of atmega 64a for breadboard...ill show you image when I am back at home...
I got atmega 64a from aliexpress and I belong to india...

MCUdude, 2018-10-30:
Great! Please post some high-res photos when you're back from work 👍

anandspathak, 2018-10-30:
Ya sure

anandspathak, 2018-11-01:
Sir MCUdude,
Sorry for being late,
here are your images,
though the pcb is homemade by me but still works without any issues...

[image]
[image]
[image]
[image]
[image]
[image]

aasimmerchant05, 2019-04-18:
Dear Sir Where have u connected the crystal? There is no internal oscillator running at 16MHz. You need an external crystal for that

MCUdude, 2019-05-21:
I suggest you take this at the Arduino forum or the Avrfreak forum. This is a hardware issue, and nothing to do with MegaCore really.

closed 2019-05-21 by MCUdude