MCUdude/MegaCore #96: Thank You!

DirtyEngineer, 2018-10-29:
This isn't an issue. @MCUdude, I would just like to express my thanks for all the work you put into MegaCore. Thank you for sharing your knowledge.  It has been extremely helpful and has made working with the ATMega128 very easy.  It is very much appreciated.

Cheers.

X3msnake, 2018-10-30:
> Thanks! I really appreciate it when people take their time to leave a
> thanks 🙂 I don't make any money of any of my github repos (not my
> intention either), but I enjoy writing and maintaining code that's being
> used by others than just my self.
>
> IIRC I didn't really develop MegaCore because I *had* to use the
> ATmega128 in a project. I did it because I was suprised that no one had
> really provided a stable, up to date and well documented Arduino core for
> this particular AVR family. I also had some experience from developing
> MightyCore, so it wasn't too hard really. And, the fact that it cost $1 at
> Ebay makes the ATmega128 perfect for larger PCBs with lots of IO!

I take this chance to extend my thanks too. it is awsome to have Masters
like you willing to share for the sake of open knowlage. ;)

Cesar-S, 2018-11-12:
Thanks to MegaCore eh used Atmega128 that had old "J.A.F. box" with Aduino

closed 2022-12-25 by MCUdude