MCUdude/MegaCore #102: MegaCore wont use one wire on PF0....PF7

AndersDenmark, 2019-01-13:
Hi i am having problems getting PF0... to PF7 (ADC) ports to work on a ATMEGA 64A with Dallastemperature, for DS18B20 Sensor. I have tested on other port pins and it works perfect. Are there something i am missing when using MegaCore and ADC Pins for digital functions. ?
i Have tested I/O Digital write, and no problem i can switch the ADC pins Digital on and off, it just wont use the one wire function.
Please help, i am almost out off ideas where to look, an why it wont work on Port PF0 to 7.

MCUdude, 2019-01-13:
There's no reason why PF0-PF7 should be different than the other pins. The OneWire library is not interrupt based either, so it should work with any pin too.

Do you have access to a digital oscilloscope? If you have you can capture a "message" sent from a working pin, store it as a reference and compare it with a "message" sent from one of the PF pins

MCUdude, 2019-01-13:
I've done a bit more research. I connected my oscilloscope to one of the pins on PORTF. Turns out the port acts as it's completely dead. It's not a timing issue, it's just that the microcontroller doesn't even try. 

I tested ATmega64, ATmega128, ATmega1281 and ATmega2561, and strangely enough it works perfectly on 1281 and 2561! At the moment I have no idea why. They all use the same Arduino source code.

PaulStoffregen, 2019-01-13:
Nope, I had no idea of this.  I also do not have any Arduino boards with those chips.  I do test with Arduino Uno, Mega and Leonardo for 8 bit AVR support.  I also have a growing pile of 32 bit boards.  For all other 8 bit AVR chips, I depend on community contributions....

MCUdude, 2019-01-13:
hmm. is the *base* value known at compile time? It may be possible to add a macro that's fixes this if so.

MCUdude, 2019-01-13:
I don't know the library well enough to be able to figure this out without spending a lot of time.

Basically I'm trying to replace four macros if port F is used (Arduino pin 45-52). However I'm not able to "tell" the compiler that pin 45 is inserted into the constructor not that port F is used.

``` c
#define DIRECT_MODE_INPUT(base, mask)   ((*((base)+0x41)) &= ~(mask))
#define DIRECT_MODE_OUTPUT(base, mask)  ((*((base)+0x41)) |= (mask))
#define DIRECT_WRITE_LOW(base, mask)    ((*((base)+0x42)) &= ~(mask))
#define DIRECT_WRITE_HIGH(base, mask)   ((*((base)+0x42)) |= (mask))
```

I don't think I'm going to spend much time on this. It's not my code and it's not really an issue for me personally.

PaulStoffregen, 2019-01-13:
Ok.  I've created an issue over on the OneWire library.  Hopefully someone using these chips will want to pursue this further and come up with a pull request to fix it for everyone.  Or at least these issues can serve as a warning to use other pins on these specific AVR chips.

MCUdude, 2019-01-13:
Great! However it does work on ATmega1281 and ATmega2561. It's ATmega64 and ATmega128 that's causing trouble

closed 2019-01-13 by MCUdude

AndersDenmark, 2019-01-13:
Hi Guys

I really appreciate all your help.
I have really liked working with the MegaCore Board, it is so cool.
the reason is that at my work we have already made regulating PCBs, all with ATMEGA 64A, and the DS18B20 mounted to PF7 on the boards. so just mowing the sensor to another pin is NOT an option.
All these are programmed with CodeVision IDE but NOT by me, so i know it works with one wire on PORTF with the ATMEGA 64A. 
I prefer the Arduino IDE, and all the nice libs you can use with it. So it would be really nice to get it to work with the MegaCore. what is the best option work around to get it to work. Is it to modify the Dallas Lib, maybe renaming the modified Lib to Dallastemp_Atmega64, or ard there a better solution, before anybody are willing to remake the Dallas Lib. ?

Thanks everybody for all your kindly and good help.

MCUdude, 2019-01-13:
Great to hear you find MegaCore useful!

If you already have ready-made hardware you can either switch to another pin compatible microcontroller such as ATmega1281 or ATmega2561, or you could modify the library.

It would probably be easiest to modify the library to make it work with PORT F. If you have an Arduino sketch you can just move the library files into the sketch folder. This way you you have a separate modified copy of the OneWire library attached to the specific sketch.

The path should look something like this. Just remember to rename the included files too. The modified library is included by adding `#include "OneWire-64-128.h"` in the beginning of your sketch (With "" not <>)

MySketch/
  MySketch.ino
  OneWire-64-128.h
  OneWire-64-128.cpp
  util/
    OneWire-64-128_direct_gpio.h
    OneWire-64-128_direct_regtype.h