MCUdude/MegaCore #103: Ability to configure Bootloader fuse from Arduino IDE

jpraus, 2019-01-13:
Hi, thank you very much for your work!

Would it be possible to add Bootloader (Yes/No) option for ATmega2561 device? Same way it is currently being done in https://github.com/MCUdude/MiniCore library for ATmega8 for example.

ATmega2561 (MegaCore)
[image]

ATmega8 (MiniCore)
[image]

Currently, when I want or don't want to use bootloader I need to manually update the appropriate file to change the value of fuses to either use bootloader or not.

MCUdude, 2019-01-13:
All you really need to do is connect your programmer and hold shift while pressing Upload. The bootloader will be gone. Why do you need that extra 1kB space when you're using an ATmega2561? IMO it makes sense to get rid of the bootloader section with devices like ATmega8 and ATmega168, but not ATmega2561

jpraus, 2019-01-15:
This is not about bootloader itself. If I want to upload program directly (without using bootloader) I need to manually change fuses configuration to start without bootloader otherwise the program wont start.

Maybe I am missing something. I am using USBasp programmer to flash ATmega2561 microcontroller, when I choose to upload using programmer it will erase whole flash including bootloader, bootloader is not included by default when uloading via programmer (onyl if you burn it by burn bootloader). Then the program wont run because fuses are set for microcontroller to run with bootloader which is not there.

jpraus, 2019-01-15:
This is what is stated in miniCore:

> Note that you need to select Bootloader: "No" and actually hit Burn bootloader in order to set the correct fuse bits. Now you're no longer able to upload using serial, until you select Bootloader: "Yes" and burn the bootloader again.

But I don't want to use bootloader at all, even if I have plenty of room, I am flashing my ATmega2561 directly via programmer (not via serial).

MCUdude, 2019-01-15:
> I need to manually change fuses configuration to start without bootloader otherwise the program wont start.

That is not true. When the bootloader fuse bits is set, the microcontroller will indeed start in the bootloader section. Since there's no instructions the program counter will roll over and start at the bottom, where your program is. I just tested this with an ATmega2561 before I wrote this post.

jpraus, 2019-01-18:
Thank you for your replies so far, appreciate your help!

However, the problem is not in the size of bootloader or a slow start of the program. With default fuses provided with the library, my program simply won't start if programming it via "Upload using programmer" - I am using USBasp programmer. To my understanding, this is because the high fuse is set to 0xd6 - meaning BOOTRST flag is on. And since I am not using the bootloader to program the chip but flashing it completely. There is no bootloader to start from.

On the other hand, if I change the high fuse to 0xd7 - BOOTRST is unset. My program starts like a charm. This is what the Bootloader > Yes/No option is doing in MiniCore right? So to my understanding, it would also solve my issue.

Of course, this problem can be between the chain and the keyboard so forgive me my ignorance. Here is attached log from avrdude programming the fuses:

avrdude fuse setup with default fuses (high fuse set to 0xd6)
[fuse2561-bootloader.txt](https://github.com/MCUdude/MegaCore/files/2774356/fuse2561-bootloader.txt)

avrdude fuse setup with the high fuse set to 0xd7
[fuse2561-nobootloader.txt](https://github.com/MCUdude/MegaCore/files/2774355/fuse2561-nobootloader.txt)

per1234, 2019-01-18:
The problem you're encountering is discussed here: https://github.com/arduino/Arduino/issues/388
Note that issue contains a parallel discussion of the same issue as it occurs with the Arduino as ISP programmer.

If you change the firmware on your USBasp to the 3rd party firmware here:
https://github.com/PeterVH/usbasp
You will be able to use the MegaCore default fuse settings for both standard Upload as well as Upload Using Programmer.

---
As for the topic of adding a Bootloader menu option a la MiniCore, I don't have a strong opinion either way. I'll defer to MCUdude's [reasoning ](https://github.com/MCUdude/MegaCore/issues/103#issuecomment-453830718)on that.

MCUdude, 2019-01-20:
To prevent any further confusions With different versions of Avrdude and USBasp firmware I've decided to add the bootloader option in the tools menu.
[image]

You can do a manual install of MegaCore to get this feature. Otherwise you'll have to wait for a new boards manager release

closed 2019-01-20 by MCUdude

jpraus, 2019-01-21:
Oh man, you are awesome! I appreciate your help and hard work!

MCUdude, 2019-01-21:
> You can do a [manual install](https://github.com/MCUdude/MegaCore#manual-installation) of MegaCore to get this feature

DeshmukhMalhar, 2019-01-21:
> 
> 
> > You can do a [manual install](https://github.com/MCUdude/MegaCore#manual-installation) of MegaCore to get this feature

I thought you made a branch or something,
anyway,
thanks.