MCUdude/MegaCore #104: Atmega64a TimerOne

AnatoliyKube, 2019-02-01:


AnatoliyKube, 2019-02-01:
Hello !
i use Atmega64a and  MegaCore
library TimerOne 
does not work, since other registers are used for setting the timer,
attempt to create an interrupt by vector ISR (TIMER1_COMPA_vect) does not work
void setup()
{
pinMode(13,OUTPUT);
digitalWrite(13,LOW);
cli();
OCR1A=31250;
//OCR1AH = 0b01111010; 
//OCR1AL = 0b00010010;
TCCR1A = 0; 
TCCR1B =0;
TCCR1B |= (1<<WGM12 , 1<<CS12);// oset the divider and  CTC OCR1A
TIMSK |= (1<<OCIE1A);//0b00010000
sei();
}
......
.....
.....

ISR (TIMER1_COMPA_vect)
{
digitalWrite(13,digitalWrite(13)^1);//led switching
}

MCUdude, 2019-02-02:
If you're having trouble with the TimerOne library you'll have to report it to the library author, not me.

The LED will never toggle with the code you posted either. This works though.

``` c
void setup()
{
  pinMode(13, OUTPUT);
  digitalWrite(13, LOW);
  cli();
  OCR1A = 31250;
  TCCR1A = 0;
  TCCR1B = (1 << WGM12) | (1 << CS12); // oset the divider and CTC OCR1A
  TIMSK |= (1 << OCIE1A); //0b00010000
  sei();
}

void loop()
{

}

ISR (TIMER1_COMPA_vect)
{
  digitalWrite(13, !digitalRead(13));//led switching
  TCNT1 = 0; // Clear counter
}
```

closed 2019-02-02 by MCUdude

AnatoliyKube, 2019-02-02:
the problem did not dare,
after experimenting with the code, only enforcing the registers helped
void setup()
{
//atmega64A MegaCore Arduino IDE 1.8.7
pinMode(13,OUTPUT);
digitalWrite(13,LOW);

OCR1A=12500;
TCCR1A =0b00000000;//TCCR1A =0X00
TCCR1B =0b00001011;//TCCR1B |= (1<<WGM12) , (1<<CS11) , (1<<CS11);
TCCR1C =0b00000000;
TIMSK  = 0B00010000;// TIMSK |= (1<<OCIE1A);
}
void loop()
{
}
ISR (TIMER1_COMPA_vect)
{
digitalWrite(13,digitalWrite(13)^1);//led switching
}

but  !!!!!!!!!!!!!
after the first interrupt, in the register OCR1A  , for some reason all zeros are set. (????????)
at setting the mode is set CTC ,and loss of values in the register OCR1A OCR1A don't have to happen
if you change ISR (TIMER1_COMPA_vect)  - 
ISR (TIMER1_COMPA_vect)
{
OCR1A=12500;
digitalWrite(13,digitalWrite(13)^1);//led switching
}
timer running , but - in rehime CTC  register values OCR1A should automatically load into the register TCNT1(L,H)
I checked everything in assembler, everything works there.
in arduino it is necessary to reset the register OCR1A - these are resource costs