MCUdude/MegaCore #105: about delay 1000 issues

FestumZ, 2019-02-19:
in arduino IDE , delay(1000); actually is 2 second, How to fix it to 1 second ?

MCUdude, 2019-02-19:
What clock frequency are you using? You need to select the correct settings in the tools meny and click "Burn Bootloader" in order for the timing to be correct.

FestumZ, 2019-02-19:
Hi  ,
Clock is using 16MHz external. 
Is there sample to show what tools(arduino IDE?) and settings for this case.
thanks >_<

MCUdude, 2019-02-19:
I'm not on a computer with MegaCore installed at the moment.

All you need to do is to select 16MHz external" and click "Burn bootloader". Settings like BOD and LTO can be left in their default state

FestumZ, 2019-02-19:
Hi ,
burn bootloader success but still is 2 second.

setting as below: 
board: ATmega128
Clock: 16 MHz external
BOD:2.7V
Compiler LTO:Disabled (default)

thanks your comment

MCUdude, 2019-02-19:
Do you upload the code using an ISP programmer or a USB to serial adapter?

If you connect an LED to PB5 (Digital pin 13) and burn the bootloader again; does the LED blink twice every second or twice every half second?

FestumZ, 2019-02-20:
I upload the code using Uno R3 as ISP programmer.

burn again the bootloader and connect LED to PB5 that I got LED blink once every four second.
the LED have two second light and two second dark...

the code as below
`
int pins = 13;

void setup() {
  pinMode(pins, OUTPUT);
}

void loop() {
  digitalWrite(pins, HIGH);   // turn the LED on (HIGH is the voltage level)
  delay(1000);                       // wait for a second
  digitalWrite(pins, LOW);    // turn the LED off by making the voltage LOW
  delay(1000);                       // wait for a second
}`

MCUdude, 2019-02-21:
This means your crystal must be 8 MHz. That will explain the wrong baud rate issues too. Try burning the bootloader with the **8 MHz external** option.

FestumZ, 2019-02-21:
Hi MCUdude , 

It's work when I change CPU to 8 MHz external or internal...Thanks and...

[image]

The ATmega128A's parametrics show us that the CPU is 16 MHz, 
Why we must use 8 MHz  ._. ?

MCUdude, 2019-02-21:
You can have it running at 16 MHz if you want. The problem is your crystal. It's not running at 16 MHz, but 8. If you have an oscilloscope you can verify this by measuring the signal on the XTAL1 pin.

FestumZ, 2019-02-21:
Got it. 
Thank you!

closed 2019-02-21 by FestumZ