MCUdude/MegaCore #108: Burning the bootloader onto the Atmega128 using the Arduino Uno as the ISP

mrdesignr, 2019-03-24:
I ordered the QFP chip from Ebay and mounted it onto an adapter board. I connected it's 16mhz crystal and it has it's own 5 v power supply. The connections are ARduino uno ground to Atmega128 ground, Uno SS to Atmega 128 pin 20, RESET pulled up with 10k resitor, Uno MOSI to Atmega 128 pin 12, Uno MISO to Atmega 128 pin 13, and Uno SCK to Atmega 128 pin 11. When I try to burn the bootloader i get an invalid signature message and no writing is done. It also said to use -F to skip the signature test. 
[Invalid signature.txt](https://github.com/MCUdude/MegaCore/files/3000966/Invalid.signature.txt)
The report from the Arduino IDE is attached above. Also attached below is a screenshot of the Atmega128 settings in the Arduino IDE. 
[image]

MCUdude, 2019-03-24:
The programmer should be connected to pin 2 and 3 on the ATmega128, not 12 and 13. See the [minimal setup section](https://github.com/MCUdude/MegaCore#minimal-setup) for details.

mrdesignr, 2019-03-25:
That worked! thanks

closed 2019-03-31 by MCUdude