MCUdude/MegaCore #109: AT90can128 chip

geraldjust, 2019-04-21:
Could it be possible to support the at90can128 chip? i looked at it and the atmega128 and they both seem to be identical. The only issue i think is that PB4 is 0C2A but it should basicly be a drop in replacement. i tried modifying the boards.txt file and got support but some thing likr the tone lib or winterupts needed some minor work.

MCUdude, 2019-04-21:
I'll look into it. I'm not sure if I'll physically add them to the boards.txt file, but I'll at least make sure libraries and core files support them. Do you own any AT90CAN* hardware? I've added the AT90CAN32/64/128 to my [Optiboot flash repo](https://github.com/MCUdude/optiboot_flash). It would be great if you could test this bootloader to see if everything works correctly.

geraldjust, 2019-04-21:
Just tested the 16000000L version. running at 115200 on uart0 and seems to work fine!

Gerald R.

MCUdude, 2019-04-26:
Great to hear the bootloader is working just fine!
I don't have any AT90CAN hardware at the moment (should probably try to get it), but can you test the latest [MCUdude_corefiles](https://github.com/MCUdude/MCUdude_corefiles)? Just download this repo as a ZIP file and replace the existing one inside the MegaCore folder. Thanks!

geraldjust, 2019-05-01:
I would be more than happy to get you some hardware to test out. So got a couple of updates. When i originally tested the opti-flash i accidentally renames a old bootloader i was already using from https://github.com/Atlantis-Specialist-Technologies/CAN485
But now that i double checked because i renames the same folder the bootloader doesnt work. 
So im using the AST bootloader for now with the rest of your core. 
So, almost all PWM work. except for PWM on pin 12. Uart0 & 1 work. But the Tone lib does not and neither does the attach interrupt functions. 
I have a feeling that it has something to do with a issue on Timer2.
But tone lib seems to work just fine. 
https://bitbucket.org/teckel12/arduino-new-tone/src/master/

MCUdude, 2019-05-01:
> I would be more than happy to get you some hardware to test out. 

I do have a good development board for these 64 pin AVRs, so all I really need is a bare chip in a TQFP package. If you would like to send me anything we can talk futher on email. hansibull[A in a circle][googles mail service].com. Sorry for the scrambling, but I'm so tired of all these spam robots picking up emails 😉 

I looked at the CAN485 bootloader source, and I couldn't really see why the bootloader I compiled should be any different that the CAN485 one. Are you sure you used [this exact file](https://github.com/MCUdude/optiboot_flash/blob/master/bootloaders/atmega128/16000000L/optiboot_flash_atmega128_UART0_115200_16000000L_BIGBOOT.hex)?

> So, almost all PWM work. except for PWM on pin 12. Uart0 & 1 work. But the Tone lib does not and neither does the attach interrupt functions.

Thanks for testing all this! There may be something wrong with timer 2, yes. I'll see what I can do.

geraldjust, 2019-05-02:
hmm well maybe its one of the fuse bits thats causing the error. These are the modifications i made to the system to add the 128 chip. 
https://github.com/geraldjust/MegaCore
And you linked it to the mega128 lol. but saw you renamed them to bigboot.

MCUdude, 2019-05-02:
Sorry, I meant to link in [this one](https://github.com/MCUdude/optiboot_flash/blob/master/bootloaders/at90can128/16000000L/optiboot_flash_at90can128_UART0_115200_16000000L_BIGBOOT.hex). I've decided to change the suffix to *BIGBOOT*. This is because the bootloader now also has flash page copy functionality if the bootloader is compiled as BIGBOOT. EEPROM support is just one of the extra features.

MCUdude, 2019-05-15:
Thank you so much for the chips! I wouldn't be able to sort these issues out without actual hardware.

 I think I have sorted out the problems you found earlier. The bootloader issues were caused by some missing defines. Turns out the AT90CAN128 is more similar to the ATmega128 than ATmega1281. That's why it should be "treated" the same way as the 128 and not as the 1281 like I guessed earlier.

The timer2a PWM issue on digital pin 12 was caused because the AT90CANs doesn't have a TCCR2B register like ATmega1281 has. This prevented PWM to be initialized correctly for this timer.

Try getting the latest bootloaders from [here](https://github.com/MCUdude/optiboot_flash) and the latest core files from [here](https://github.com/MCUdude/MCUdude_corefiles). I've also added AT90CAN32/64/128 to [avrdude.conf](https://github.com/MCUdude/MegaCore/blob/master/avr/avrdude.conf).

geraldjust, 2019-05-16:
Great so the code for avrdude and the new bootloader seems to be working great! im still having a issue with the pin 12 pwm. still having issues too with the attachinterrupt function and the tone lib. i will keep looking as to why this hasn't been working.

Also fuse bits are way different compared to the 128.
https://github.com/geraldjust/MegaCore/blob/master/avr/boards.txt
I edited the boards.txt to add the support for the chip. with the corrected fuse bit location to get it working. Tested it with this and the new bootloaders and its working like a charm besides the above issues.

MCUdude, 2019-05-16:
> im still having a issue with the pin 12 pwm

That's strange. Please have a look in the wiring.h and see if the TCCR2A register is present there.

> Also fuse bits are way different compared to the 128.

I've added CAN32, CAN64, and CAN128 to the boards.txt on my computer. I haven't pushed it yet though. I can paste these options here on this page when I get home from work.

MCUdude, 2019-05-16:
For reference, here are the boards.txt entries for CA32, 64 and 128:

```
#####################
#### AT90CAN128  ####
#####################

# General
can128.name=AT90CAN128
can128.upload.tool=avrdude
can128.upload.maximum_data_size=4096
can128.bootloader.tool=avrdude
can128.bootloader.unlock_bits=0x3f
can128.bootloader.lock_bits=0x0f
can128.build.core=MCUdude_corefiles
can128.build.variant=64-pin-avr
can128.build.board=AVR_AT90CAN128
can128.build.mcu=at90can128

# Upload port select
can128.menu.bootloader.uart0=Yes (UART0)
can128.menu.bootloader.uart0.upload.maximum_size=130048
can128.menu.bootloader.uart0.upload.protocol=arduino
can128.menu.bootloader.uart0.upload.port=UART0
can128.menu.bootloader.uart0.bootloader.high_fuses=0xd6
can128.menu.bootloader.uart0.bootloader.file=optiboot_flash/bootloaders/{build.mcu}/{build.f_cpu}/optiboot_flash_{build.mcu}_{upload.port}_{upload.speed}_{build.f_cpu}_BIGBOOT.hex

can128.menu.bootloader.uart1=Yes (UART1)
can128.menu.bootloader.uart1.upload.maximum_size=130048
can128.menu.bootloader.uart1.upload.protocol=arduino
can128.menu.bootloader.uart1.upload.port=UART1
can128.menu.bootloader.uart1.bootloader.high_fuses=0xd6
can128.menu.bootloader.uart1.bootloader.file=optiboot_flash/bootloaders/{build.mcu}/{build.f_cpu}/optiboot_flash_{build.mcu}_{upload.port}_{upload.speed}_{build.f_cpu}_BIGBOOT.hex

can128.menu.bootloader.no_bootloader=No bootloader
can128.menu.bootloader.no_bootloader.upload.maximum_size=131072
can128.menu.bootloader.no_bootloader.bootloader.high_fuses=0xd7
can128.menu.bootloader.no_bootloader.bootloader.file=empty/empty.hex

# Brown out detection
can128.menu.BOD.2v7=BOD 2.7V
can128.menu.BOD.2v7.bootloader.extended_fuses=0xf5
can128.menu.BOD.4v1=BOD 4.1V
can128.menu.BOD.4v1.bootloader.extended_fuses=0xfd
can128.menu.BOD.4v0=BOD 4.0V
can128.menu.BOD.4v0.bootloader.extended_fuses=0xfb
can128.menu.BOD.3v9=BOD 4.1V
can128.menu.BOD.3v9.bootloader.extended_fuses=0xf9
can128.menu.BOD.3v8=BOD 4.0V
can128.menu.BOD.3v8.bootloader.extended_fuses=0xf7
can128.menu.BOD.2v6=BOD 4.1V
can128.menu.BOD.2v6.bootloader.extended_fuses=0xf3
can128.menu.BOD.2v5=BOD 4.0V
can128.menu.BOD.2v5.bootloader.extended_fuses=0xf1
can128.menu.BOD.disabled=BOD disabled
can128.menu.BOD.disabled.bootloader.extended_fuses=0xff

# Compiler link time optimization
can128.menu.LTO.Os=LTO disabled
can128.menu.LTO.Os.compiler.c.extra_flags=
can128.menu.LTO.Os.compiler.c.elf.extra_flags=
can128.menu.LTO.Os.compiler.cpp.extra_flags=
can128.menu.LTO.Os.ltoarcmd=avr-ar

can128.menu.LTO.Os_flto=LTO enabled
can128.menu.LTO.Os_flto.compiler.c.extra_flags=-Wextra -flto -g
can128.menu.LTO.Os_flto.compiler.c.elf.extra_flags=-w -flto -g
can128.menu.LTO.Os_flto.compiler.cpp.extra_flags=-Wextra -flto -g
can128.menu.LTO.Os_flto.ltoarcmd=avr-gcc-ar

[... 217 lines cut ...]
can32.menu.clock.18_432MHz_external=18.432 MHz external
can32.menu.clock.18_432MHz_external.upload.speed=115200
can32.menu.clock.18_432MHz_external.bootloader.low_fuses=0xf7
can32.menu.clock.18_432MHz_external.build.f_cpu=18432000L

can32.menu.clock.12MHz_external=12 MHz external
can32.menu.clock.12MHz_external.upload.speed=57600
can32.menu.clock.12MHz_external.bootloader.low_fuses=0xf7
can32.menu.clock.12MHz_external.build.f_cpu=12000000L

can32.menu.clock.8MHz_external=8 MHz external
can32.menu.clock.8MHz_external.upload.speed=38400
can32.menu.clock.8MHz_external.bootloader.low_fuses=0xf7
can32.menu.clock.8MHz_external.build.f_cpu=8000000L

can32.menu.clock.8MHz_internal=8 MHz internal
can32.menu.clock.8MHz_internal.upload.speed=38400
can32.menu.clock.8MHz_internal.bootloader.low_fuses=0xe2
can32.menu.clock.8MHz_internal.build.f_cpu=8000000L

can32.menu.clock.1MHz_internal=1 MHz internal
can32.menu.clock.1MHz_internal.upload.speed=9600
can32.menu.clock.1MHz_internal.bootloader.low_fuses=0x62
can32.menu.clock.1MHz_internal.build.f_cpu=1000000L
```

MCUdude, 2019-05-18:
I've done some more testing, and I can verify that I have PWM and tone working on pin 12 (timer2). The last commit to MCUdude_corefiles solves the issue with the external interrupt pins. All interrupts seems to work fine now.

MCUdude, 2019-05-18:
I've uploaded all AT90CAN related code to the master branch. @per1234 can you please add AT90CAN32, AT90CAN64, and AT90CAN128 to the Travis CI script?

per1234, 2019-05-18:
Great news! I will do so.

per1234, 2019-05-19:
@MCUdude there is a problem with the SoftwareSerial library due to the missing `digitalPinToPCICR` and `digitalPinToPCMSK`:
```
/home/travis/Arduino/hardware/MegaCore/avr/libraries/SoftwareSerial/src/SoftwareSerial.cpp: In member function 'void SoftwareSerial::begin(long int)':
/home/travis/Arduino/hardware/MegaCore/avr/libraries/SoftwareSerial/src/SoftwareSerial.cpp:330:36: error: 'digitalPinToPCICR' was not declared in this scope
   if (digitalPinToPCICR(_receivePin)) {
                                    ^
In file included from /home/travis/arduino-ide/arduino-1.8.9/hardware/tools/avr/avr/include/avr/io.h:99:0,
                 from /home/travis/arduino-ide/arduino-1.8.9/hardware/tools/avr/avr/include/avr/interrupt.h:38,
                 from /home/travis/Arduino/hardware/MegaCore/avr/libraries/SoftwareSerial/src/SoftwareSerial.cpp:43:
/home/travis/Arduino/hardware/MegaCore/avr/libraries/SoftwareSerial/src/SoftwareSerial.cpp:371:76: error: 'digitalPinToPCICRbit' was not declared in this scope
     *digitalPinToPCICR(_receivePin) |= _BV(digitalPinToPCICRbit(_receivePin));
                                                                            ^
/home/travis/Arduino/hardware/MegaCore/avr/libraries/SoftwareSerial/src/SoftwareSerial.cpp:374:51: error: 'digitalPinToPCMSK' was not declared in this scope
     _pcint_maskreg = digitalPinToPCMSK(_receivePin);
                                                   ^
In file included from /home/travis/arduino-ide/arduino-1.8.9/hardware/tools/avr/avr/include/avr/io.h:99:0,
                 from /home/travis/arduino-ide/arduino-1.8.9/hardware/tools/avr/avr/include/avr/interrupt.h:38,
                 from /home/travis/Arduino/hardware/MegaCore/avr/libraries/SoftwareSerial/src/SoftwareSerial.cpp:43:
/home/travis/Arduino/hardware/MegaCore/avr/libraries/SoftwareSerial/src/SoftwareSerial.cpp:375:60: error: 'digitalPinToPCMSKbit' was not declared in this scope
     _pcint_maskvalue = _BV(digitalPinToPCMSKbit(_receivePin));
```

There is a problem with the Ethernet library due to a name collision with that library's `LISTEN` variable:
```
/home/travis/Arduino/hardware/MegaCore/avr/libraries/Ethernet/src/utility/w5100.h:101:24: error: expected unqualified-id before numeric constant
   static const uint8_t LISTEN      = 0x14;
```
how should I proceed?

MCUdude, 2019-05-19:
The SoftwareSerial library just needed some #ifdefs. But I can't quite figure out what the problem with the ethernet library is. It does work with ATmega128, but not with any of the AT90CAN micros. Do you have any idea what may be wrong?

MCUdude, 2019-05-19:
It should all be good now!

per1234, 2019-05-19:
I rebased my fork and will am running the build again to make sure there aren't any other issues. After that passes, I'll submit the PR.

MCUdude, 2019-05-19:
@per1234 any idea what's going on with [build 91.2](https://travis-ci.org/MCUdude/MegaCore/builds/534443434)?

per1234, 2019-05-19:
codespell's list of commonly misspelled words is continuously being improved. A new version of codespell was released and so it caught new misspelled words (and one false positive). I already fixed the misspelled words and false positive in my fork and will submit that as part of my PR.

geraldjust, 2019-05-20:
Does it have to do with pin change interrupts? The at90can128 doesn't have any on most pins. only the regular int0~7

and checked out the file and seems to work like a charm now!

MCUdude, 2019-05-21:
Big thanks to both of you! I think I can safely close this issue now 😃

closed 2019-05-21 by MCUdude