MCUdude/MegaCore #111: Ethernet library outdated

jandrassy, 2019-05-24:
Hallo Hans. The included Ethernet library is old. The Arduino Ethernet library now supports W5100, W520 and W5500 and has some new functions. It works over SPI library so it is multiplatform now and is not bundled with Arduino AVR core. I think it would be best to remove Ethernet library from MCUdude cores.

MCUdude, 2019-05-24:
Well, I do need to provide a dedicated library in order to be able to compile for the CAN32/64/128. This is because there is a variable in the library that collides with a CAN related register. About a year ago I updated the ethernet library that's bundled with MightyCore. I used Paul Stoffregen's [Ethernet library](https://github.com/PaulStoffregen/Ethernet). I'll update it for MegaCore as well.

I did include the copy_flash_pages functionality as an option when building the bootloader. However, it's not enabled in the pre-compiled binaries. To be honest, I haven't tested it yet. I probably should 😉

closed 2019-05-25 by MCUdude

MCUdude, 2019-05-25:
@per1234 it seems as the last Travis build fails due to tabs, trailing white spaces and missing EOL. To be honest it's very difficult to be able to fix all this in one commit without having Travis to check everything again and again. How do you usually deal with issues like this? If you have a trick up in your sleeve, would you like tu submit a PR to fix this? Thanks!

per1234, 2019-05-26:
I've been thinking a lot lately on how to make it easy for developers to comply with the requirements imposed by Travis CI.

For simply detecting the issues locally, you can run the commands from .travis.yml (the "detection scripts" below).

For fixing up repositories in bulk when I first add a formatting check to its Travis CI build, I have a collection of scripts, which I have provided below ("Fix scripts").

Both the detection and fix scripts require bash. If using Windows, you can install Git for Windows, which includes Git Bash, and then run the scripts using Git Bash.

Git pre-commit hooks are run when you commit to a repository. If the script returns a non-0 exit status, the commit will be aborted. To install a Git pre-commit hook, create a file named `pre-commit` in `.git/hooks`. Note that hooks are installed on a per-repository basis. `.git/hooks` is not tracked by Git so you can't push your hooks to the remote repository to be installed for anyone cloning the repo.

I'm not super happy with the pre-commit hook system I'm using. I wanted the pre-commit hook to run a check identical to that done by Travis CI. That means the repository should be checked as it is at the point of the commit. Unstaged changes should not be checked. To accomplish this, I `git stash` before running the check, then `git stash pop` after. Untracked files are also skipped (which is no trouble). I'm not crazy about the hook automatically messing with the repository's index with the stash operations. The only alternative I've found is to to `git checkout` the repository to a temporary folder, run the check on that folder, and then delete the folder. That's not an ideal solution either, and is more complex since the user should be allowed to configure the location of the folder, but it may end up being preferable to the stash solution. The other option is to forget about the goal of skipping unstaged changes and run the check directly on the entire repository folder. I'd welcome any input on which is best, or any other possible solutions. Maybe @SpenceKonde would have some input or interest in this topic as well.

Let me know if you think it would be useful to add any of this to the repository. I now have a repository that is a collection of Git hook scripts. I'm planning to add the formatting check hooks below to it but I will need to remove the MCUdude *Core-specific paths from them. I'm planning to make it easy to configure the paths to skip via a configuration variable at the top of the script eventually but that means going down the ugly path of `eval`.

### Non-Unix line endings

#### Detection script:
```bash
find . -path './.git' -prune -or -path './avr/travis-ci' -prune -or -path './avr/cores/MCUdude_corefiles' -prune -or -path './avr/bootloaders' -prune -or -type f -exec grep --files-with-matches --binary-files=without-match --regexp=$'\r$' '{}' \; -exec echo 'Non-Unix EOL detected.' \; -exec false '{}' +
```

#### Git pre-commit hook:
```bash
#!/bin/bash

# stash so that unstaged changes are not checked
git stash -q --keep-index

# set default exit status
exitStatus=0

while read -r filename; do
  # check if the file is tracked by git
  if ! git ls-files --error-unmatch "$filename" &>/dev/null; then
    # don't check untracked files
    continue
  fi

  if grep --files-with-matches --binary-files=without-match --regexp=$'\r$' "$filename"; then
    echo "ERROR: Non-Unix EOL detected in $filename"
    # make the hook fail
    exitStatus=1
  fi
done <<<"$(find . -path './.git' -prune -or -path './avr/travis-ci' -prune -or -path './avr/cores/MCUdude_corefiles' -prune -or -path './avr/bootloaders' -prune -or -type f)"

git stash pop -q

exit $exitStatus
```

#### Fix script:
```bash
# find all files under the current directory that have non-Unix EOL and convert them to Unix EOL
find . -path './.git' -prune -or -type f -exec grep --files-with-matches --binary-files=without-match --regexp=$'\r$' '{}' \; -exec echo 'Non-Unix EOL detected.' \; -exec dos2unix '{}' \; -exec false '{}' +
# Pause so that the output may be examined before script window is exited
read -rsp $'\n\nFinished converting files to Unix EOL. Press any key to continue...\n' -n 1
```

### Trailing whitespace

#### Detection script:
```bash
find . -path './.git' -prune -or -path './avr/travis-ci' -prune -or -path './avr/cores/MCUdude_corefiles' -prune -or -path './avr/bootloaders' -prune -or -type f -exec grep --with-filename --line-number --binary-files=without-match --regexp='[[:blank:]]$' '{}' \; -exec echo 'Trailing whitespace found.' \; -exec false '{}' +
```

#### Git pre-commit hook:
```bash
#!/bin/bash

# stash so that unstaged changes are not checked
[... 274 lines cut ...]
    # there is no ignore words file so don't use the --ignore-words option
    if ! codespell "$filename"; then
      echo "ERROR: Misspelled word found in $filename"
      # make the hook fail
      exitStatus=1
    fi
  else
    if ! codespell --ignore-words="$ignoreWordsListPath" "$filename"; then
      echo "ERROR: Misspelled word found in $filename"
      # make the hook fail
      exitStatus=1
    fi
  fi
done <<<"$(find . -path './.git' -prune -or -path './avr/bootloaders' -prune -or -path './avr/cores' -prune -or -path './avr/travis-ci' -prune -type f)"

git stash pop -q

exit $exitStatus
```

#### Fix script:
```bash
codespell --write-changes --skip="./.git","./avr/bootloaders","./avr/cores","./avr/travis-ci" --ignore-words="./avr/travis-ci/etc/codespell-ignore-words-list.txt"
```
NOTE: `codespell -w` will only automatically fix misspelled words when there is only one suggested fix in the database. In the event there are multiple suggested fixes in the database, codespell will display the misspelled word, but not automatically fix it. You will need to manually fix the misspelled word.

MCUdude, 2019-05-26:
Thank you so much for the scripts! As for now, it's easier for me to just manually run these detection scripts. I only really need them when I'm updating code from another repository. Do you know if there is a CRLF check as well? I prefer LF line endings only.

per1234, 2019-05-26:
You're welcome. I'm glad if they can be of use. The "Non-Unix line endings" script checks for CRLF (Windows EOL) or CR (Macintosh EOL).

MCUdude, 2019-05-26:
I forgot to check all files for extra blank lines before I pushed the code, so the current build will also fail. But when trying to run the script on my mac I get this output:

```
$ find . -path './.git' -prune -or -path './avr/travis-ci' -prune -or -path './avr/cores/MCUdude_corefiles' -prune -or -path './avr/bootloaders' -prune -or -type f -print0 | xargs -0 -L1 bash -c 'if test "$(grep --files-with-matches --binary-files=without-match --max-count=1 --regexp='.*' "$0")" && test "$(tail --bytes=1 "$0")"; then echo "No new line at end of $0."; false; fi'
tail: illegal option -- -
usage: tail [-F | -f | -r] [-q] [-b # | -c # | -n #] [file ...]
```

It seems like the version bundled with macOS is a little bit different than on Ubuntu for instances. How can I run this on my mac?

per1234, 2019-05-26:
The error message is confusing. I guess it's the `--bytes` option that's not supported? I prefer to use long option names for scripts because it makes the command more self documenting, but I have noticed that sometimes the long option names aren't as well supported. Try this:
```bash
find . -path './.git' -prune -or -path './avr/travis-ci' -prune -or -path './avr/cores/MCUdude_corefiles' -prune -or -path './avr/bootloaders' -prune -or -type f -print0 | xargs -0 -L1 bash -c 'if test "$(grep --files-with-matches --binary-files=without-match --max-count=1 --regexp='.*' "$0")" && test "$(tail -b 1 "$0")"; then echo "No new line at end of $0."; false; fi'
```

per1234, 2019-05-26:
Sorry, I got that wrong it should be `-c`, not `-b`:
```bash
find . -path './.git' -prune -or -path './avr/travis-ci' -prune -or -path './avr/cores/MCUdude_corefiles' -prune -or -path './avr/bootloaders' -prune -or -type f -print0 | xargs -0 -L1 bash -c 'if test "$(grep --files-with-matches --binary-files=without-match --max-count=1 --regexp='.*' "$0")" && test "$(tail -c 1 "$0")"; then echo "No new line at end of $0."; false; fi'
```

MCUdude, 2019-05-26:
Thanks, this worked!

SpenceKonde, 2019-05-28:
Is there a way to do a similar thing automatically if you normally use the windows github client, rather than raw git on Linux? I'm just not that hardcore, but since I started using travis, at least 50% of my commits have been to fix trailing whitespace, codespell, or missing end of line on files. u_u

per1234, 2019-05-28:
The Git hooks I shared above will work with a Git client as well as when using Git from command line.

Please let me know if you have any troubles with the hooks or ideas for how this can be made better. I only started working with Git hooks less than a week ago and the more I use them, the more I'm feeling like the `git stash`/`git stash pop` approach I settled on is not the way to go. The alternative of checking out to a temporary folder where the check happens would be pretty slow with a monster repository like the MCUdude cores though.

I use a Git client ([Git Extensions](https://github.com/gitextensions/gitextensions)) 97% of the time because I find a good GUI to be faster than typing out commands on the command line. I only use the command line for things the GUI doesn't do, or does poorly.

jandrassy, 2019-05-29:
Hans, you could PR the changes for the Ethernet library to the library repo. There are important pull requests now, so I expect a release in the near future. Support for W6100 and for MKR Vigor are those important PRs

per1234, 2019-05-29:
@jandrassy by "library repo" do you mean the repo for the official Arduino Ethernet library:
https://github.com/arduino-libraries/ethernet

I know MCUdude already submitted one to the PaulStoffregen/Ethernet repository for compatibility with the AT90CAN32/64/128: https://github.com/PaulStoffregen/Ethernet/pull/33

I do think it would be ideal to get that change made in the Arduino Ethernet library because that might pave the way to one day being able to remove the Ethernet library from this repository. That would eliminate the maintenance burden of keeping a bundled copy of the library up to date in this repo and simplify the Travis CI build a bit. I believe that most of the enhancements from PaulStoffregen/Ethernet were incorporated into arduino-libraries/Ethernet for the 2.0.0 release of the library.

jandrassy, 2019-05-29:
> @jandrassy by "library repo" do you mean the repo for the official Arduino Ethernet library:
> https://github.com/arduino-libraries/ethernet
yes.

per1234, 2019-06-03:
I have published the Git hooks and "fix scripts" I shared above (minus the MCUdude core-specific folder exclusions) to this repository:
https://github.com/per1234/formatting-checks

commit dbf5f6c1 (2019-05-25): Update Ethernet library
Closes #111