MCUdude/MegaCore #112: Travis is just too picky

MCUdude, 2019-06-22:
@per1234 At the moment I feel like I'm working against rather than with Travis.

Almost every time push something It complains about silly code and text formatting that has no impact on the looks or the functionality. I'd definitely prefer a warning in the output log (yellow text?), but it's not significant enough to break the build.

Here's what I want to convert into warnings instead.
* Tailing whitespaces
  - Most editors add this automatically when hitting enter. I don't see any reason why tailing white spaces is a problem.
* More than one blank line at the end of a file
* Spelling

Is this doable?

per1234, 2019-06-22:
>I'd definitely prefer a warning in the output log

Sure, I can do that. I'll prepare a PR now.

>Most editors add this automatically when hitting enter

I've never seen an editor add trailing whitespace, any editor that does that is absolute garbage.

>I don't see any reason why tailing white spaces is a problem.

My opinion is that every character in code or documentation should have a purpose. Trailing whitespace has no purpose. Any character that has no purpose is a problem because it pollutes the diff. Avoiding trailing whitespace is a basic skill any developer should be able to achieve without effort.

MCUdude, 2019-06-23:
> > Most editors add this automatically when hitting enter

> I've never seen an editor add trailing whitespace, any editor that does that is absolute garbage.

The editor I'm often using (BBEdit) does this when I hit enter twice. I should probably use something else, I know. Another issue is trailing white spaces in markdown. Two blank spaces at the end of a line indicate a line shift. Travis complained about this too.

>> I don't see any reason why tailing white spaces is a problem.

> My opinion is that every character in code or documentation should have a purpose. Trailing whitespace has no purpose. Any character that has no purpose is a problem because it pollutes the diff. Avoiding trailing whitespace is a basic skill any developer should be able to achieve without effort.

Well, I do agree that it can pollute a diff.

closed 2019-06-23 by MCUdude

per1234, 2019-06-23:
>Another issue is trailing white spaces in markdown. Two blank spaces at the end of a line indicate a line shift.

Yeah, that's the one thing I don't like about Markdown I don't know what they were thinking. Actually I think it's really silly that they didn't just treat a newline as a newline. There is the alternative of adding a backslash at the end of the line to force a line break. That's much better than trailing whitespace, but still not intuitive since in C++/C we use that same syntax for what is essentially the opposite purpose.