MCUdude/MegaCore #118: Can't use name "AVR" in a struct

kbickham, 2019-08-05:
For a school project I'm building a machine. I've fixed the error, but this is a very peculiar bug.
 
I have the following struct defined, note that I call one member "AVR1"  ...

typedef struct
{ float irtemp[4];
  float lmtemp[4];
  long weight;
  int AVRstate;
  board2 AVR1;
  board2 ESP;  
  timer CHECKT[5];
  dhtDAT INTAKE;
  dhtDAT EXHAUST;
  dhtDAT AMBIENT;
  air AIR;
  dryp DRYP;
  heat HEAT; 
  disp DISP;
  loadc LC;
} SYSstate, *pSYSstate;

the board 2 struct is as follows: 

typedef struct
{ timer STATE[6];
  int currentstate; 
  sensors2 SENSOR;
} board2, *pboard2;

If instantiate as simply "AVR" i get the following error message at my first instance of the word, a switch case 

....

void loop()                                          
{   genie.DoEvents();

        switch(HDC.AVR.currentstate)
          {  case 0: break;
       
          };
....

Yielding this error message upon compilation:

"C:\\Users\\Lauren\\Downloads\\arduino-nightly\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -MMD -flto -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10810 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR "-IC:\\Users\\Lauren\\Downloads\\arduino-nightly\\hardware\\arduino\\avr\\cores\\arduino" "-IC:\\Users\\Lauren\\Downloads\\arduino-nightly\\hardware\\arduino\\avr\\variants\\mega" "-IC:\\Users\\Lauren\\Downloads\\arduino-nightly\\hardware\\arduino\\avr\\libraries\\Wire\\src" "-IC:\\Users\\Lauren\\Documents\\Arduino\\libraries\\DHT_sensor_library" "-IC:\\Users\\Lauren\\Documents\\Arduino\\libraries\\genieArduino\\src" "-IC:\\Users\\Lauren\\Documents\\Arduino\\libraries\\HX711_ADC\\src" "-IC:\\Users\\Lauren\\Documents\\Arduino\\libraries\\EasyTransfer" "-IC:\\Users\\Lauren\\Documents\\Arduino\\libraries\\Adafruit_Unified_Sensor" "C:\\Users\\Lauren\\AppData\\Local\\Temp\\arduino_build_257593\\sketch\\HDCnewimplementstates.ino.cpp" -o "C:\\Users\\Lauren\\AppData\\Local\\Temp\\arduino_build_257593\\sketch\\HDCnewimplementstates.ino.cpp.o"
C:\Users\Lauren\Desktop\SUNDAY WORK FOLDER\HDCnewimplementstates\HDCnewimplementstates.ino: In function 'void loop()':

HDCnewimplementstates:206:20: error: expected unqualified-id before numeric constant

         switch(HDC.AVR.currentstate)

                    ^

HDCnewimplementstates:206:20: error: expected ')' before numeric constant

Using library Wire at version 1.0 in folder: C:\Users\Lauren\Downloads\arduino-nightly\hardware\arduino\avr\libraries\Wire 
Using library DHT_sensor_library at version 1.3.7 in folder: C:\Users\Lauren\Documents\Arduino\libraries\DHT_sensor_library 
Using library genieArduino at version 1.4.5 in folder: C:\Users\Lauren\Documents\Arduino\libraries\genieArduino 
Using library HX711_ADC at version 1.1.5 in folder: C:\Users\Lauren\Documents\Arduino\libraries\HX711_ADC 
Using library EasyTransfer in folder: C:\Users\Lauren\Documents\Arduino\libraries\EasyTransfer (legacy)
Using library Adafruit_Unified_Sensor at version 1.0.3 in folder: C:\Users\Lauren\Documents\Arduino\libraries\Adafruit_Unified_Sensor 
exit status 1
expected unqualified-id before numeric constant

Now I've spend hours, and I'm just renaming the object, as this leads to everything working fine. Just thought I'd mention it.

MCUdude, 2019-08-07:
Well, that doesn't surprise me. There's a lot of reserved keywords in the Arduino framework and avr-libc. Not all of them are color highlighted in the IDE, so It's hard to know. It isn't really a bug though.

closed 2019-08-07 by MCUdude

per1234, 2019-08-07:
From https://www.microchip.com/webdoc/avrlibcreferencemanual/using_tools_1using_avr_gcc_mach_opt.html:
>The macro `AVR` will be defined as well when using the standard levels gnu89 (default) and gnu99 but not with c89 and c99.

They don't mention `-std=gnu11` and `-std=gnu++11`, as used in this repository (apparently that documentation hasn't been updated in a really long time), but it also applies to those. It's really just the gnu/c that determines whether it's defined or not.

kbickham, 2019-08-07:
Well, thank you. I did not know this.