MCUdude/MegaCore #119: Sketch is erased on each reboot

ilium007, 2019-09-01:
Not sure why this is happening but after loading this bootloader to a 2560 each time the board is reset the sketch is lost, uploading the sketch again enables it to work until next reset.

MCUdude, 2019-09-01:
I'm not able to reproduce this. Does the LED flash two times when you press the reset button?

I burned a 16 MHz bootloader to an Arduino Mega 2560, and have uploaded the blink example several times. Everything works as expected here!

ilium007, 2019-09-01:
For some reason now I cannot burn the bootloader. I am using a Arduino Uno with the ArduinoISP sketch loaded (the USBTinyISP board I bought doesn't work with the 2560 i found out)

When I use the Uno as ArduinoISP and with the MCUdude 2560 board selected I get this error (still looking for the USBTinyISP)

`avrdude: Version 6.3-20171130
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "/Users/brantwinter/Library/Arduino15/packages/MegaCore/hardware/avr/2.0.2/avrdude.conf"
         User configuration file is "/Users/brantwinter/.avrduderc"

         Using Port                    : usb
         Using Programmer              : arduinoisp
avrdude: Error: Could not find USBtiny device (0x2341/0x49)

avrdude done.  Thank you.

Error while burning bootloader.`

MCUdude, 2019-09-01:
You can't communicate with the programmer because you've chosen the wrong programmer option. Select *Arduino as ISP* instead of *ArduinoISP*. I know, it's confusing.

ilium007, 2019-09-01:
Arghg - ok thanks. Why on earth is the sketch called ArduinoISP !!!!

So it went further but then got this error (I've seen before, not sure how I fixed)

[image]

ilium007, 2019-09-01:
```
avrdude: Version 6.3-20171130
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "/Users/brantwinter/Library/Arduino15/packages/MegaCore/hardware/avr/2.0.2/avrdude.conf"
         User configuration file is "/Users/brantwinter/.avrduderc"

         Using Port                    : /dev/cu.usbmodem14301
         Using Programmer              : arduino
         Overriding Baud Rate          : 19200
         AVR Part                      : ATmega2560
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PA0
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    10     8    0 no       4096    8      0  9000  9000 0x00 0x00
           flash         65    10   256    0 yes    262144  256   1024  4500  4500 0x00 0x00
           lfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           lock           0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : Arduino
         Description     : Arduino
         Hardware Version: 2
         Firmware Version: 1.18
         Topcard         : Unknown
         Vtarget         : 0.0 V
         Varef           : 0.0 V
         Oscillator      : Off
         SCK period      : 0.1 us

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude: Device signature = 0x000000 (retrying)

Reading | ################################################## | 100% 0.01s

avrdude: Device signature = 0x000000 (retrying)

Error while burning bootloader.
Reading | ################################################## | 100% 0.01s

avrdude: Device signature = 0x000000
avrdude: Yikes!  Invalid device signature.
         Double check connections and try again, or use -F to override
         this check.

avrdude done.  Thank you.

```

ilium007, 2019-09-01:
I changed the 10uF cap for another on the programmer board and it worked - go figure. Thanks and apologies for the bug report.

closed 2019-09-01 by ilium007

MCUdude, 2019-09-01:
No worries! Thank you for explaining so clearly what you did in order to solve this 🙂